iiab/iiab #3057: Internal WiFi not working on a Old PC (Panasonic Toughbook CF-52)

icebox827, 2021-12-02:
# Internal WiFi Hotspot issue

Unable to use iaab internal WiFi hotspot on a Old PC (Panasonic CF-52) running Ubuntu desktop 21.10

## Diagnostic links

- http://sprunge.us/9zTw1Z?en

holta, 2021-12-02:
I see on Line 467 of http://sprunge.us/9zTw1Z?en that you also tried `wifi_up_down: false`

holta, 2021-12-02:
> http://sprunge.us/9zTw1Z?en

@jvonau might possibly know whether this laptop has any hope (or not) to run **AP Mode** ?

Some say Mint has somewhat better driver/firmware support than Ubuntu on older laptops.  But I don't know if that's proven true.

Certainly the Toughbook is unlikely to be able to run [AP+STA](https://github.com/iiab/iiab/issues/2853#issuecomment-934372625) if indeed its internal WiFi cannot be put into AP Mode...?

jvonau, 2021-12-03:
No duel mode available when the upstream wifi is using 5GHz as shown below, while the hotspot is always 2.4GHz on channels 1-12 only.
```
 395 host_channel = 6
 396 client_wifi_channel = 149
1365 		channel 149 (5745 MHz), width: 40 MHz, center1: 5755 MHz
```

holta, 2021-12-03:
> No [dual] mode available when the upstream wifi is using 5GHz as shown below, while the hotspot is always 2.4GHz on channels 1-12 only.
> 
> ```
>  395 host_channel = 6
>  396 client_wifi_channel = 149
> 1365 		channel 149 (5745 MHz), width: 40 MHz, center1: 5755 MHz
> ```

Thanks @jvonau for spotting that:

@icebox827 can you clarify &mdash; did you run `sudo ./iiab-network` (in /opt/iiab/iiab) and then reboot &mdash; after you set `wifi_up_down: false` in /etc/iiab/local_vars.yml ?

(I'd assume that would be your best shot at achieving a 2.4GHz internal hotspot?  If there's any remaining chance in the universe...that this Toughbook CF-52 from almost a decade ago...might actually be capable of AP Mode ;-)

jvonau, 2021-12-03:
> I see on Line 467 of http://sprunge.us/9zTw1Z?en that you also tried `wifi_up_down: false`

That is a bit of a red herring, the actions below do not reflect the state of the machine's setting file at the time of running iiab-network, doesn't jive with ap0 = True note the changed and skipping with the task being first and the result below.
```
2275 2021-12-01 21:05:55,945 p=75776 u=root n=ansible | TASK [network : Copy ap0-manage.conf for NetworkManager] ************************************************************************************************************************************************
2276 2021-12-01 21:05:56,472 p=75776 u=root n=ansible | changed: [127.0.0.1]
2277 2021-12-01 21:05:56,487 p=75776 u=root n=ansible | TASK [network : Copy manage.conf for NetworkManager when wifi_up_down False] *******************************************************************************************************************************
2278 2021-12-01 21:05:56,512 p=75776 u=root n=ansible | skipping: [127.0.0.1]
2279 2021-12-01 21:05:56,531 p=75776 u=root n=ansible | TASK [network : Remove manage.conf for NetworkManager when wifi_up_down True] ******************************************************************************************************************************
2280 2021-12-01 21:05:56,840 p=75776 u=root n=ansible | ok: [127.0.0.1]
```

One really needs to run iiab-hotspot-off before changing wifi_up_down: false **before** running iiab-network, bit of a quagmire when the uplink is wifi and ssh'd in but straight forward when at the desktop.

1. manually dis-associate/disconnect the wifi uplink because of the wifi is 5GHz... This is the main issue, don't connect and the hotspot should work on 2.4GH without messing around much.
2. toggle wifi_up_down: to False if that is what you **really** want, think most just would want to disable/disconnect the 5GHz upstream connection at this point to see if the hotspot comes up. Not sure a connection will come up? just jack a ethernet cable into it.
3. iiab-hotspot-off because we need hostapd in a static state while the files are replaced.
4. iiab-network replaces key template files with the setting changed above, iiab-hotspot-on|off have different contents depending on the setting of wifi_up_down, different interfaces are created or torn down, that is the reason for the hotspot to be in a stopped state while the transition occurs.
5. Now being set to 'False' the upstream wifi is still connected and operational for the sake of being ssh'd in over the wifi, until you run `iiab-hotspot-on`and the machine is rebooted.
6. On a desktop where you have working control over the GUI never try to use upstream wifi while the wifi_up_down: is set to False that a) won't work or b) stop the hotspot from working after trying until you reboot.

jvonau, 2021-12-03:
What can be better expressed on https://download.iiab.io/ ?
>An Ethernet cable is HIGHLY RECOMMENDED during installation, as this is more reliable than Wi-Fi (and faster too!) WARNING: IF YOU CONNECT YOUR IIAB'S INTERNAL WIFI TO THE INTERNET OVER 5 GHz, YOU'LL PREVENT OLDER LAPTOPS/PHONES/TABLETS (WHICH REQUIRE 2.4 GHz) FROM CONNECTING TO YOUR IIAB'S INTERNAL HOTSPOT. See: "wifi_up_down: True" in /etc/iiab/local_vars.yml

holta, 2021-12-03:
Profound Thanks @jvonau for the detailed feedback for @icebox827.

He is new to Internet-in-a-Box but extremely enthusiastic.

I'm assuming he has a backup plan if the old Toughbook CF-52 doesn't work out.  But it's almost impossible to find Raspberry Pi's in Haiti and the Dominican Republic.  So he has to start with the equipment he has.

icebox827, 2021-12-03:
Missing diagnostics URL: 

Thanks @jvonau for your help 

Here are the steps I ran, along with the diagnostics URL): 
1. Turn off Wifi Manually 
2. Connect the computer to an Ethernet cable 
3. Run iiab-hotspot-off 
4. Reboot the computer 
5. Run iiab-network
6. Reboot the computer
7. Run iiab-hotspot-on
8. Run iiab-network
9. Reboot the computer

It was very instructive, and a great opportunity to learn more, but unfortunately the situation remain the same.

**iiab-diagnostics:** http://sprunge.us/dCbzG5

jvonau, 2021-12-03:
```
COMMAND: /usr/bin/systemctl status hostapd    # Downstream Wi-Fi: Is hostapd running?

× hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
     Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2021-12-03 13:55:13 AST; 25min ago
   Main PID: 721 (code=exited, status=1/FAILURE)
        CPU: 10ms

dic 03 13:55:13 box hostapd[721]: Could not set interface wlp11s0 flags (UP): Operation not possible due to RF-kill
```

There is a hardware button or a software setting that has been enabled so that RF-kill has been come active and is actively blocking the WiFi. 
```
-rw------- 1 root root 355 dic  1 14:00 /etc/NetworkManager/system-connections/Altice a3999f.nmconnection
``` 
This connection should be removed if you are within range of the AP, being a "system-connections" this one will always be tried if the interface is not RF-kill blocked. You could try moving out of range of the AP and leave the file in place to test a theory. Think the GUI will try to scan every so often, depending on the chipset the AP functionality may not auto recover, I find things more stable with a non-functional upstream wifi entry removed.

jvonau, 2021-12-04:
```
Dec  3 14:11:01 box CRON[6006]: (www-data) CMD (/usr/bin/php /opt/iiab/moodle/admin/cli/cron.php >/dev/null)
Dec  3 14:11:18 box internetarchive[999]: 2021-12-03T18:11:18.709Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:11:19 box internetarchive[999]: 2021-12-03T18:11:19.032Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
Dec  3 14:11:48 box internetarchive[999]: 2021-12-03T18:11:48.709Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:11:49 box internetarchive[999]: 2021-12-03T18:11:49.093Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
Dec  3 14:12:01 box CRON[6020]: (www-data) CMD (/usr/bin/php /opt/iiab/moodle/admin/cli/cron.php >/dev/null)
Dec  3 14:12:18 box internetarchive[999]: 2021-12-03T18:12:18.710Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:12:19 box internetarchive[999]: 2021-12-03T18:12:19.050Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
Dec  3 14:12:20 box systemd[3859]: app-gnome-org.gnome.DejaDup.Monitor-5518.scope: Deactivated successfully.
Dec  3 14:12:48 box internetarchive[999]: 2021-12-03T18:12:48.710Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:12:49 box internetarchive[999]: 2021-12-03T18:12:49.078Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
Dec  3 14:13:01 box CRON[6028]: (www-data) CMD (/usr/bin/php /opt/iiab/moodle/admin/cli/cron.php >/dev/null)
Dec  3 14:13:18 box internetarchive[999]: 2021-12-03T18:13:18.711Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:13:19 box internetarchive[999]: 2021-12-03T18:13:19.003Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
```
internetarchives is spamming the logs every 30 seconds with an error, that might be an issue on a long running machine.

jvonau, 2021-12-04:
> Turned off Wifi Manually (top-right of Ubuntu's Desktop)

That is the RF-kill in action, I said to dis-associate/disconnect from from the uplink, not turn the wifi chipset off altogether.

Here is what I would do:
1. Leave the ethernet connected 
2. Delete the upstream wifi file /etc/NetworkManager/system-connections/Altice a3999f.nmconnection
3. Reboot
4. The GUI might have other stored wifi connection information so when you turn the wifi back on ensure none are connected, if connected use 'disconnect' that is provided when you access the menu.
5. Edit local_vars.yml's wifi_up_down to your choice, I would prefer the stock setting be maintained. If you are using False you promise to never ever touch the wifi menu in the GUI(think I locked it out well enough that it is a pita to undo), using True you promise to run iiab-hotspot-off before connecting with other APs in the menu then run iiab-hotspot-on.
6. `cd /opt/iiab/iiab` `sudo ./iiab-network`
7. Quick testing to see if it's working, don't panic if it's not some things take a reboot to straighten out easily, then reboot
8. Retest and report.

@holta the comment at https://github.com/iiab/iiab/issues/2863#issuecomment-907613218

icebox827, 2021-12-06:
Interesting @jvonau 

I've done the following:

1. Left the ethernet connected
2. Deleted the the upstream WiFi file /etc/NetworkManager/system-connections/Altice a3999f.nmconnection
3. Edited /etc/iiab/local_vars.yml and set wifi_up_down to True
4. cd /opt/iiab/iiab
5. Ran `git pull` to install more informative iiab-diagnostics command
6. Ran `./runrole 0-init`
7. Rebooted the Toughbook
8. Ran `iiab-hotspot-off`
9. Rebooted the Toughbook
10. Ran `iiab-hotspot-on`
11. Rebooted the Toughbook
12. Ran `sudo ./iiab-network`
13. Rebooted the Toughbook

Tried to connect my smartphone to IIAB (but WiFi SSID "Internet in a Box" does not appear on the air). When running `rfkill list` the output is disappointing:  *line 1330 	Soft blocked: yes*

*iiab-diagnostics*: http://sprunge.us/diMAk0?en

jvonau, 2021-12-06:
> Thanks @jvonau for your help.
> 
> Here's what I tried:
> 
>     1. Turned off Wifi Manually (top-right of Ubuntu's Desktop)

This is what you need to undo, however you managed to to that... this is what mine looks like with the MATE desktop note the checkbox, yours may differ but much the same idea.
[image]

icebox827, 2021-12-06:
Thanks @jvonau 

Attached are some interesting screenshot

[image]

[image]

I ran the tests for **wifi_up_down: True** and **wifi_up_down: False**

1. Manually disconnect the WiFi
2. Ran `iiab-hotspot-off`
3. Rebooted the Toughbook
4. Ran `iiab-hotspot-on`
5. Rebooted the Toughbook
6. Ran `sudo ./iiab-network` in /opt/iiab/iiab
7. Rebooted the Toughbook

Can't connect my smartphone to IIAB (WiFi SSID "Internet in a Box" does not appear on the air).

**iiab-dignostics :** http://sprunge.us/FVrmTN?en

jvonau, 2021-12-06:
```
1385 	Supported interface modes:
1386 		 * IBSS
1387 		 * managed
1388 		 * monitor

```
Missed this earlier, looks your wifi chipset does not support becoming an access point at all, 'AP' is missing from the listing. Other hints below.
```
1508 	software interface modes (can always be added):
1509 		 * monitor
1510 	interface combinations are not supported
```
This would denote that duel mode with ap0 would be a no go. 

```
1569 dic 06 10:09:52 box hostapd[717]: Configuration file: /etc/hostapd/hostapd.conf
1570 dic 06 10:09:52 box hostapd[717]: nl80211: Could not configure driver mode
1571 dic 06 10:09:52 box hostapd[717]: nl80211: deinit ifname=wlp11s0 disabled_11b_rates=0
1572 dic 06 10:09:52 box hostapd[717]: nl80211 driver initialization failed.
```
Do you know the type of chipset the laptop uses?  manufacture & model?

icebox827, 2021-12-06:
@jvonau 

According to the [tech specs](https://www.hbrtech.com/panasonic/cf52.aspx), I'm using a `Intel WiFi Link 5100, Bluetooth 2.0 + EDR; optional GOBI 3G global mobile Internet`
Please take a look at the [tech specs](https://www.hbrtech.com/panasonic/cf52.aspx) for more informations

holta, 2021-12-06:
Gigantic Thank You to @jvonau AND @icebox827, who've work so hard on this over 4 days now, even if the final news isn't pure fireworks:

_This ticket will help **many other people** to understand this process of (trying!) to get their old laptops working with IIAB &mdash; thanks to your hard work having now published key details as de facto documentation &mdash; that others worldwide can now Google (search for) and read~_

jvonau, 2021-12-07:
`lspci -nn | grep Network` should give the kernel's take on the hardware, advertisements for the hardware can be misleading.

holta, 2021-12-07:
> `lspci -nn | grep Network` should give the kernel's take on the hardware, advertisements for the hardware can be misleading.

Is the entire `lspci -nn` command important enough to be included in [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) ?

(Raspberry Pi OS output below, on RPi 4, doesn't contain the word 'Network' FWIW...)

```
root@box:~# lspci -nn
00:00.0 PCI bridge [0604]: Broadcom Limited Device [14e4:2711] (rev 10)
01:00.0 USB controller [0c03]: VIA Technologies, Inc. VL805 USB 3.0 Host Controller [1106:3483] (rev 01)

jvonau, 2021-12-07:
> Gigantic Thank You to @jvonau AND @icebox827, who've worked so hard on this over 4 days now, even if the final news isn't pure fireworks:
> 
> _This ticket will help **many other people** to understand this process of (trying!) to get their old laptops working with IIAB — thanks to your hard work having now published key details as de facto documentation — that others worldwide can now Google (search for) and read~_

Might help if developer instructions are followed to the tee in place of doing what the user thinks is the best course of action. Always run `iiab-hotspot-off` before making any network changes like the upstream WiFi connection in a GUI or edits to local_vars.yml especially network's **'wifi_up_down'** before running ./iiab-network and finally iiab-hotspot-on.

> Is the entire `lspci -nn` command important enough to be included in [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) ?
>
> (Raspberry Pi OS output below, [...]

That is the only hardware that is connected to the pci bus on that RPi. Add the command if you want, drop capturing iiab-gen-iptables there is no useful information now that the ports are not shown in a human readable format with the use of `iiab_var_value()`.

icebox827, 2021-12-08:
Hi @jvonau 

1. Ran `lspci -nn`
```
00:00.0 Host bridge [0600]: Intel Corporation Core Processor DRAM Controller [8086:0044] (rev 02)
00:01.0 PCI bridge [0604]: Intel Corporation Core Processor PCI Express x16 Root Port [8086:0045] (rev 02)
00:19.0 Ethernet controller [0200]: Intel Corporation 82577LM Gigabit Network Connection [8086:10ea] (rev 06)
00:1a.0 USB controller [0c03]: Intel Corporation 5 Series/3400 Series Chipset USB2 Enhanced Host Controller [8086:3b3c] (rev 06)
00:1b.0 Audio device [0403]: Intel Corporation 5 Series/3400 Series Chipset High Definition Audio [8086:3b56] (rev 06)
00:1c.0 PCI bridge [0604]: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 1 [8086:3b42] (rev 06)
00:1c.1 PCI bridge [0604]: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 2 [8086:3b44] (rev 06)
00:1c.2 PCI bridge [0604]: Intel Corporation 5 Series/3400 Series Chipset PCI Express Root Port 3 [8086:3b46] (rev 06)
00:1d.0 USB controller [0c03]: Intel Corporation 5 Series/3400 Series Chipset USB2 Enhanced Host Controller [8086:3b34] (rev 06)
00:1e.0 PCI bridge [0604]: Intel Corporation 82801 Mobile PCI Bridge [8086:2448] (rev a6)
00:1f.0 ISA bridge [0601]: Intel Corporation QM57 Chipset LPC Interface Controller [8086:3b07] (rev 06)
00:1f.2 SATA controller [0106]: Intel Corporation 5 Series/3400 Series Chipset 6 port SATA AHCI Controller [8086:3b2f] (rev 06)
00:1f.3 SMBus [0c05]: Intel Corporation 5 Series/3400 Series Chipset SMBus Controller [8086:3b30] (rev 06)
00:1f.6 Signal processing controller [1180]: Intel Corporation 5 Series/3400 Series Chipset Thermal Subsystem [8086:3b32] (rev 06)
01:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Madison [Mobility Radeon HD 5650/5750 / 6530M/6550M] [1002:68c1]
01:00.1 Audio device [0403]: Advanced Micro Devices, Inc. [AMD/ATI] Redwood HDMI Audio [Radeon HD 5000 Series] [1002:aa60]
0b:00.0 Network controller [0280]: Intel Corporation Centrino Advanced-N 6200 [8086:422c] (rev 35)
0c:00.0 CardBus bridge [0607]: Ricoh Co Ltd RL5c476 II [1180:0476] (rev ba)
0c:00.1 FireWire (IEEE 1394) [0c00]: Ricoh Co Ltd R5C832 IEEE 1394 Controller [1180:0832] (rev 04)
0c:00.2 SD Host controller [0805]: Ricoh Co Ltd R5C822 SD/SDIO/MMC/MS/MSPro Host Adapter [1180:0822] (rev 21)
3f:00.0 Host bridge [0600]: Intel Corporation Core Processor QuickPath Architecture Generic Non-core Registers [8086:2c62] (rev 02)
3f:00.1 Host bridge [0600]: Intel Corporation Core Processor QuickPath Architecture System Address Decoder [8086:2d01] (rev 02)
3f:02.0 Host bridge [0600]: Intel Corporation Core Processor QPI Link 0 [8086:2d10] (rev 02)
3f:02.1 Host bridge [0600]: Intel Corporation 1st Generation Core i3/5/7 Processor QPI Physical 0 [8086:2d11] (rev 02)
3f:02.2 Host bridge [0600]: Intel Corporation 1st Generation Core i3/5/7 Processor Reserved [8086:2d12] (rev 02)
3f:02.3 Host bridge [0600]: Intel Corporation 1st Generation Core i3/5/7 Processor Reserved [8086:2d13] (rev 02)
```
2. Ran `lscpi -nn | grep -i net`
```
00:19.0 Ethernet controller [0200]: Intel Corporation 82577LM Gigabit Network Connection [8086:10ea] (rev 06)
0b:00.0 Network controller [0280]: Intel Corporation Centrino Advanced-N 6200 [8086:422c] (rev 35)
```

holta, 2021-12-08:
> Always run `iiab-hotspot-off` before making any network changes like the upstream WiFi connection in a GUI or edits to local_vars.yml especially network's **'wifi_up_down'** before running ./iiab-network and finally iiab-hotspot-on.

Excellent Reminder!

( FYI another version of much this same warning is posted at:
https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#whats-upgraded )

jvonau, 2022-04-13:
What is the remaining issue here?

holta, 2022-04-13:
> What is the remaining issue here?

I don't think there is one, let's close it!

closed 2022-04-13 by holta