iiab/iiab #3058: internetarchive log file is growing out of control

holta, 2021-12-04:
@mitra42 can you recommend a solution here please if possible?  Thank if you can!

`/var/log/syslog` shows :

```
Dec  3 14:11:01 box CRON[6006]: (www-data) CMD (/usr/bin/php /opt/iiab/moodle/admin/cli/cron.php >/dev/null)
Dec  3 14:11:18 box internetarchive[999]: 2021-12-03T18:11:18.709Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:11:19 box internetarchive[999]: 2021-12-03T18:11:19.032Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
Dec  3 14:11:48 box internetarchive[999]: 2021-12-03T18:11:48.709Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:11:49 box internetarchive[999]: 2021-12-03T18:11:49.093Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
Dec  3 14:12:01 box CRON[6020]: (www-data) CMD (/usr/bin/php /opt/iiab/moodle/admin/cli/cron.php >/dev/null)
Dec  3 14:12:18 box internetarchive[999]: 2021-12-03T18:12:18.710Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:12:19 box internetarchive[999]: 2021-12-03T18:12:19.050Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
Dec  3 14:12:20 box systemd[3859]: app-gnome-org.gnome.DejaDup.Monitor-5518.scope: Deactivated successfully.
Dec  3 14:12:48 box internetarchive[999]: 2021-12-03T18:12:48.710Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:12:49 box internetarchive[999]: 2021-12-03T18:12:49.078Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
Dec  3 14:13:01 box CRON[6028]: (www-data) CMD (/usr/bin/php /opt/iiab/moodle/admin/cli/cron.php >/dev/null)
Dec  3 14:13:18 box internetarchive[999]: 2021-12-03T18:13:18.711Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Dec  3 14:13:19 box internetarchive[999]: 2021-12-03T18:13:19.003Z dweb-transports:http Error status call to info failed Transport error thrown by https://dweb.me/info: request to https://dweb.me/info failed, reason: Hostname/IP does not match certificate's altnames: Host: dweb.me. is not in the cert's altnames: DNS:*.archive.org, DNS:archive.org
```

Longer excerpt starts on Line 2552 of:
http://sprunge.us/dCbzG5?en

Earlier @jvonau wrote:

> [interactivearchive] is spamming the logs every 30 seconds with an error, that might be an issue on a long running machine.

Originally at:

- https://github.com/iiab/iiab/issues/3057#issuecomment-985928986

mitra42, 2021-12-04:
Looks to me like the certificate no longer has dweb.me in it, though I see it in the internet archive code. 

I'm not really maintaining that code - as far as I was aware noone was using it, and the project wasn't continued, though in theory it should still work. It looks like @jvonau  is just testing, rather than using ...but anyway I'll see if I can get it fixed.

mitra42, 2021-12-04:
Its Saturday in the US, so I suspect it will be a few days.

jvonau, 2021-12-05:
I wasn't testing at all just have noticed other "large" installs have been spammed for awhile. Perhaps @holta might want to tone down the loudspeaker in local_vars.yml, setting the install default to False or comment out internetarchives altogether if work doesn't continue?

mitra42, 2021-12-05:
Sorry - @jvonau  what I meant was that you probably weren't actively using it. I expect I'll hear back from Tracie after the weekend, and hopefully get the certificate fixed (its a fix that needs to happen at archive.org, not in the IIAB distro) . 

If noone is using it, then defaulting to false would be fine. It *should* still be working, and I will fix problems if I can.

holta, 2021-12-05:
Thank You @mitra42 !

mitra42, 2021-12-09:
The certificate has been updated on dweb.me so there should be no more log entries generated. I don't have a way to test this currently (though I have tested that `https://dweb.me/info` works without HTTPS / cert problems, which is what the log is reporting.

holta, 2022-03-24:
Good enough.

Thanks all.

closed 2022-03-24 by holta