iiab/iiab #3060: Change to imager design

georgejhunt, 2021-12-06:
I would like to change the design of imager such that when the Rasperrypi foundation changes their imager, it does not require updating our code.

I'd also like to scrap the "experimental" item on our menu.  I'm thinking that experimental images should be downloaded as an image, and then use the last item "'Use custom" for installation of such.

This proposal requires  only that we mount imagingutility_iiab_subitems.json as images.json on iiab.me.

And of course it means that the iiab-imager will no longer be a superset of the stock imager. Both will be required by our end users

tim-moody, 2021-12-06:
I think I agree with all this, but I don't entirely understand the implication of 'we mount imagingutility_iiab_subitems.json as images.json on iiab.me'. Would the menu on iiab.me be static and reference submenus that would still be in github and vary as we publish new images or would we somehow get the submenu from github to iiab.me? I think the former is preferred.

I had wanted our submenu one day to be one of the rpi imagers sub menus, and I see that the list of included submenus has grown. If and when they agree to include us there would be no need for our submenu to be on iiab.me.

georgejhunt, 2021-12-06:
As Tim points out, the new layout of the foundation's menu tree is much
expanded. I like his idea of integrating within their context better.

holta, 2021-12-06:
Coincidentally I was trying to simplify this in much the same way @georgejhunt mentioned.

I didn't get a chance to post/publish but will later today.

holta, 2021-12-06:
> Coincidentally I was trying to simplify this in much the same way @georgejhunt mentioned.

I should have published this last week in its early draft phase, but here's the latest runnable prototype including the 3 standard-for-now images for [IIAB 7.2 Preview 2](https://github.com/iiab/iiab/releases/tag/7.2-preview-2) just published today;

```
rpi-imager --repo http://iiab.io/fast.json
```

For outreach purposes, to keep newcomers laser-focused-and-on-message, this intentionally includes a much smaller number of choices (just 3 images {DESKTOP, SMALL, MEDICAL}) with the idea being that @avni and @icebox827 are right now this week and in December helping craft highly simplified, no-fuss outreach/marketing/community-building messaging to work with this during IIAB 7.2's full release sometime in coming weeks.

Coincidentally this is almost exactly the same conclusion George reached: an "experimental" folder muddies outreach and simplicity among newcomers especially, and removing it indeed best.  As George says, more advanced folk should use the "Use custom" option at the very bottom of Raspberry Pi Imager's "CHOOSE OS" menu.

Just FYI .json code hereunder is also highly simplified, with the `zip-json-sd` command generating 8-line .json snippets[*] immediately during zipping of the .img, as seen here;

- https://github.com/iiab/iiab-factory/pull/199

[*] Manually cutting + pasting these 8-line .json snippets into a file like http://iiab.io/fast.json works extremely well FYI.  And in fact is likely best going forward on a monthly/quarterly basis, so that outreach/marketing can be arranged artistically (per the outreach needs of that particular quarter, etc).

tim-moody, 2021-12-06:
@holta have you also reserved for yourself the exclusive ability to publish images? Who else is able to modify fast.json on iiab.io?

holta, 2021-12-06:
Please see https://github.com/iiab/iiab-factory/blob/master/box/rpi/fast.json which is the source of http://iiab.io/fast.json = https://iiab.io/fast.json which I'd neglected earlier.  Sorry for that.

It's important, both as a secondary backup, and for community deliberation -> collaboration as @tim-moody outlines above.

Anybody can and should suggest a Pull Request against that file (**fast.json**) when updates are needed quasi-quarterly or whenever, e.g. when a better outreach/marketing/community-building strategy emerges etc.

An immediate example (this is still a prototype) is the currently non-elegant image titles / non-elegant image descriptions that @avni and @icebox827 (or others!) can hopefully make friendlier in coming days, and in any case before IIAB 7.2's final release later this month.

"Outreachy" work takes many, many weeks to make solid, friendly and useful: _"A woman's work is never done?"_ as the old saying goes.  And as politically incorrect as that may be, it remains as true as ever.  Those currently working very hard to make this possible  (@icebox827 and @avni and others) are not unappreciated!

Finally: the [standard-for-now ~3 IIAB images](https://github.com/iiab/iiab/wiki/Raspberry-Pi-Images:-Summary#steps-to-install-an-image-on-a-raspberry-pi) will one day evolve in one direction or another, and might even be auto-generated quasi-monthly in future, as @jvonau is making great progress on these fronts last I checked.  e.g. Especially if GitHub Actions and/or QEMU (or similar) are hopefully capable of hopefully very soon emulating Raspberry Pi in the cloud.

_While keeping in mind the new focus on IIAB image downloads that are Dramatically Faster Than Archive.org (approximately 100X, hence the name **fast.json**) wherever we possibly can~_

closed 2022-03-23 by holta

jvonau, 2022-03-23:
Personally think using `rpi-imager --repo https://raw.githubusercontent.com/iiab/iiab-factory/master/box/rpi/fast.json` might work, although a longer url it would be using the true source of the fast.json file with less web upkeep going forward.