iiab/iiab #3062: ZIM Files that lack menudefs aren't clickable on IIAB home page, and {Size, Articles, Media, Date} are missing

holta, 2021-12-09:
@tim-moody this seems to be a bug?
Weren't these things working in the past?

These ZIM files were downloaded into IIAB using Admin Console.  They appear on the IIAB's home page,
`iiab-make-kiwix-lib` was run for good measure, but it did not resolve the problems.

An example broken link URL &mdash; when hovering over the top ZIM file in the screenshot below:
http://10.8.0.70/js-menu/menu-files/html/undefined.html?menu_item=en-wikipedia_en_top_maxi&zim_name=wikipedia_en_top_maxi
...which leads to a page that says: "Undefined Zim in a Menu Definition"

The {Size, Articles, Media, Date} information [footnote detail](https://github.com/iiab/iiab/wiki/IIAB-Menuing#menu-item-definition-fields) is missing for each of these ZIM files shows:

```
Size: undefined, Articles: NaN, Media: NaN, Date: undefined
```

[image]

FYI here's the menu-def for that ZIM file at the top of screenshot, pasted in from 10.8.0.70:

```
root@box:~# cat /library/www/html/js-menu/menu-files/menu-defs/en-wikipedia_en_top_maxi.json                                                                            {
  "name": "en-wikipedia_en_top_maxi",
  "lang": "en",
  "intended_use": "zim",
  "zim_name": "wikipedia_en_top_maxi",
  "start_url": "",
  "title": "Best of Wikipedia, in English",
  "logo_url": "en-wikipedia.png",
  "description": "50,000 articles from the world's largest encyclopedia, created, moderated, and edited by users for users.",
  "extra_description": "Can be searched using any word(s) that appear in article titles (this is fast!)<p>Or, do a deep search using any keyword(s) that appear further down in the articles themselves (have patience, this takes time!)",
  "extra_html": "",
  "footnote": "Size: ##SIZE##, Articles: ##ARTICLE_COUNT##, Media: ##MEDIA_COUNT##, Date: ##zim_date##",
  "edit_status": "repo",
  "commit_sha": "612cab0c5b6b8a913029f7942c550c479ba8edc6",
  "change_ref": "copy from repo",
  "change_date": "2021-12-01",
  "upload_flag": false,
  "download_flag": true
}
```

[iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) are: http://sprunge.us/wwvQBv?en

tim-moody, 2021-12-09:
if 10.8.0.70 has the menu def why doesn't the machine under test have it? it seems to be a stock menu def and not in need of generation.

still, if absent it should have been generated. sounds like zim_version_idx.json is absent.

anything in the log? journalctl  | grep CMDSRV

what happens if you click Update Home Menu on the Content Menus submenu? not sure if it generates menu defs or just puts them on the home page.

holta, 2021-12-09:
Confoundingly, 1 of the 4 menudefs on the main page has suddenly come to life (is now fully working, with valid URL http://10.8.0.70/kiwix/wiktionary_en_simple_all_maxi_2021-12/ and all 4 footnote details appearing correctly) as seen in this new screen shot:

[image]

(Mysteriously, the other 3 are still not working.)

holta, 2021-12-09:
> still, if absent it should have been generated. sounds like zim_version_idx.json is absent.

```
root@box:~# locate zim_version_idx.json
/library/www/html/common/assets/zim_version_idx.json
```

> anything in the log? journalctl | grep CMDSRV

```
root@box:~# journalctl | grep CMDSRV | pastebinit -b sprunge.us
http://sprunge.us/fIlsBy
```

> what happens if you click Update Home Menu on the Content Menus submenu? not sure if it generates menu defs or just puts them on the home page.

Working with @icebox827, we had similarly tried:

```
root@box:~# iiab-update-menus
Updating Home Page Menu
Updating kiwix menu items
Updating iiab installed services' menu items
Updating installed OER2Go menu items
```

And just to be sure, we also tried http://10.8.0.70/admin **> Content Menus > Update Home Menu** and it  responded:

```
10.8.0.70 says

Home Page Menu Updated.
```

_Unfortunately, IIAB home page http://10.8.0.70 appears unchanged, as in the most recent screen shot above (:_

(@tim-moody do you want to ssh in to 10.8.0.70 to dig a bit more?)

tim-moody, 2021-12-09:
I notice that the automatically update menus flag is not checked. Not sure what else happened.

holta, 2021-12-09:
> I notice that the automatically update menus flag is not checked.

Is that normal?

FYI this was a vanilla LARGE-sized IIAB install 8 days ago (Wedn Dec 1) onto @icebox827's Panasonic Toughbook CF-52:

- #3057

holta, 2021-12-09:
> if 10.8.0.70 has the menu def why doesn't the machine under test have it? it seems to be a stock menu def and not in need of generation.

Possibly this ticket is titled poorly.  If this is not caused by missing menu-defs.

@tim-moody and @icebox827 if you make progress in understanding cause/effect, let's please re-title this issue!

holta, 2021-12-09:
@icebox827 now that you've reverted to the non-nightly kiwix-serve, can you now rerun `iiab-make-kiwix-lib` and `iiab-update-menus` and see if that helps?

icebox827, 2021-12-09:
@tim-moody @holta 

```
root@box:~# iiab-make-kiwix-lib
Waiting to run iiab-make-kiwix-lib.py
Now running iiab-make-kiwix-lib.py
Writing zim_versions_idx
Finished making Kiwix library.xml
````
```
root@box:~# iiab-update-menus
Updating Home Page Menu
Updating kiwix menu items
Updating iiab installed services' menu items
Updating installed OER2Go menu items
```

[image]

It's interesting, even after running theses commands I still have some broken links

holta, 2021-12-09:
Thanks @icebox827 who also identified this additional broken link:

[image]

tim-moody, 2021-12-09:
none of those items is installed

holta, 2021-12-09:
@tim-moody could any idea if these [1+3 broken links](https://github.com/iiab/iiab/issues/3062#issuecomment-990280455) have arrived onto the IIAB main page (http://10.8.0.70) from a [Preset / Quick Pick](https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/InstContent.rst#quick-pick) possibly?

FYI I looked through all 139 jobs at http://10.8.0.70/admin/ > Utilities > Display Job Status and I don't see any of the 4 missing items, but the 4th one at the bottom may provide some hints ??

- https://rachel.worldpossible.org/viewmod/en-blockly-games (< 0.1 GB)
- http://download.kiwix.org/zim/wikipedia/wikipedia_en_top_maxi_2021-12.zim (5.9G, 2021-12-05)
- http://download.kiwix.org/zim/wikipedia/wikipedia_en_simple_all_maxi_2021-12.zim (1.9G, 2021-12-05)
- " https://s3.us-east-2.wasabisys.com/iiab-zims/crashcourse_en_top_2021-01.zip " as listed in https://github.com/iiab-share/iiab-content/blob/main/catalogs/iiab-zim-cat.json#L106-L123 (9.3G, 2021-01-08)

tim-moody, 2021-12-09:
>could any idea if these 1+3 broken links have arrived onto the IIAB main page (http://10.8.0.70) from a Preset / Quick Pick possibly?

yes

>" https://s3.us-east-2.wasabisys.com/iiab-zims/crashcourse_en_top_2021-01.zip " as listed in https://github.com/iiab-share/iiab-content/blob/main/catalogs/iiab-zim-cat.json#L106-L123 (9.3G, 2021-01-08)

catalog says zim; you say zip

tim-moody, 2021-12-09:
impossible to reconstruct what has gone on with this machine. I see zim tools in the middle of /library/zim/content.

You might try journaling your testing so you can reconstruct the steps you took and what resulted.

holta, 2021-12-10:
@tim-moody: Denis just explained to me that (almost a week ago) he had clicked on **Install Content > selected School Server in Spanish > Install Selected Collection**.

Does that help a bit...to explain how some-or-all of these 1+3 links appeared?

tim-moody, 2021-12-10:
Those links are all in that preset (collection). Doesn't say why the content is not there.

icebox827, 2021-12-12:
@holta  @tim-moody 

1) I wiped the Toughbook and did a fresh install of Ubuntu Desktop 21.10

2) I installed LARGE-sized IIAB.

3) I clicked Admin Console > Install Content > Quick Pick > School Server in Spanish > Install Selected Collection.

4) I waited many hours until it completed, as confirmed by Admin Console > Utilities > Display Job Status.

5) The Blockly Games link remains broken, on my IIAB home page http://10.8.0.14

[image]

6)  [**iiab-diagnostics**](http://sprunge.us/lKs7vl?en)

Can you help me investigate why this keeps failing?

tim-moody, 2021-12-12:
I can't find an es-blocky-games in the OER2Go catalog, only en and fr. So this is a bug in the preset.

There may have been a popup that went by quickly announcing this, but not sure how that would work during presets that stuff many jobs on the queue.

tim-moody, 2021-12-12:
Fixed in https://github.com/iiab/iiab-admin-console/pull/440. Thanks for persisting in finding it.

closed 2021-12-26 by holta