iiab/iiab #3064: [OSM] Failed to install/reinstall OpenStreetMap on iiab 7.2 version via ./runrole

boholder, 2021-12-11:
### Expected behavior and actual behavior:

After installed iiab (the "large(? that allows OSM to be installed)" install option), go to `box.lan/admin` -> `Install Content` -> `Get Map Regions` or `Get Map Addons`, there are only "MAPS NOT INSTALLED OR NOT ENABLED" text and no map select check box on the left side.

Then I found `Configuration` -> `Services Enabled` has an enable config for OSM, so I checked it on and restart the box via admin panel.

But still I can't access OSM after reboot, so I assumed that the OSM installation is failed. I searched the method that reinstall OSM, and found issue #2370 : `./runrole --reinstall osm-vector-maps`.

Running  `./runrole --reinstall osm-vector-maps` will output 

```
ERROR: /etc/iiab/iiab_state.yml has no line(s) that begin with 'osm_vector_maps_'
Try again without the '--reinstall' flag?
```

Running `./runrole osm-vector-maps` will output

[image]

Now... maybe my real question is: **is there any other way to reinstall the OSM alone?** 

BTW, my network isn't stable and run `./iiab-install --reinstall` will output some network-related errors (see [here](https://github.com/iiab/iiab/issues/1501#issuecomment-927121603)),  so another trickier possibility is that some files is crushed or even not exist in right position.

Big thank to hardworking maintainers.

### IIAB version, OS version, hardware details:

IIAB: 7.2, Debian 11 amd64 (bullseye) (in virtualbox), host is Windows10

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

https://pastebin.com/Yqf4PxAL

holta, 2021-12-11:
Thanks @boholder for the report!

@georgejhunt do you have time to look into this today or in coming days possibly?

holta, 2021-12-11:
1) @boholder can you please repost the output of [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) to a new pastebin URL?

   The link you have provided (https://pastebin.com/Yqf4PxAL) does not work.

2) @boholder the `./runrole` command should always be run from the directory `/opt/iiab/iiab`

   (Your screenshot makes it appear as if you're running `./runrole` from an incorrect directory.)

boholder, 2021-12-11:
If you need me to provide more informations, feel free to mention me. The outputs of the OSM installation script shows that maybe I haven't installed the OSM correctly so I don't know why it is a potential admin console issue.

boholder, 2021-12-11:
> 1) @boholder can you please repost the output of [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) to a new pastebin URL?
> 
> 
> 
>    The link you have provided (https://pastebin.com/Yqf4PxAL) does not work.

On the way.

holta, 2021-12-11:
@boholder as a side comment (and general FYI for everyone!) please use the **new** http://FAQ.IIAB.IO going forward.

It is now hosted here:
https://wiki.iiab.io/go/FAQ

boholder, 2021-12-11:
> 1. @boholder can you please repost the output of [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) to a new pastebin URL?

http://sprunge.us/Uvv2G9

>     2. @boholder the `./runrole` command should always be run from the directory `/opt/iiab/iiab`
>        (Your screenshot makes it appear as if you're running `./runrole` from an incorrect directory.)

Rerun in  `/opt/iiab/iiab` works: `./runrole --reinstall osm-vector-maps`  output same message, but without `--reinstall` flag, it pops up a proxy error (urlopen error [Errno 113] No route to host), which shows that maybe I should not bother you further and keep fighting with my proxy settings.

holta, 2021-12-11:
Thanks @boholder for http://sprunge.us/Uvv2G9?en

Please also consider a completely fresh install of the latest pre-release of IIAB 7.2 (as your original install appears to have been about 3 months ago, in September) if you can?

_Thanks for reporting back what you discover!_

closed 2021-12-11 by boholder

boholder, 2021-12-11:
The OSM works well after performing installation of pre-release of IIAB 7.2 (via one-line install script). 😄

holta, 2021-12-11:
> The OSM works well after performing installation of pre-release of IIAB 7.2 (via one-line install script). smile

Thanks @boholder for your reports!

PS do you happen to have a Raspberry Pi available?

If so, you can also try a much faster way to install IIAB &mdash; _that also works well for those wanting to avoid the Command-Line:_ https://github.com/iiab/iiab/wiki/Raspberry-Pi-Images:-Summary

boholder, 2021-12-12:
[Mini Tutorial: How to manually install region maps]

When admin console couldn't directly download tile files (due to network issues) and reports an error but you can download the files in a different way. For me who using virtual machine to run iiab, my  iiab guest os can't access network due to wrong proxy setting, but I can download files in host os and copy them onto iiab guest os.

Do what admin console wants to do when you schedule a region map download job:

1. Download map files from [archive.org](https://archive.org/search.php?query=osm-vector&sort=-publicdate), copy them to `/library/working/maps`.
2. Run two installation scripts:

```
cd /opt/iiab/iiab-admin-console/roles/cmdsrv/files
# move files to www directory and delete old ones
bash scripts/osm-vect_v2_install_step2.sh <map_file_name (e.g. osm_xxx.mbtiles)>
# config map view?  
# "!:$" is a bash trick, for repeating last argument of last command -- osm_xxx.mbtiles
python3 scripts/osm-vect_v2_finish_install.py !:$
```

3. If everything is ok, the python script will print currently installed maps.

tim-moody, 2021-12-12:
>For me who using virtual machine to run iiab, my iiab guest os can't access the Internet due to wrong proxy setting, but I can download files in host os and copy them onto iiab guest os

how were you able to install iiab without an internet connection?

also we don't usually run scripts out of the ansible repo. they could contain variables that only get substituted during installation

boholder, 2021-12-13:
> how were you able to install iiab without an internet connection?

After installation I changed virtual machine software's virtual network settings for letting my host can browser iiab's website (doesn't install GUI desktop on iiab), and I messed up something so previous tool-specific proxy settings on iiab are invalid.

Well... the behavior of the admin console when I download region maps is pretty confusing: 

At first I didn't correctly configure wget's proxy (wrong proxy server port), so it couldn't connect to the site where the map data was downloaded and set the download  job statu as `FAIL`, two after-installation jobs as `CANCELED`, which can be seen in admin console `Utilities` -> `Job status(?)` menu.

Then I configured wget's proxy (`/etc/.wgetrc`), rebooted the iiab, tested it (by using wget to download a online text file),  re-clicked map download option in admin console, but still the console poped up same error, and there is no new job status about this operation in the `Job status(?)` menu. Retried clicking three times, three pop up, three log in `Utilities` -> `??? Log` (log is same content as pop up window), but no job status.

> also we don't usually run scripts out of the ansible repo. they could contain variables that only get substituted during installation

Therefore... I started trying to install maps manuall. This is the operation *after installation*, so I guessed I can handle it by hand, all I needed is to know in which directory I should run the two scripts, so I viewed their contents and thought I could run them at `/opt/iiab/iiab-admin-console/roles/cmdsrv/files`, which is logged in job status running-output column.

Sincere apologies for disrespectfully fiddling with your hrad work. 💦 Need I delete that comment?

boholder, 2021-12-14:
> My concern is that your tutorial contains misinformation based on reverse engineering and an assumption that Admin Console runs scripts from /opt/iiab/iiab-admin-console/roles/cmdsrv/files. It doesn't. These are also not published scripts so they could get changed at any time. So while this worked for you it might not work for others.

You are right, I broke the encapsulation. I've hidden that comment. Thank you for pointing out that.

holta, 2021-12-14:
@boholder here's an IIAB reference doc that I'll try to update (with others' help) in coming weeks:

https://github.com/iiab/iiab/wiki/IIAB-Maps

It's about a year-old so not everything is gospel truth as the sands if time pass — but just FYI, as yes looking under the hood is sometimes necessary — e.g. if this general topic of helping communities customize the offline OpenStreetMap "packs" they most need...is of interest to you!