iiab/iiab #3065: Admin Console -> Quick Pick -> EN School leads to 1 + 4 broken links

icebox827, 2021-12-13:
@tim-moody @avni 

After doing these steps on a freshly installed IIAB on a Raspberry Pi 400 : http://10.8.0.30/admin -> Quick Pick -> Install Content -> English speaking School 128G Sampler, @avni and I identified 1 + 4 broken links: 

All the 17 jobs show status SUCCEEDED in http://10.8.0.30/admin -> Utilities -> Display Jobs Status
[image]

1. Clicking on **PhET module** leads to **404 Not Found**
[image]

2. Clicking on **Wikispecies, Raspberry Pi Stack Exchange, Chemistry Stack Exchange, and Wiktionary in Simple English** leads to **Undefined Zim in a Menu Definition**
[image]

Can you help me understand where these links are breaking?

[**iiab-diagnostics:**](http://sprunge.us/X4dYBG?en)

holta, 2021-12-13:
Thanks @icebox827 & @avni for your very hard work over the weekend here!

tim-moody, 2021-12-13:
I can confirm that the preset has the wrong names for each of the items you mentioned. Do you mind if I use your install to test the changes?

icebox827, 2021-12-13:
@tim-moody 

Sure! It is a great idea.

tim-moody, 2021-12-13:
@holta should we change from the html PhET to zim?

holta, 2021-12-13:
> @holta should we change from the html PhET to zim?

Definitely a good idea.  Yes please do that if possible.

tim-moody, 2021-12-13:
Having trouble sorting this out. The zim names are wrong as are the menu defs. Not sure if Kiwix changed the names or I just got it wrong. Thought each of the original presets were tested.

holta, 2021-12-13:
@icebox827 please see:

- PR iiab/iiab-admin-console#441

tim-moody, 2021-12-14:
@icebox827 I have pulled iiab-admin-console#441, installed it and rerun installing the preset. Please verify.

tim-moody, 2021-12-14:
#3066 is irrelevant to this issue as it deals with a networking problem, not a preset configuration problem.

icebox827, 2021-12-14:
> @icebox827 I have pulled iiab-admin-console#441, installed it and rerun installing the preset. Please verify.

Thanks @tim-moody , I will verify

closed 2021-12-26 by holta