iiab/iiab #3068: Admin Console -> Quick Pick -> FR School leads to 1 + 1 broken links

icebox827, 2021-12-14:
@tim-moody

I installed a fresh copy of Ubuntu 21.10 Desktop on a Panasonic Toughbook CF52 and freshly installed IIAB : http://10.8.0.14/admin -> Quick Pick -> Install Content -> French speaking School Sampler, and I identified 1 + 1 broken links:

1. Clicking on **Khan Academy Lite** leads to **This site can’t be reached**
[image]

2. Clicking on **Wiki livre** leads to **Undefined Zim in a Menu Definition**
[image]

All the 43 jobs show status SUCCEEDED in http://10.8.0.14/admin -> Utilities -> Display Jobs Status
[image]

And the dropdown search on Kiwix  doens't not work properly
[image]

Please help me figure out why these links are broken.

[iiab-diagnostics](http://sprunge.us/gRpunT?en)

holta, 2021-12-14:
> dropdown search on Kiwix doens't not work properly

I can't reproduce this at http://10.8.0.14/kiwix/wikivoyage_fr_all_maxi_2021-11/ which redirects to http://10.8.0.14/kiwix/wikivoyage_fr_all_maxi_2021-11/A/Accueil

@icebox827 have you cleared your browser cache since changing versions of kiwix-tools?  Thanks for clarifying.  Background here:

- kiwix/kiwix-tools#505

holta, 2021-12-14:
Thanks @icebox827 for this report, which is a huge help (like your others!!)

As we together (1) Sort out this Quick Pick [fr-school](https://github.com/iiab/iiab-admin-console/tree/master/roles/cmdsrv/files/presets/fr-school)'s particulars and also (2) Identify **community QA workflows** that will nurture the best "Learning Bouquets" needed by any community worldwide... _willing to step up to the plate and help others build it out!_

(1) In this specific case, the 2 non-working links are:

- http://10.8.0.14:8006/learn/khan/math/
- http://10.8.0.14/js-menu/menu-files/html/undefined.html?menu_item=fr-wikibooks_fr_all_maxi&zim_name=wikibooks_fr_all_maxi

(2) No clear answers just yet, but a few quick/initial ideas here: https://github.com/iiab/iiab/issues/3065#issuecomment-993634420

Related:

- #3062 
- #3065 
- #3067

icebox827, 2021-12-14:
> > dropdown search on Kiwix doens't not work properly
> 
> I can't reproduce this at http://10.8.0.14/kiwix/wikivoyage_fr_all_maxi_2021-11/ which redirects to http://10.8.0.14/kiwix/wikivoyage_fr_all_maxi_2021-11/A/Accueil
> 
> @icebox827 have you cleared your browser cache since changing versions of kiwix-tools? Thanks for clarifying. Background here:
> 
> * [Search regressions in nightly builds of kiwix-serve: (1) punctuation floods the UI (2) Title Search goes to the wrong URL [mitigated by clearing browser cache] kiwix/kiwix-tools#505](https://github.com/kiwix/kiwix-tools/issues/505)

You are right, I need to clear the browser cache after changing the kiwix-tools versions, it works fine now. Thank you!

tim-moody, 2021-12-15:
The dropdown is a separate matter. 

>http://10.8.0.14/js-menu/menu-files/html/undefined.html?menu_item=fr-wikibooks_fr_all_maxi&zim_name=wikibooks_fr_all_maxi

kiwix added maxi to some names and the preset doesn't reflect that.

The Khan link is a bigger problem and one that @jvonau raised in another context.

When we have multiple languages of kalite installed we put them on different ports, though this is not exactly how Learning Equality intended it to be done. When we have a single language, even non-English I think we put kalite on port 8008, same as EN. The es-kalite menu def reflects this in that it has a name that suggests ES, but the declared lang is EN, which triggers port 8008. The FR menu def fr-kalite has lang FR, which triggers port 8006. But kalite was installed on 8008 so the link is broken.

I'm still inclined to use port 8008 for any single language, but the menu defs need adjusting and we need a naming convention,

Maybe en-kalite means 8008 in EN, fr-kalite is 8008 in FR, and es-kalite is 8008 in ES. fr-kalite_fr would be FR on 8006 and es-kalite_es would be ES on 8007. Or maybe clearer en-kalite_en, fr-kalite_en, es-kalite_en are all 8008.

icebox827, 2021-12-15:
@tim-moody 

The dropdown has been solved, it was a minor issue. It requires the user to clear the browser cache.
Thanks for theses valuables and high quality feedback, it helps me have a better understanding of the Khan link problem.

tim-moody, 2021-12-15:
The alternative is to specify kalite port with kalite_server_port in the preset's vars.yml file.

tim-moody, 2021-12-15:
I will expect one instance and port 8008 until kalite role(s) change

holta, 2021-12-15:
FYI

- PR https://github.com/iiab/iiab-admin-console/pull/443

tim-moody, 2021-12-15:
in addition to pulling and installing adm cons it will be necessary to click Sync Menu defs

icebox827, 2021-12-15:
@tim-moody 

I did the following:
1. `sudo su -`
2. `cd /opt/iiab/iiab-admin-console`
3. `git pull`
4. `./install`

This message is been displayed after doing as suggested: http://10.8.0.14/admin > Content Menus > Sync Menu Defs.
[image]

tim-moody, 2021-12-15:
I'm a little surprised, but I guess the install already did the sync. Can you verify that /library/www/html/js-menu/menu-files/menu-defs/fr-kalite_en.json exists.

holta, 2021-12-15:
> Can you verify that /library/www/html/js-menu/menu-files/menu-defs/fr-kalite_en.json exists.

I logged into @icebox827's Toughbook (10.8.0.14) and confirmed:

```
root@box:~# ls -l /library/www/html/js-menu/menu-files/menu-defs/fr-kalite_en.json
-rw-r--r-- 1 root root 685 dic 15 16:45 /library/www/html/js-menu/menu-files/menu-defs/fr-kalite_en.json
```

```
root@box:~# cat /library/www/html/js-menu/menu-files/menu-defs/fr-kalite_en.json
{
  "name": "fr-kalite_en",
  "lang": "en",
  "intended_use": "kalite",
  "start_url": "learn/khan/math/",
  "title": "Khan Academy Lite",
  "logo_url": "khan-logo-vertical-transparent.png",
  "description": "KA Lite est la version hors ligne de Khan Academy qui comprend des milliers de vidéos d'apprentissage et de quiz. Il est très populaire dans les lycées et comprend également quelques cours universitaires.",
  "extra_description": "",
  "extra_html": "",
  "footnote": "",
  "edit_status": "repo",
  "commit_sha": "96ca8eeb8fbc4ebd1d18f9426ad902d1408c41cc",
  "change_ref": "copy from repo",
  "change_date": "2021-12-15",
  "upload_flag": false,
  "download_flag": true
}
```

FYI for those interested this originates from:
https://github.com/iiab-share/js-menu-files/blob/master/menu-defs/fr-kalite_en.json

Which is almost identical to:
https://github.com/iiab-share/js-menu-files/blob/master/menu-defs/fr-kalite.json

tim-moody, 2021-12-15:
perhaps someone could rerun the preset now.

holta, 2021-12-15:
> perhaps someone could rerun the preset now.

@icebox827: I just now clicked http://10.8.0.14/admin **> Quick Pick > French Speaking School > Install Selected Collection** on @tim-moody's suggestion.

Let's take a look at how long it takes to complete!

(It began at 6:35PM NYC Time.)

holta, 2021-12-16:
Good News: both links now work on @icebox827's http://10.8.0.14

- http://10.8.0.14:8008/learn/khan/math/
- http://10.8.0.14/kiwix/wikibooks_fr_all_maxi_2021-11/ which redirects to...
  http://10.8.0.14/kiwix/wikibooks_fr_all_maxi_2021-11/A/Accueil

Even Better News: all 5 Admin Console jobs completed in 10 minutes, as confirmed by the screenshot below. (that's instead of about 4.5 hours taken by yesterday's 43 jobs, that installed the bulk of the content making up [presets/fr-school](https://github.com/iiab/iiab-admin-console/tree/master/roles/cmdsrv/files/presets/fr-school))

[image]

holta, 2021-12-16:
For the heck of it, I clicked **Quick Pick > French Language School** a 2nd time, which spawned 3 jobs, all of which completed in 7min13sec.

Just FYI the output (from the **Result** field of the final job [# 108] within http://10.8.0.14/admin **> Utilities > Display Job Status**) kept deleting itself (as it was running, very odd).

But here are 6 Copy/Pastes showing the bulk of that output...arising from command `scripts/kalite_install_videos.py fr khan/` just for the record:

```
--2021-12-15 20:06:55-- http://s3.us-east-2.wasabisys.com/ka-lite-0.17-resized-videos/fr/qLQIqZdNlww.png
Resolving s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)... 38.73.225.19, 38.73.225.20, 38.73.225.22
Connecting to s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)|38.73.225.19|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-12-15 20:06:56 ERROR 404: Not Found.

--2021-12-15 20:06:56-- http://s3.us-east-2.wasabisys.com/ka-lite-0.17-resized-videos/fr/cX9LC6NDNGg.png
Resolving s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)... 38.73.225.19, 38.73.225.20, 38.73.225.22

[...]
	
--2021-12-15 20:08:13-- http://s3.us-east-2.wasabisys.com/ka-lite-0.17-resized-videos/fr/YsNugQGQHHI.png
Resolving s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)... 38.73.225.20, 38.73.225.22, 38.73.225.19
Connecting to s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)|38.73.225.20|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-12-15 20:08:13 ERROR 404: Not Found.

--2021-12-15 20:08:13-- http://s3.us-east-2.wasabisys.com/ka-lite-0.17-resized-videos/fr/8dnugjhwZTA.png
Resolving s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)... 38.73.225.20, 38.73.225.19, 38.73.225.22
Connecting to s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)|38.73.225.20|:80... connected.

[...]
	
--2021-12-15 20:09:13-- http://s3.us-east-2.wasabisys.com/ka-lite-0.17-resized-videos/fr/dI3Fey_e9vE.png
Resolving s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)... 38.73.225.22, 38.73.225.20, 38.73.225.19
Connecting to s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)|38.73.225.22|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-12-15 20:09:13 ERROR 404: Not Found.

--2021-12-15 20:09:13-- http://s3.us-east-2.wasabisys.com/ka-lite-0.17-resized-videos/fr/2mqPzgHyYBM.png
Resolving s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)... 38.73.225.19, 38.73.225.22, 38.73.225.20
Connecting to s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)|38.73.225.19|:80... connected.

[...]

2021-12-15 20:09:40 ERROR 404: Not Found.

--2021-12-15 20:09:40-- http://s3.us-east-2.wasabisys.com/ka-lite-0.17-resized-videos/fr/aARnDeibUqw.png
Resolving s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)... 38.73.225.22, 38.73.225.20, 38.73.225.19
Connecting to s3.us-east-2.wasabisys.com (s3.us-east-2.wasabisys.com)|38.73.225.22|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-12-15 20:09:40 ERROR 404: Not Found.

--2021-12-15 20:09:40-- http://s3.us-east-2.wasabisys.com/ka-lite-0.17-resized-videos/fr/tfxCBZFbfv4.png

[...]

Error downloading yUuJr40GCCs.png
Skipping yUuJr40GCCs.png
Error downloading FxasOvtghO8.png
Skipping FxasOvtghO8.png
Error downloading EB9d5pJv01Q.png
Skipping EB9d5pJv01Q.png
Error downloading mNjUd5xeqIY.png
Skipping mNjUd5xeqIY.png
The following were not downloaded:

Skipping FxasOvtghO8.png
Error downloading EB9d5pJv01Q.png
Skipping EB9d5pJv01Q.png
Error downloading mNjUd5xeqIY.png
Skipping mNjUd5xeqIY.png

[...]

The following were not downloaded:
['hMlnp_LjDbM.png', 'kujXvvdVMx0.png', 'X-j_dhVj88w.png', 'P0WBQtUzfhA.png', 'p8A7T9gFEJ4.png', 'zTWDlgdGuuc.png', '579w41BCPzo.png', 'gesE29mT2qI.png', 'J2TNj1nP22A.png', 'CIHpE6SwQWk.mp4', 'CIHpE6SwQWk.png', 'xuYU9RUTYLw.png', 'TXUHA2E02ks.png', '43ZRhApac4E.png', 'PFnxD8uN2ko.png', 'aqghioN_oWM.png', 'H7aSeuJJEwE.png', 'cqA1Rqrdw1Q.png', 'Zi5iv1PlTmA.png', 'fqYiHRw65-8.png', 'UAiCoS5m_eM.png', '4lxdYp6Am2Y.mp4', '4lxdYp6Am2Y.png', 'DaNDYqCf5n4.png', 'E7eRJTgxVz4.png', 'qGty5I6Q2j4.mp4', 'qGty5I6Q2j4.png', 'RniMln1mrn0.mp4', 'RniMln1mrn0.png', 'C1kd9nn-nwk.png', 'JFzmFniBK2k.mp4', 'JFzmFniBK2k.png', 'Hdu1zemn1Gw.png', 'OAyy44yv6-Q.png', 'oAhwOli8mrk.png', 'Ed1bep8r-Do.png', 'r_JmKC5RvsU.png', 'N4FaKU22mXQ.png', 'IJcI7iGRmIE.mp4', 'IJcI7iGRmIE.png', 'kN_0ed10CKI.png', 'a17x0qgymaw.png', 'AVV-Q3K6zPY.png', 'hMjQt-_-mnM.png', 'IaJU8yp_BzM.png', '2Nhb_0Qd3eQ.png', 'AQvBI6xktCc.png', 'z-dBVnKQL9s.png', 'S4XOMUfa5_M.png', 'DXlMqOF5oJA.png', 'o5ibEvbxBdw.png', 'fB5DsXPkvfc.png', 'a6LVJgHKBso.png', 'G8rAQfVYIKs.png', 'd9VtA6-cWSo.png', '0H9hP7EPB6Q.png', 'mDX2LONqwvA.png', 'QwxAaYqEfzA.png', 'bQTA12LrWok.png', 'VJuprEsWbAM.png', 'kJrVF1z_yeg.png', 'yPZhlbJo66s.png', 'r5ETVf71bPk.png', 'm52nSdMrf2c.png', 'H3zA-HLoyso.png', 'sOtn2O00b3w.png', '6QLCiOx6Iok.mp4', '6QLCiOx6Iok.png', 'cOogKzqYYfU.mp4', 'cOogKzqYYfU.png', 'iYzN094wRlE.png', 'c4JzGNPd8Yg.png', 'ghRkhlgAE3s.png', 'cIO0TswzQxA.png', 'dUQWvL-RssM.png', 'qki2EO2hd7U.png', 'BQTbjkx67wk.png', '4OU72U-sUwg.png', 'gRu8SGCgank.png', '0HWe9mfXMZ4.png', 'JXwlLcj8Hmo.png', 'Y7PFpdXvyq4.png', '8wx96etjINo.png', 'Zo4uW-jitIw.png', 'aDKFTmgPuTw.png', 'GpQGyD26U9Q.png', '4kvYHO7e5Do.png', '8p2_08qH3xQ.png', 'gZqaonnqkTQ.png', 'v3VMyfQAcv0.png', '-QL5EabZvEM.png', 'BJcYKnf1a_k.png', 'JzokCZQXCvc.png', 'Q78AnlvrqWk.png', 'lPozGe_hmAQ.png', '5oAzmiXGJZU.png', 'HCV0s-iFTVw.png', 'NQYwfKUCz8E.png', '3hBQSOBjqE [... cut ...] .png', 'W2PcnbFc990.png', 'Wmoj5P_17LI.png', 'RNrN_286S0o.png', 'hCQef-LxFTw.png', 'nWzWC-oySu4.png', 'QRd6iNB8x7c.png', 'ZgFXL6SEUiI.png', 'C_iuFl07U8Y.png', '1HwNS_1kPjE.png', 't7eWWjseJxY.png', 'akORpFsqLIE.png', '8VKIabQ0Bic.png', 'vPTMTM_rRww.png', 's6a0G_U_Vmg.png', 'u1qn5r7eks0.png', 'kIEqCAaY5zg.png', 'yKuO9v5l9a0.png', 'j_dWbEGVzEA.png', 'Oim0OH0J7bo.png', 'ngyZsVQoFBg.png', 'yUuJr40GCCs.png', 'FxasOvtghO8.png', 'EB9d5pJv01Q.png', 'mNjUd5xeqIY.png']
Starting videoscan
Videoscan complete

closed 2021-12-26 by holta