iiab/iiab #3069: For Discussion: Nextcloud 'news' is 64-bit only?

jvonau, 2021-12-15:
https://github.com/nextcloud/server/issues/30169
https://github.com/nextcloud/server/issues/28926
https://github.com/nextcloud/nextcloudpi/issues/1318
https://github.com/nextcloud/news/issues/1320

jvonau, 2021-12-15:
The user base on 32bit RaspiOS might run into [this](https://github.com/nextcloud/news/issues/1423) at some point post-install in the field.

jvonau, 2021-12-15:
https://github.com/nextcloud/server/issues/27797

holta, 2022-05-08:
Can we close this ticket?

Especially now that 64-bit Raspberry Pi OS has proved very solid[*] over the past 3-4 months.

[*] And "48% faster" according to extensive tests run by Phoronix.

jvonau, 2022-05-09:
Not that we install any of the plugins, but there should be a warning to not enable the 'news' plugin post install. Have you documented nextcloud's 'news' issue on 32 bit RaspiOS somewhere? If you have the warning in place feel free to close this issue.

closed 2022-05-09 by holta