iiab/iiab #3071: DNS Errors with box/wordpress (box’s DNS  could not be found. Diagnosing the problem.) and box/moodle (DNS_PROBE_FINISHED_NXDOMAIN).  Both  fail to connect.

shanti-bhardwa, 2021-12-17:
I have IIAB 7.2 Installed on Linux Mint 20.2 on a DELL laptop. It is wired with Ethernet to a TP-Link Router  configured as Local Access Point  with DHCP Server . The router is NOT connected to the Internet. It provides the local LAN. 

Wordpress and Moodle are installed on IIAB 

The router provides IIAB server the IP Address  192.168.0.101.
I have another Client machine. Connected to the same router and it has the address 192.168.0.100

From this Client machine I am using Chrome as my browser.

I get the home page with 192.168.0.101/home and then clicking on Wordpress Icon the url is changed to box/wordpress and the connection fails with these errors.

Chrome gives this error.
This site can’t be reached,
box’s DNS  could not be found. Diagnosing the problem.
DNS_PROBE_STARTED
No Internet
Then
No Internet
No Internet
Try:
    • Checking the network cables, modem and router
    • Reconnecting to Wi-Fi
DNS_PROBE_FINISHED_NO_INTERNET

192.169.1.99/moodle  turns into http://box.lan/moodle
and  the error is 

This site can’t be reached
Check if there is a typo in box.lan.
DNS_PROBE_FINISHED_NXDOMAIN
The following entries work
192.168.0.101/home
192.168.0.101/admin
192.168.0.101/kolibri
192.168.0.101/kiwix
192.168.0.101/usb
192.168.0.101/modules

shanti-bhardwa, 2021-12-17:
We are using Wordpress in Kenya and the the servers are connected to local router as the wifi-access point, so this needs to work.

tim-moody, 2021-12-17:
this is a feature of wp that it expects to know its host or domain name and return urls with that name.

can  you turn  on dns  in the router so that it will resolve box?

tim-moody, 2021-12-17:
alternatively if iiab has a fixed ip, you may be able to point wp to it

tim-moody, 2021-12-17:
https://wpengine.com/resources/change-domains-wordpress/

jvonau, 2021-12-17:
https://wiki.laptop.org/go/IIAB/FAQ#WordPress_.26_Moodle_Administration:_What_tips_.26_tricks_exist.3F

holta, 2021-12-17:
> https://wiki.laptop.org/go/IIAB/FAQ#WordPress_.26_Moodle_Administration:_What_tips_.26_tricks_exist.3F

Please use IIAB's new http://FAQ.IIAB.IO if possible!  i.e. see item number 36 of 48 here:

https://wiki.iiab.io/go/FAQ#WordPress_&_Moodle_Administration:_What_tips_&_tricks_exist?

jvonau, 2021-12-17:
Since you have control over the the DNS at the router level what have you configured the router to hand out for the clients to use? Try `cat /etc/resolv.conf` from the iiab box and the chrome client machine and attach the results to this issue. How far along are you, as in this is just a test box or you have a ton of content installed? These 2 programs install and run best when in they are in their final deployed environment not juggling around across the different subnet's. I can get this to work but I would have to walk you through step by step from a fresh install with the install passing though the router that you are going to deploy with becoming a matched set. Maybe turn the experience into a how-to along the way?

jvonau, 2021-12-17:
Some screen shots of the router's config would be helpful also, and a description of what you expect to type/appear in the address bar of the browser.

jvonau, 2021-12-18:
I recall fiddling with trusted_domains in https://github.com/iiab/iiab/blame/master/roles/nextcloud/tasks/setup.yml#L90-L93 with upstream nextcloudpi being explicit in the definitions used:  
```
pi@nextcloudpi:~ $ sudo ncc config:list
{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "7": "nextcloudpi",
            "5": "nextcloudpi.local",
            "8": "nextcloudpi.lan",
            "3": "nextcloudpi",
            "11": "public ip on internet",
            "1": "private ip on local lan",
            "14": "nextcloudpi"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "22.2.2.0",
        "overwrite.cli.url": "https:\/\/nextcloudpi\/",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
        },
        "tempdirectory": "\/var\/www\/nextcloud\/data\/tmp",
        "mail_smtpmode": "sendmail",
        "mail_smtpauthtype": "LOGIN",
        "mail_from_address": "***REMOVED SENSITIVE VALUE***",
        "mail_domain": "***REMOVED SENSITIVE VALUE***",
        "preview_max_x": "2048",
        "preview_max_y": "2048",
        "jpeg_quality": "60",
        "overwriteprotocol": "https",
        "trusted_proxies": "***REMOVED SENSITIVE VALUE***"
    },
    "apps": {
        "accessibility": {
            "enabled": "yes",
            "installed_version": "1.8.0",
            "types": ""
        },
        "activity": {
            "enabled": "yes",
            "installed_version": "2.15.0",
            "types": "filesystem"
        },
        "backgroundjob": {
            "lastjob": "33"
        },
        "calendar": {
            "enabled": "yes",
            "installed_version": "3.0.1",
            "types": ""
        },
        "circles": {
            "enabled": "yes",
[... 242 lines cut ...]
        },
        "user_status": {
            "enabled": "yes",
            "installed_version": "1.2.0",
            "types": ""
        },
        "viewer": {
            "enabled": "yes",
            "installed_version": "1.6.0",
            "types": ""
        },
        "weather_status": {
            "enabled": "yes",
            "installed_version": "1.2.0",
            "types": ""
        },
        "workflowengine": {
            "enabled": "yes",
            "installed_version": "2.4.0",
            "types": "filesystem"
        }
    }
}
```
I don't recall the significance of the numbers on the left side of trusted_domains other than being a key pair value.

shanti-bhardwa, 2021-12-18:
I have tried this fix and it works for me.

1. Connected my TP_LINK router to IIAB server on DELL PC using Ethernet.

2. Fixed the IP address of IIAB server based on its MAC address using Address Reservation with Router's DHCP. This means the router will always give this IP address to the Server.

        DELL PC  MAC address is  D4-BE-D9-3E-BA-70 and the IP is fixed at 192.168.0.100

3. Then I have edited this file /library/wordpress/wp-config.php and added these two lines to the file.

You will have to do sudo nano /library/wordpress/wp-config.phpto edit

        define('WP_Home', 'http://192.168.0.100/wordpress');

        define('WP_SITEURL', 'http://192.168.0.100/wordpress');

4. Then from another laptop connected to the same router using Ethernet, I can do 192.168.0.100/wordpress and get the Site home page

5. Also from IIAB home page, Clicking on Wordpress ICON works as expected.

6. Connecting to Router's WIFI SSID and then using 192.168.0.100/wordpress also works but NOT 172.18.96.1/wordpress or box/wordpress

7. Connecting to IIAB Server's SSID ( e.g. Kenya IIAB) works with 172.18.96.1/wordpress and also 192.168.0.100/wordpress
 
Will do some more testing with couple of other routers and also play with DNS within the router

jvonau, 2021-12-18:
> 6. Connecting to Router's WIFI SSID and then using 192.168.0.100/wordpress also works but NOT 172.18.96.1/wordpress or box/wordpress

 172.18.96.1/wordpress will never be reachable from the other subnet on 192.168.0.0/24. they are completely different networks, but getting box/wordpress to work would be at the router level where you would map the static ip address to the name 'box'. What is the router using as a domain name for the clients? That is the info from /etc/resolv.conf I asked for above. FWIW moodle is not as easy, but still do-able.

tim-moody, 2021-12-18:
>172.18.96.1/wordpress will never be reachable from the other subnet

and even if it were reachable, wp is a one host animal, though there have been attempts to softcode host.

jvonau, 2021-12-18:
The idea to have both the wired wan side and wireless lan side resolve to the same name on both subnets, but these considerations really need to be in place before installing and local_vars will need to be tweaked to match the layout.

jvonau, 2021-12-18:
Depending on if the other clients on 192.168.0.0/24 have the support installed using http://box.local might be viable as that is a different name resolution stack using mdns working with local broadcasts.

jvonau, 2021-12-18:
I see moodle has gotten easier over the years, there is a search and replace [tool](https://www.inmotionhosting.com/support/edu/moodle/change-url-moodle/) and better directions available.

shanti-bhardwa, 2021-12-18:
have tried this in wp-config.php 
define('WP_Home', 'http://box.local/wordpress');
define('WP_SITEURL', 'http://box.local/wordpress');

and works with two different  routers .Will test more.

shanti-bhardwa, 2021-12-18:
How do I configure the DNS in my router to use dnsmasq with IIAB?. Generally DNS is set up with Primary & Secondary servers e.g. with Google 8..8.8.8 & 8.8.4.4

jvonau, 2021-12-18:
Yea I was kind of thinking like that if your router doesn't support dynamic dns hostname entries from dhcp leases, replace 8.8.8 & 8.8.4.4 with the reserved address of 192.168.0.100 for the mac of the iiab box. Now there would be some tinkering on the IIAB box dnsmasq wise, but answering my questions makes this exercise less painful, documented, allows me some time so I can form some answers|actions in advance, quickest is with vpn access but I want a step by step done in plain sight

jvonau, 2021-12-19:
https://discourse.pi-hole.net/t/how-do-i-configure-my-devices-to-use-pi-hole-as-their-dns-server/245 is the basic idea substitute IIAB for Pi-Hole in that routine.

holta, 2022-05-23:
@shanti-bhardwa can we close this yet?

shanti-bhardwa, 2022-05-23:
yes can be closed Thanks

closed 2022-05-23 by shanti-bhardwa