iiab/iiab #3072: IIAB Kiwix proxy fails to proxy certain articles?

holta, 2021-12-18:
Apparently there's a bug in our 11-line Kiwix proxy...

https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/kiwix-nginx.conf.j2

...that prevents very real articles like ".apple" from being displayed.  As you can see this leads to *403 Forbidden*:

http://iiab.me/kiwix/wikipedia_en_all_maxi_2021-03/A/.apple

Whereas it works without our NGINX Kiwix proxy:

http://10.8.0.14:3000/kiwix/wikipedia_en_all_maxi_2021-03/A/.apple

As mentioned here:

- kiwix/kiwix-tools#513

jvonau, 2021-12-18:
http://iiab.me/kiwix/search?content=wikipedia_en_all_maxi_2021-03&pattern=apple

holta, 2021-12-18:
> http://iiab.me/kiwix/search?content=wikipedia_en_all_maxi_2021-03&pattern=apple

The above is a "Full Text Search" for "apple" with 97,264 search results.

Whereas the easier way to experience this bug is using a "Title Search" for "apple" (a search dropdown should appear almost instantly, before you hit Enter).

jvonau, 2021-12-18:
http://iiab.me/kiwix/search?content=wikipedia_en_all_maxi_2021-03&pattern=TLD Ok I see it is the leading . in the final part of the url. The nginx kwix template passes everything though to the port, the filtering might be in one of the other templates in use.

jvonau, 2021-12-18:
Seems more of a security feature of nginx, https://www.reddit.com/r/nginx/comments/i6et27/displaying_hidden_files/ A normal doc_root could have sensitive data within the .file or .directory, that is the point of hiding them there to begin with.

jvonau, 2021-12-18:
```
jvonau@box:/media/usb1/git/iiab$ ls -lh
total 3.2M
-rwxr-xr-x  1 root root  444 Oct 20 17:25 ansible.cfg
-rwxr-xr-x  1 root root   23 Oct 20 17:25 ansible_hosts
-rwxr-xr-x  1 root root  574 Oct 20 17:25 collections.yml
-rwxr-xr-x  1 root root  139 Oct 20 17:25 CONTRIBUTING.md
-rwxr-xr-x  1 root root 1.9K Oct 20 17:25 iiab-configure
-rwxr-xr-x  1 root root  396 Oct 20 17:25 iiab-from-cmdline.yml
-rwxr-xr-x  1 root root  420 Oct 20 17:25 iiab-from-console.yml
-rwxr-xr-x  1 root root 8.8K Oct 20 17:33 iiab-install
-rwxr-xr-x  1 root root 3.7K Oct 20 17:33 iiab-network
-rwxr-xr-x  1 root root  239 Oct 20 17:25 iiab-network.yml
-rwxr-xr-x  1 root root 1.4K Oct 20 17:25 iiab-stages.yml
-rwxr-xr-x  1 root root  227 Oct 20 17:33 install-support.yml
-rwxr-xr-x  1 root root  18K Oct 20 17:25 LICENSE
-rwxr-xr-x  1 root root  914 Oct 20 17:33 LICENSING.md
-rwxr-xr-x  1 root root 5.3K Oct 20 17:33 README.md
drwxr-xr-x 59 root root 128K Oct 20 17:33 roles
-rwxr-xr-x  1 root root  251 Oct 20 17:25 run-one-role.yml
-rwxr-xr-x  1 root root 4.8K Oct 20 17:33 runrole
-rwxr-xr-x  1 root root  571 Oct 20 17:25 runroles
-rwxr-xr-x  1 root root  186 Oct 20 17:25 runroles-base.yml
drwxr-xr-x  2 root root 128K Oct 20 17:33 scripts
drwxr-xr-x  2 root root 128K Oct 20 17:33 tests
-rwxr-xr-x  1 root root 1.4K Oct 20 17:33 test.yml
-rwxr-xr-x  1 root root  220 Oct 20 17:33 unmaintained-roles.txt
drwxr-xr-x  2 root root 128K Oct 21 06:29 vars
jvonau@box:/media/usb1/git/iiab$ ls -lah
total 3.9M
drwxr-xr-x  8 root root 128K Oct 20 17:33 .
drwxr-xr-x  7 root root 128K Dec 14 23:04 ..
-rwxr-xr-x  1 root root  444 Oct 20 17:25 ansible.cfg
-rwxr-xr-x  1 root root   23 Oct 20 17:25 ansible_hosts
-rwxr-xr-x  1 root root  574 Oct 20 17:25 collections.yml
-rwxr-xr-x  1 root root  139 Oct 20 17:25 CONTRIBUTING.md
drwxr-xr-x  8 root root 128K Oct 20 17:34 .git
drwxr-xr-x  2 root root 128K Oct 20 17:25 .github
-rwxr-xr-x  1 root root   69 Oct 20 17:33 .gitignore
-rwxr-xr-x  1 root root 1.9K Oct 20 17:25 iiab-configure
-rwxr-xr-x  1 root root  396 Oct 20 17:25 iiab-from-cmdline.yml
-rwxr-xr-x  1 root root  420 Oct 20 17:25 iiab-from-console.yml
-rwxr-xr-x  1 root root 8.8K Oct 20 17:33 iiab-install
-rwxr-xr-x  1 root root 3.7K Oct 20 17:33 iiab-network
-rwxr-xr-x  1 root root  239 Oct 20 17:25 iiab-network.yml
-rwxr-xr-x  1 root root 1.4K Oct 20 17:25 iiab-stages.yml
-rwxr-xr-x  1 root root  227 Oct 20 17:33 install-support.yml
-rwxr-xr-x  1 root root  18K Oct 20 17:25 LICENSE
-rwxr-xr-x  1 root root  914 Oct 20 17:33 LICENSING.md
-rwxr-xr-x  1 root root 5.3K Oct 20 17:33 README.md
drwxr-xr-x 59 root root 128K Oct 20 17:33 roles
-rwxr-xr-x  1 root root  251 Oct 20 17:25 run-one-role.yml
-rwxr-xr-x  1 root root 4.8K Oct 20 17:33 runrole
-rwxr-xr-x  1 root root  571 Oct 20 17:25 runroles
-rwxr-xr-x  1 root root  186 Oct 20 17:25 runroles-base.yml
drwxr-xr-x  2 root root 128K Oct 20 17:33 scripts
drwxr-xr-x  2 root root 128K Oct 20 17:33 tests
-rwxr-xr-x  1 root root 1.4K Oct 20 17:33 test.yml
-rwxr-xr-x  1 root root 2.4K Oct 20 17:33 .travis.yml
-rwxr-xr-x  1 root root  220 Oct 20 17:33 unmaintained-roles.txt
drwxr-xr-x  2 root root 128K Oct 21 06:29 vars

```
[image]

note .travis.yml and .git* files and directories are hidden from the enduser.

jvonau, 2021-12-18:
The bug would be in the original source where '.apple' was used in place of '(dot)apple' the current online version is different https://en.wikipedia.org/w/index.php?title=.apple&redirect=no with the history being https://en.wikipedia.org/w/index.php?title=.apple&action=history

jvonau, 2021-12-18:
The dot file blocking is done at https://github.com/iiab/iiab/blob/master/roles/nginx/templates/server.conf.j2#L36-L37, #'d out does in fact allow the hidden .git directory to be accessible (add the path in the url bar of the browser) but not viewable in the listing above. See the attached PR, could be that simple or much more complete, I don't have the data-set to set against.

holta, 2021-12-18:
The following fix seems good enough:

- PR #3074 

But let's test this thoroughly to be sure...so this issue can hopefully be closed in coming days.

jvonau, 2021-12-19:
If the fix was demonstrated to work close this issue.

holta, 2021-12-19:
> If the fix was demonstrated to work close this issue.

Yes, at the same time to do this properly I'd like to give it a few days, to monitor that the fix does not have unintended consequences 🤞

jvonau, 2021-12-19:
The danger I see is if some other request not intended for /kiwix but has /kiwix/ somewhere in the url this change could intercept that request.

holta, 2021-12-19:
> The danger I see is if some other request not intended for /kiwix but has /kiwix/ somewhere in the url this change could intercept that request.

The `location ^~ /kiwix` directive specifies a prefix (if I understand NGINX correctly:

https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/kiwix-nginx.conf.j2#L1

So hopefully we're on increasingly solid ground, following this official doc on "" :

https://docs.nginx.com/nginx/admin-guide/web-server/web-server/#nginx-location-priority

jvonau, 2021-12-19:
This passes my test of browsing to http://localhost/local_content/USB1/git/iiab/roles/kiwix/tasks/ so that one is off the table.

jvonau, 2021-12-19:
> > The danger I see is if some other request not intended for /kiwix but has /kiwix/ somewhere in the url this change could intercept that request.
> 
> The `location ^~ /kiwix` directive specifies a prefix (rather than any grep-like substring) here — if I understand NGINX correctly:

OK ^ anchors the left / to be the root dir of the request.

holta, 2021-12-26:
The fix looks solid (good enough) after a week:

- PR #3074

Thanks @jvonau for having reviewed this.

closed 2021-12-26 by holta