iiab/iiab #3077: (A) "size" erroneous in map-catalog.json (B) NEW OSM dataset from Oct/Nov 2020 (C) Prep for future datasets incl Satellite Photos

holta, 2021-12-21:
@georgejhunt

1) Should we fix `"size": 16841445376` that appears 3 times in http://download.iiab.io/content/OSM/vector-tiles/map-catalog.json which is apparently drawn from https://github.com/iiab/maps/blob/master/public/2021-01-27/map-catalog.json ?

   - [Line 21](https://github.com/iiab/maps/blob/master/public/2021-01-27/map-catalog.json#L21) under osm-planet_z0-z10_2019.mbtiles
   - [Line 47](https://github.com/iiab/maps/blob/master/public/2021-01-27/map-catalog.json#L47) under satellite_z0-z9_v3.mbtiles
   - [Line 75](https://github.com/iiab/maps/blob/master/public/2021-01-27/map-catalog.json#L75) under osm_africa_z11-z14_2019.mbtiles

2) Is [2020-10-06/modify_catalog.py](https://github.com/georgejhunt/maps/blob/addbase/public/2020-10-06/modify_catalog.py) the most recent and best way of doing this?

   Do you have any recommendations here, in general?

3) In contrast, 16841445376 appears only once (under osm_africa_z11-z14_2019.mbtiles) in `/library/www/html/common/assets/adm-map-catalog.json` apparently because it's auto-created by https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/files/scripts/osm-vect_v2_finish_install.py ?

4) NEW OpenStreetMap datasets from Oct/Nov 2020 are now posted to https://archive.org/details/osm-vector-mbtiles that we should take a look at in coming months (when we can!) in service to low-income communities worldwide and their students who will greatly benefit...

   We should lay out a rough plan for 2022 so these communities get what they want (is Sentinel satellite data...essentially auto-updated at the moment?!)

Thanks!

tim-moody, 2021-12-21:
there are two catalogs because the osm module and adm cons do the math differently, and the difference with respect to size reflect the different math.

holta, 2021-12-21:
> there are two catalogs because the osm module and adm cons do the math differently, and the difference with respect to size reflect the different math.

Thanks for context.  I haven't dug into the details, but just FYI `"size": 16841445376` is marked as identical in both files (all 3 + 1 = 4 cases) so this doesn't look like a math issue...?

1) `"size": 16841445376` appears [3 times in map-catalog.json](https://github.com/iiab/iiab/issues/3077#issue-1085321794)
2) `"size": 16841445376` appears [once at adm-map-catalog.json#L124](https://github.com/iiab/iiab-admin-console/blob/master/roles/common/files/map/adm-map-catalog.json#L124)

PS I revised the [quick explanation of adm-map-catalog.json in point 3.](https://github.com/iiab/iiab/issues/3077#issue-1085321794) above, and it's probably not 100% perfect (!) but hopefully at least clarifies a bit &mdash; towards helping everyone sort this all out for 2022 & Onwards.

tim-moody, 2021-12-21:
I'm not clear on what the issue is here. There could be as many as three.

1. "size": 16841445376 has the wrong value

What should the value be and why do you think so? Is it just that three things probably don't have the same size?

2. This size appears three times in one catalog and once in another

3. The tiles files could be upgraded.

Is there also a question about synchronization between the repo and d.iiab.io?

tim-moody, 2021-12-21:
The last question is why does it matter?

holta, 2021-12-21:
_The 3 issues in the subject line (A), (B) and (C) are to get offline communities and offline students of all kinds ready for the maps they need in 2022 and many years beyond._

**Item (E)** is definitely less important, but could be added as appropriate: should the stale "url" field be removed from iiab-catalog.json in future &mdash; if perhaps this information from ~5 years ago...proves no longer irrelevant?  An example from https://github.com/iiab/maps/blob/master/public/2021-01-27/map-catalog.json#L78 is here:

```
      "url": "https://archive.org/download/en-osm-omt_africa_2017-07-03_v0.23/en-osm-omt_africa_2017-07-03_v0.23.zip",
```

**Item (F)** that @tim-moody (almost but not quite!) suggests [above](https://github.com/iiab/iiab/issues/3077#issuecomment-998752123): Should iiab-catalog.json itself possibly be hosted on GitHub, if not now, then perhaps sometime in the future?  Visible versioning might greatly help field communities in future, to show lineage of different .mbtiles and Map Pack offerings available to them each year and in recent years.  (Possibly not needed in 2022?  But curatorial/library-oriented people will greatly appreciate choices / transparency of this nature, if and when it's achievable.)

**Item (G)** can happen along the way (the social glue that holds everything together) and is implicit, and rarely talked about (until it's too late, then "Where Are The Docs?" is suddenly the most popular question, when they don't exist...)

Specifically: which community docs and/or pipeline automation might be most important to make this quality mapmaking welcoming to new talent contributors over the coming ~5 years, helping @georgejhunt and others make this extremely valuable contribution fresh every 1-2 years or so, and for many years to come?

_RECAP: Routine technical glitches can easily be solved now or in due course.  At the same time, the higher-level purpose should be protecting our foundation so that in 5 years from now...the extremely useful foundation we've built over the past 5 years...is honored and disseminated increasingly broadly (and inevitably in new ways, that we cannot yet predict, as 2020 becomes 2030...)_

holta, 2022-04-21:
Some catalog and hosting and menu-def cleanups may remain in coming days/weeks, e.g. relating to things like:

- #3111

But the larger task is now done!

_Thanks to all the work below, from @georgejhunt and @tim-moody:_

- PR iiab/maptools#4
  - https://github.com/iiab/maptools/blob/main/README.md from
  - https://github.com/georgejhunt/maptools/blob/main/README.md
- PR iiab/maps#45
  - SEE package-lock.json & package.json within:
  - https://github.com/iiab/maps/tree/master/osm-source/pages/webpack
- PR #3165
  - Code for generation is a Jupyter Notebook at https://github.com/georgejhunt/maps/blob/maps7.3/osm-source/jupyter-code/maps.ipynb
  - For "2020" (this version) https://github.com/georgejhunt/maps/blob/maps7.3/osm-source/jupyter-code/README.md
- PR iiab/iiab-admin-console#478

holta, 2022-04-28:
Just to clarify now that the above PR's are all merged, progress is ongoing here:

- PR #3192
- PR iiab/maps#54
- iiab/maps#55

holta, 2022-04-28:
Huge progress continues:

- PR iiab/iiab-admin-console#484
- PR #3196
- PR #3197

holta, 2022-04-29:
Question(s):

- [ ] Is [geodb.pl](https://github.com/iiab/maps/blob/master/tools.unused/geodb.pl) the only (or most important) thing left we need to migrate from [tools.unused](https://github.com/iiab/maps/tree/master/tools.unused) to https://github.com/iiab/maptools ?
- [ ] Probably there are other items on the checklist arising from Thursday morning's call that I'm forgetting !

holta, 2022-04-30:
Subtle font adjustments yesterday and today:

- PR #3192
- PR #3196
- PR #3197
- PR #3198
- iiab/maps#57
- PR #3203

tim-moody, 2022-09-21:
still needed?

closed 2022-09-21 by holta