iiab/iiab #3091: "Refresh Kiwix Catalog" failed 3 times, on 3 different RPi 4's (& "Refresh OER2Go Catalog" failed once)

holta, 2022-01-01:
Likely this is as a result of [kiwix-tools 2022-01-01](http://download.kiwix.org/nightly/2022-01-01/) that is part of IIAB 7.2, and if so waiting a couple days for a cleaner version of kiwix-tools will hopefully solve this.

@tim-moody is there an easy way to tell if this arising from a regression (or change) within `/opt/iiab/kiwix/bin/kiwix-manage` ?

This "Refresh Kiwix Catalog" failure happened on 3 different Raspberry Pi 4's, each running 32-bit Raspberry Pi OS 11:

```
Downloading Catalogs and Building Local Data Files.
Starting xml download from Kiwix
library_zim.xml received from Kiwix
Reading Our Catalog
Parsing xml downloads from Kiwix
Starting of processing xml download from Kiwix to /etc/iiab/kiwix_catalog.json
Traceback (most recent call last):
  File "/opt/admin/cmdsrv/scripts/get_kiwix_catalog", line 331, in <module>
    main()
  File "/opt/admin/cmdsrv/scripts/get_kiwix_catalog", line 116, in main
    zim_attr = parse_attr_new(attr)
  File "/opt/admin/cmdsrv/scripts/get_kiwix_catalog", line 197, in parse_attr_new
    if attributes['date'] < magicDate:
KeyError: 'date'
Getting the Kiwix Catalog Failed.
Please run iiab-get-kiwix-cat again later.
```

Whereas this "Refresh OER2Go Catalog" happened only on the "MEDICAL" install of IIAB:

```
Traceback (most recent call last):
  File "/opt/admin/cmdsrv/scripts/get_oer2go_catalog", line 183, in <module>
    main()
  File "/opt/admin/cmdsrv/scripts/get_oer2go_catalog", line 85, in main
    err_num, err_str, oer2go_catalog = get_oer2go_cat()
  File "/opt/admin/cmdsrv/scripts/get_oer2go_catalog", line 157, in get_oer2go_cat
    url_handle = urllib.request.urlopen(adm.CONST.oer2go_cat_url)
  File "/usr/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
  File "/usr/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/usr/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.9/urllib/request.py", line 1375, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/usr/lib/python3.9/urllib/request.py", line 1350, in do_open
    r = h.getresponse()
  File "/usr/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/usr/lib/python3.9/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.9/http/client.py", line 276, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response
Getting the OER2Go/RACHEL Catalog Failed.
Please run iiab-get-oer2go-cat again later.
```

This final phase (or very similar) was visible on all 3 Raspberry Pi 4's:

```
Waiting to run iiab-make-kiwix-lib.py
Now running iiab-make-kiwix-lib.py
Writing zim_versions_idx
Finished making Kiwix library.xml
Updating Home Page Menu
Updating kiwix menu items
Updating iiab installed services' menu items
Adding en-mediawiki to Menu
Updating installed OER2Go menu items
(Re)Starting CMDSRV
Created symlink /etc/systemd/system/multi-user.target.wants/iiab-cmdsrv.service → /etc/systemd/system/iiab-cmdsrv.service.
```

FYI the http://box/kiwix/test/A/index.html page appears to load just as it should, on all 3 Raspberry Pi 4's.

Likely related:

- PR #3089

PS announcing IIAB 7.2 will be delayed until this is worked out in coming days.

holta, 2022-01-01:
> Likely this is as a result of [kiwix-tools 2022-01-01](http://download.kiwix.org/nightly/2022-01-01/)

The above suspicion is incorrect.

As I've confirmed the exact same problem occurs with https://download.kiwix.org/nightly/2021-11-30/kiwix-tools_linux-armhf-2021-11-30.tar.gz from 32 days ago, and several other nightly builds of kiwix-tools from December.

@tim-moody do you have other ideas?

holta, 2022-01-01:
Another Data Point:

Retrofitting http://download.kiwix.org/release/kiwix-tools/kiwix-tools_linux-armhf-3.1.2-5.tar.gz from 2021-06-10 (i.e. putting its 4 binaries in the usual place in /opt/iiab/kiwix/bin) does not fix the problem.

So is this problem perhaps resulting from a recent change in Kiwix's catalog structure at http://download.kiwix.org/library/library_zim.xml ?

tim-moody, 2022-01-01:
The following have no date attribute and only one has a name attribute:

{'id': '35f06a52-1e99-8369-ec39-c93ff758b20b', 'path': '../var/www/download.kiwix.org/zim/wikipedia/wikipedia_en_wp1-0.5_2007-03.zim', 'title': 'wikipedia en wp1-0.5 2007-03', 'tags': '_ftindex:no;_pictures:yes;_videos:yes;_details:yes', 'url': 'http://download.kiwix.org/zim/wikipedia/wikipedia_en_wp1-0.5_2007-03.zim.meta4', 'articleCount': '1997', 'mediaCount': '16891', 'size': '256588'}
{'id': '43b94328-4e20-9ccd-1e07-8d8fa551bfaf', 'path': '../var/www/download.kiwix.org/zim/other/plume-app.co_fr_all_2021-12.zim', 'title': 'Le Dojo de Plume', 'description': 'Tu trouveras dans le Dojo de nombreuses activités amusantes pour progresser en français.', 'language': 'fra', 'creator': 'Plume', 'publisher': 'Kiwix', 'name': 'plume-app.co_fr_all', 'tags': '_ftindex:no;_pictures:yes;_videos:yes;_details:yes', 'url': 'http://download.kiwix.org/zim/other/plume-app.co_fr_all_2021-12.zim.meta4', 'articleCount': '47', 'mediaCount': '119', 'size': '7488'}
{'id': '90bfe9e9-6e9e-3c4c-d18f-0d00107b0000', 'path': '../var/www/download.kiwix.org/zim/wikipedia/wikipedia_ml_500+_2010-05.zim', 'title': 'wikipedia ml 500+ 2010-05', 'tags': '_ftindex:no;_pictures:yes;_videos:yes;_details:yes', 'url': 'http://download.kiwix.org/zim/wikipedia/wikipedia_ml_500+_2010-05.zim.meta4', 'articleCount': '515', 'mediaCount': '6862', 'size': '56156'}
{'id': '9bcd90f8-db78-5d68-1a81-2528ca25e806', 'path': '../var/www/download.kiwix.org/zim/wikipedia/wikipedia_en_wp1-0.7_2009-05.zim', 'title': 'wikipedia en wp1-0.7 2009-05', 'tags': '_ftindex:no;_pictures:yes;_videos:yes;_details:yes', 'url': 'http://download.kiwix.org/zim/wikipedia/wikipedia_en_wp1-0.7_2009-05.zim.meta4', 'articleCount': '90284', 'mediaCount': '167086', 'size': '2388987'}

holta, 2022-01-01:
> The following have no date attribute and only one has a name attribute:

Interesting.

Should we...possibly...try to trap + announce these missing fields...so as to become more resilient?

(Or conversely is this too much work &mdash; e.g. within https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/templates/scripts/get_kiwix_catalog &mdash; if it's not expected to happen too often?)

tim-moody, 2022-01-01:
these have no name:
../var/www/download.kiwix.org/zim/stack_exchange/productivity.stackexchange.com_en_all_2017-10.zim
../var/www/download.kiwix.org/zim/other/ubuntuusers_de_all_2017-03.zim
../var/www/download.kiwix.org/zim/other/bouquineux_fr_all_2013-07.zim
../var/www/download.kiwix.org/zim/wikipedia/wikipedia_en_wp1-0.5_2007-03.zim
../var/www/download.kiwix.org/zim/other/ubuntudoc_fr_all_2015-12.zim
../var/www/download.kiwix.org/zim/wikipedia/wikipedia_en_wp1-0.8_orig_2010-12.zim
../var/www/download.kiwix.org/zim/wikibooks/wikibooks_ang_all_nopic_2015-02.zim
../var/www/download.kiwix.org/zim/other/oer4schools_en_all_2015-11.zim
../var/www/download.kiwix.org/zim/stack_exchange/programmers.stackexchange.com_en_all_2017-10.zim
../var/www/download.kiwix.org/zim/wikipedia/wikipedia_ml_500+_2010-05.zim
../var/www/download.kiwix.org/zim/wikipedia/wikipedia_en_wp1-0.7_2009-05.zim
../var/www/download.kiwix.org/zim/wiktionary/wiktionary_no_all_nopic_2016-08.zim
../var/www/download.kiwix.org/zim/other/rezeptewiki_de_all_2014-06.zim
../var/www/download.kiwix.org/zim/other/ubuntuusers_de_all_nopic_2017-03.zim
../var/www/download.kiwix.org/zim/stack_exchange/startups.stackexchange.com_en_all_2017-06.zim

tim-moody, 2022-01-01:
so yes, the question is what to do. easiest is just to skip them. the no-names are all pretty old

plume-app.co_fr_all_2021-12.zim with no date is new, whatever it is. given that it is in 'other' was likely build by some special scrape, which means we could see more non-standard zims.

holta, 2022-01-01:
> so yes, the question is what to do. easiest is just to skip them. the no-names are all pretty old

Presumably...if both those fields are "essential" ?

> plume-app.co_fr_all_2021-12.zim with no date is new, whatever it is. given that it is in 'other' was likely build by some special scrape, which means we could see more non-standard zims.

Indeed: all are many years old &mdash; except for download.kiwix.org/zim/other/plume-app.co_fr_all_2021-12.zim published 3 days ago.

(Any idea why is appears to be suddenly happening now?  Has this ever happened before, or not really?)

tim-moody, 2022-01-01:
looks to me like some old zims got added back into the downloads.

I use date to distinguish old zims, where the default was no video from new zims where default is with videos

I don't remember if the front end uses name or not

looking into rewriting the old date logic

tim-moody, 2022-01-01:
looks like _videos tag is in all current zims, so no need to compute it

tim-moody, 2022-01-01:
in fact the four required tags are in all zims

```
for child in x:
  attr = child.attrib
  #tags = attr['tags'].split(';') if 'tags' in attr else []
  found = True
  if '_ftindex' not in attr['tags']:
    found = False
  if '_pictures' not in attr['tags']:
    found = False
  if '_videos' not in attr['tags']:
    found = False
  if '_details' not in attr['tags']:
    found = False                
  if not found:    
    print(attr['path'])  

```

tim-moody, 2022-01-01:
this should be two issues, kiwix and oer2go

tim-moody, 2022-01-01:
https://github.com/iiab/iiab-admin-console/pull/454

holta, 2022-01-01:
@tim-moody are there any lessons we should draw from these kinds of catalog fragilities, to reduce everyone's exposure from similar glitches in future?

(While declaring victory today, and facing these year by year as they inevitably arise...)

tim-moody, 2022-01-01:
>facing these year by year as they inevitably arise

you have answered your own question. this could not have been anticipated and in fact required changing code that handles an earlier such event.

I might add that few software suppliers can turn around such fixes in less than 24 hours.

closed 2022-01-01 by holta