iiab/iiab #3092: "Refresh OER2Go Catalog" failed

tim-moody, 2022-01-01:
not sure if this is reproducible

holta, 2022-01-01:
Whether this is reproducible or not, the failure output is here:

- https://github.com/iiab/iiab/issues/3091#issue-1091752250

holta, 2022-01-02:
I just ran `/usr/bin/iiab-get-oer2go-cat` 10 times, and it failed 8 times, while succeeding 2 times: (regardless which IIAB machine I use)

```
root@box:~# iiab-get-oer2go-cat
SUCCESSroot@box:~# iiab-get-oer2go-cat
Traceback (most recent call last):
  File "/opt/admin/cmdsrv/scripts/get_oer2go_catalog", line 183, in <module>
    main()
  File "/opt/admin/cmdsrv/scripts/get_oer2go_catalog", line 85, in main
    err_num, err_str, oer2go_catalog = get_oer2go_cat()
  File "/opt/admin/cmdsrv/scripts/get_oer2go_catalog", line 157, in get_oer2go_cat
    url_handle = urllib.request.urlopen(adm.CONST.oer2go_cat_url)
  File "/usr/lib/python3.7/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.7/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/usr/lib/python3.7/urllib/request.py", line 543, in _open
    '_open', req)
  File "/usr/lib/python3.7/urllib/request.py", line 503, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.7/urllib/request.py", line 1352, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/usr/lib/python3.7/urllib/request.py", line 1327, in do_open
    r = h.getresponse()
  File "/usr/lib/python3.7/http/client.py", line 1352, in getresponse
    response.begin()
  File "/usr/lib/python3.7/http/client.py", line 310, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.7/http/client.py", line 279, in _read_status
    raise RemoteDisconnected("Remote end closed connection without"
http.client.RemoteDisconnected: Remote end closed connection without response
Getting the OER2Go/RACHEL Catalog Failed.
Please run iiab-get-oer2go-cat again later.
```

Whereas success on the very same machine (just moments before/later) appears as follows:

```
root@box:~# iiab-get-oer2go-cat
SUCCESS
```

@tim-moody do you have any guesses...as to why the failure rate is so very high?

tim-moody, 2022-01-02:
RemoteDisconnected suggests that the remote disconnected.

http://dev.worldpossible.org/cgi/json_api_v1.pl seems to work.

so no idea.

tim-moody, 2022-01-02:
I did a couple of url_handle = urllib.request.urlopen(adm.CONST.oer2go_cat_url) (line that fails) with no failure.

any pattern in the OS or version of python?

I don't like what I'm reading about urllib.request and it's handlers. maybe they are retaining state and are busy at the os level.

this could be made a lot simpler with 

```
r = requests.get(adm.CONST.oer2go_cat_url)
if r.status_code:
    oer2go_catalog = r.json()
else:
    oer2go_catalog = None
```

holta, 2022-01-02:
> any pattern in the OS or version of python?

It appears to fail regardless of OS (Raspberry Pi OS "Lite" or "with desktop" or Ubuntu 22.04 daily build).

The failure rate is currently about 60% (about 3 in 5 attempts failed, after I tried running `iiab-get-oer2go-cat` about 50 times).

I can't see a pattern to date.  Almost as if it's completely random.  Certainly I don't understand it.

holta, 2022-06-04:
Should this ticket possible be closed?
(If the situation's much less severe 5-6 months ago later?)

As an aside, oer2go_catalog_json was just refined in roles/cmdsrv/templates/scripts/get_oer2go_catalog3.py :

- PR iiab/iiab-admin-console#497

tim-moody, 2022-06-04:
there are two problems. 1) the catalog is not always available or perhaps times out. 2) when that situation is trapped there is a programming error. I just addressed the second problem.

holta, 2022-06-05:
> there are two problems. 1) the catalog is not always available or perhaps times out. 2) when that situation is trapped there is a programming error. I just addressed the second problem.

Problem 1) too!

- PR iiab/iiab-admin-console#498

jvonau, 2022-07-10:
>Just a data point while installing IIAB onto RPi 4 today:
"Finished writing to /etc/iiab/kiwix_catalog.json
SUCCESSRetrying download of catalog.
Retrying download of catalog.
Retrying download of catalog.
Retrying download of catalog.
Retrying download of catalog.
OER2Go Catalog download failed. Using existing catalog."

>will never succeed until the Indonesia casino implements it

>RACHEL is "disappearing" all references to OER2Go it would appear:
https://github.com/rachelproject/oer2go-web

holta, 2022-07-14:
Thanks to @tim-moody for this important fix/workaround a week ago:

- PR iiab/iiab-admin-console#506

tim-moody, 2022-09-21:
why is this still open?

holta, 2022-09-21:
Seems reliable (or reliable enough!) in recent months.

So let's close this.

closed 2022-09-21 by holta