iiab/iiab #3097: Should host_country_code in local_vars.yml update wpa_supplicant.conf not just hostapd.conf ?

holta, 2022-01-03:
Should https://github.com/iiab/iiab/blob/master/roles/network/tasks/rpi_debian.yml#L23-L33 change the country code in /etc/wpa_supplicant/wpa_supplicant.conf when someone proactively changes `host_country_code` in local_vars.yml and then runs ./iiab-network ?

(I just noticed that https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/hostapd.conf.j2#L18 does indeed change it in /etc/hostapd/hostapd.conf — in these same circumstances — whereas wpa_supplicant.conf does not get changed.)

holta, 2022-01-03:
FYI I just spoke with @jvonau who indicates he doesn't care, either way.

holta, 2022-01-04:
FYI whatever is decided, this should be aligned with http://FAQ.IIAB.IO -> "23 Any other networking tips?"

Which currently says: (as the #1 recommendation on top)

> Please change "host_country_code: US" to the correct country code, in `/etc/iiab/local_vars.yml`, prior to installing IIAB. If IIAB is already installed, also run `cd /opt/iiab/iiab` then `sudo ./iiab-network` and then reboot.

jvonau, 2022-01-04:
"Use the same country code as you used with raspi-config"

holta, 2022-01-04:
For now I've added a context note to:

https://wiki.iiab.io/go/FAQ#Any_other_networking_tips?

holta, 2022-05-20:
A different approach is implemented by:

- PR #3179

In short:

https://github.com/iiab/iiab/blob/743d8f260157178faa588870efda4f215a1ad153/vars/default_vars.yml#L109-L119

closed 2022-05-20 by holta