iiab/iiab #3100: There should be a way to drag Content Packs (on an IIAB Home Page) into the trash

holta, 2022-01-06:
Using something like IIAB Admin Console (http://box.lan/admin) **> Content Menus > Edit Content Menus > Content Item List** >[DRAG CONTENT ITEM...SOMEWHERE ?]

Specifically: people populating their IIAB Home Page with custom content items (typically RACHEL-like modules, but also other things!) that do not appear on the right-side ("Available Menu Item Definitions") of the above interface.

So dragging these from the left-side ("Menu Items in Current Menu") to the right-side does not work, when trying to remove such items from their IIAB Home Page.

What they really want is...some kind of Trash Can or Recycle Bin?

tim-moody, 2022-01-07:
how do you add a menu item to the menu if it does not exist?

holta, 2022-01-07:
> how do you add a menu item to the menu if it does not exist?

FileZilla is apparently being used to upload their own RACHEL-like modules to `/library/www/html/modules` but I don't know all the details as to why this is happening, and how they got in this situation.

I'll try to hopefully find out more details in coming weeks.

tim-moody, 2022-01-07:
so now we have modules that are not in the catalog. these should be in local_content.
if a menu contains the name of a menu def, that menu def must exist or the menu will show an error.
if a menu def exists it should be in the list on the right, but if it was just created it may be necessary to refresh the screen.

tim-moody, 2022-01-07:
also, if you create a menu def it is automatically added to the menu, I think. So that way a menu def could get added that is not in the list on the right until you refresh the screen.

A trash might be nice, but there is a lot to work out. It will scroll of the screen, and the screen is already very busy. Maybe the best answer would be to allow drag right even if the item is not already on the right.

holta, 2022-01-07:
> if a menu contains the name of a menu def, that menu def must exist or the menu will show an error.

They did mention error "file not found or improperly formatted" but that's all I know.

Does that shed any light, on this particular situation anyway?

> if a menu def exists it should be in the list on the right, but if it was just created it may be necessary to refresh the screen.

Great to know.
Sorry I don't have more details, but I'll try to find out.

> A trash might be nice, but there is a lot to work out. It will scroll of the screen, and the screen is already very busy. Maybe the best answer would be to allow drag right even if the item is not already on the right.

Either way would be great!

tim-moody, 2022-01-07:
>They did mention error "file not found or improperly formatted" but that's all I know.

sounds like manual editing of the menu file. I don't support that.

holta, 2022-01-07:
> sounds like manual editing of the menu file.

I don't think so.  As this person made clear they never use Linux, and had no idea about `/library/www/html/home/menu.json` when I mentioned that yesterday.

They do everything from Windows.

Which seems to mean FileZilla on Windows to port Content Packs / Items from RACHEL.

And then Admin Console where they got stuck unable to delete from their IIAB Home Page

holta, 2022-01-09:
FYI I asked him for feedback on his overall experience never having used IIAB before.  He replied:

> **[Admin Console > Content Menus]** needs some work to make it more intuitive:
>
> a.       It should default to Content Item List and not Menu Properties.
> b.       The Available Menu Item Definitions should be sorted, duplicates removed, and when a module is moved to the left column it should disappear from the right column.
> c.       It is not clear the functions of Load Menu, Refresh Lists, Sync Menu Defs and Update Home Menu.
> d.       The Edit Menu Item page was quite confusing and took me many trial-and-error attempts before I was successful in bringing in our custom modules from RACHEL.

> That being said, I was pleased that in a relatively short period of time...I was able to produce a finished product that nicely mirrors what our users have been seeing on the RACHEL.

shanti-bhardwa, 2022-01-09:
We should add an option on Admin portal to install a module from an external device where the module is NOT defined in the catalogue. People can then create their own modules and get them on the main Home page. I have not used local_content so not sure how it works.
There are existing scripts which should be easy to modify. Doing it manually to add the image , create a .json file and then edit menu.json is cumbersome.

tim-moody, 2022-01-09:
>We should add an option on Admin portal to install a module from an external device where the module is NOT defined in the catalogue. People can then create their own modules and get them on the main Home page. I have not used local_content so not sure how it works.
There are existing scripts which should be easy to modify. Doing it manually to add the image , create a .json file and then edit menu.json is cumbersome.

Please create a Feature Request on iiab-admin-console repo with details about what should be automated. This is not a trivial task, so this detail will be needed in order to implement it. Please mention the names of the existing scripts you refer to.

tim-moody, 2022-01-09:
>FYI I asked him for feedback on his overall experience never having used IIAB before. He replied:

>>[Admin Console > Content Menus] needs some work to make it more intuitive:

>>a. It should default to Content Item List and not Menu Properties.
b. The Available Menu Item Definitions should be sorted, duplicates removed, and when a module is moved to the left column it should disappear from the right column.
c. It is not clear the functions of Load Menu, Refresh Lists, Sync Menu Defs and Update Home Menu.
d. The Edit Menu Item page was quite confusing and took me many trial-and-error attempts before I was successful in bringing in our custom modules from RACHEL.

I have a request and a question. 

The request is that this item be disintermediated. The title of this ticket is not in the list nor are other items reported above. It would be helpful if the person with questions or problems reported them.

The question is what supporting documentation was consulted before using this Adm Cons item, faq, wiki, help?

a. This has been suggested before and I still disagree. A user not presented first with Menu Properties will never visit it and end up with support questions.
b. They are sorted and there are no duplicates. The problem is that they are sorted by menu def id, but the title is displayed. Further there is nothing to prevent duplicate titles for different menu defs. At a minimum the menu def id should be included on the title line so the sorting and versioning are more transparent. It is already in the mouse over tip, but you have to know to look for it. Would you prefer sorting by menu def id, as at present, or title? There can still be duplicates as they are not really duplicates only duplicate titles.
c. See my question regarding help.
d. Can you provide some details on what you found confusing? How many modules did you add and about how much time did they take? How much time did the last one take? See c.

tim-moody, 2022-01-09:
>when a module is moved to the left column it should disappear from the right column.

This is a good idea, but I don't like hiding things. I wonder if marking it as ON MENU would meet your needs.