iiab/iiab #3105: Files from USB Drive Fail to Appear

jackrodonnell, 2022-01-17:
I have a USB stick which contains html files.  I've mounted the USB on my Lenovo T440p laptop which is running with UbuntuMate 20.04.3 and IIAB 8.0  I select "Insert a USB key/drive to publish files here" from the menu.  I see my mounted Parent Directory and my mounted USB as USB0.  However none of the HTML files appear when I select USB0

 http://sprunge.us/2zzXQZ
Thanks

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

jackrodonnell, 2022-01-17:
The HTML files that failed to display were on a USB with ext4.  I moved the files to a USB with Fat32.  Same problem. I removed the HTML files and added a PDF file on each of these USBs.  Both PDF files on these 2 USBs were visible via IIAB.  I then placed one HTML file and one PDF file on each of these 2 USBs.  IIAB was not able to display any of these files

tim-moody, 2022-01-17:
>I select "Insert a USB key/drive to publish files here" from the menu

what menu is that?

what url are you viewing when you say the files do not display? what when they do? http://box/USB?

what does ll /library/www/html/local_content/ show?

>I then placed one HTML file and one PDF file on each of these 2 USBs

was the html file named index.html? something else?

jackrodonnell, 2022-01-17:
Tim
See the screenshot for the menu.  
[image]

Concerning the URL: On my local laptop that is running IIAB, I have tried both http://172.18.96.1/usb and box/usb.  Neither one results in a display of the HTML files. However I just tried accessing IIAB from my phone.  When I enter http://172.18.96.1/usb the HTML files DO NOT  appear on my phone.  When I access using box/usb the HTML files DO appear on my phone

Concerning /library/www/html/local_content/  I'm not sure where this directory is at??

Concerning "was the html file named index.html?"   YES.  When I accessed IIAB on my phone I could open index.html and all of the subsequent links functioned

Thanks
Jack

tim-moody, 2022-01-17:
what happens if you use box/local_content as the url on the two devices?

jackrodonnell, 2022-01-17:
When I access box/local_content on my local device with IIAB or remotely on my  iPhone, the HTML files are not visible

tim-moody, 2022-01-17:
puzzling. It can happen that the mount and unmount functions get confused in creating and removing links to the usb. I would try examining the file system with the Mate File Manager, especially /library/www/html/local_content/.

You could also try removing the USB and rebooting.

jackrodonnell, 2022-01-17:
I suspected a mounting/unmounting issue.  I've constantly rebooted and restarted IIAB both locally on the IIAB laptop and on my phone.  
I've checked mounted devices and see

Filesystem      Size  Used Avail Use% Mounted on
udev            7.7G     0  7.7G   0% /dev
tmpfs           1.6G  1.8M  1.6G   1% /run
/dev/sda5       916G  199G  671G  23% /
tmpfs           7.7G   12M  7.7G   1% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           7.7G     0  7.7G   0% /sys/fs/cgroup
/dev/loop2       56M   56M     0 100% /snap/core18/2284
/dev/loop3      165M  165M     0 100% /snap/gnome-3-28-1804/161
/dev/loop1       56M   56M     0 100% /snap/core18/2128
/dev/loop0      128K  128K     0 100% /snap/bare/5
/dev/loop4       66M   66M     0 100% /snap/gtk-common-themes/1519
/dev/loop5       16M   16M     0 100% /snap/ubuntu-mate-welcome/626
/dev/loop7      128K  128K     0 100% /snap/software-boutique/54
/dev/loop6      168M  168M     0 100% /snap/spotify/56
/dev/loop8       16M   16M     0 100% /snap/ubuntu-mate-welcome/646
/dev/loop9       44M   44M     0 100% /snap/snapd/14295
/dev/loop10      33M   33M     0 100% /snap/snapd/12704
/dev/sda1       511M  4.0K  511M   1% /boot/efi
tmpfs           1.6G   28K  1.6G   1% /run/user/1001
/dev/sdb1       113G   18G   90G  17% /media/usb0
Filesystem      Size  Used Avail Use% Mounted on
udev            7.7G     0  7.7G   0% /dev
tmpfs           1.6G  1.8M  1.6G   1% /run
/dev/sda5       916G  199G  671G  23% /
tmpfs           7.7G   12M  7.7G   1% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           7.7G     0  7.7G   0% /sys/fs/cgroup
/dev/loop2       56M   56M     0 100% /snap/core18/2284
/dev/loop3      165M  165M     0 100% /snap/gnome-3-28-1804/161
/dev/loop1       56M   56M     0 100% /snap/core18/2128
/dev/loop0      128K  128K     0 100% /snap/bare/5
/dev/loop4       66M   66M     0 100% /snap/gtk-common-themes/1519
/dev/loop5       16M   16M     0 100% /snap/ubuntu-mate-welcome/626
/dev/loop7      128K  128K     0 100% /snap/software-boutique/54
/dev/loop6      168M  168M     0 100% /snap/spotify/56
/dev/loop8       16M   16M     0 100% /snap/ubuntu-mate-welcome/646
/dev/loop9       44M   44M     0 100% /snap/snapd/14295
/dev/loop10      33M   33M     0 100% /snap/snapd/12704
/dev/sda1       511M  4.0K  511M   1% /boot/efi
tmpfs           1.6G   28K  1.6G   1% /run/user/1001
/dev/sdb1       113G   18G   90G  17% /media/usb0
iiab-admin@box:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            7.7G     0  7.7G   0% /dev
tmpfs           1.6G  1.8M  1.6G   1% /run
/dev/sda5       916G  199G  671G  23% /
tmpfs           7.7G   12M  7.7G   1% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           7.7G     0  7.7G   0% /sys/fs/cgroup
/dev/loop2       56M   56M     0 100% /snap/core18/2284
/dev/loop3      165M  165M     0 100% /snap/gnome-3-28-1804/161
/dev/loop1       56M   56M     0 100% /snap/core18/2128
/dev/loop0      128K  128K     0 100% /snap/bare/5
/dev/loop4       66M   66M     0 100% /snap/gtk-common-themes/1519
/dev/loop5       16M   16M     0 100% /snap/ubuntu-mate-welcome/626
/dev/loop7      128K  128K     0 100% /snap/software-boutique/54
/dev/loop6      168M  168M     0 100% /snap/spotify/56
/dev/loop8       16M   16M     0 100% /snap/ubuntu-mate-welcome/646
/dev/loop9       44M   44M     0 100% /snap/snapd/14295
/dev/loop10      33M   33M     0 100% /snap/snapd/12704
/dev/sda1       511M  4.0K  511M   1% /boot/efi
tmpfs           1.6G   28K  1.6G   1% /run/user/1001
/dev/sdb1       113G   18G   90G  17% /media/usb0

Just now I rebooted the laptop and phone and cleared caches.  I attempted to view box/usb/USB0 on my phone and this time it failed to appear on both laptop anIfd phone.

I know it would be nice to solve this mystery but I'm setting up for 12 laptops going to a school in Haiti.  By using the USB with IIAB I was trying to consolidate WorldComputerExchange Content with IIAB Content.  There seems to be a USB issue which I dont want to pass on to those in Haiti.  All of the Content on my USB is on the hard drive of the computers that we will send.  So I may just leave the content in 2 places.  I can create web links on the laptops that will lead to the IIAB or to the hard drive content

holta, 2022-01-17:
Thanks @jackrodonnell for pushing.

Just to be clear this feature[*] is Rock Solid for others, so we definitely do want to help you get closer to identifying what's not working (intermittently?) in your unique situation??

Does a call help in coming days — simply to review all facts + assumptions, so others can try to reproduce exactly what's failing for you?
<br>
[*] http://FAQ.IIAB.IO -> **"4 Can teachers display their own content?"**

jackrodonnell, 2022-01-17:
Yes a call would be fine
      773-316-2944
    Tues-Fri best
      Jack
    
    On 1/17/22 1:49 PM, A Holt wrote:
    
    
      
      Thanks @jackrodonnell
        for pushing.
      Just to be clear this feature[*] is Rock Solid for
        others, so we definitely do want to help you get closer to
        identifying what's not working (intermittently?) in your unique
        situation??
      Does a call help in coming days — simply to review
        all facts + assumptions, so others can try to reproduce exactly
        what's failing for you?
        
        
        [*] http://FAQ.IIAB.IO -> "4
          Can teachers display their own content?"
      —
        
      [
{
"target": "https://github.com/iiab/iiab/issues/3105#issuecomment-1014844934",
"url": "https://github.com/iiab/iiab/issues/3105#issuecomment-1014844934",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

holta, 2022-01-17:
@jackrodonnell FYI I also do IIAB work in Haiti — please to feel free to write / coordinate on these topics anytime:

holt@unleashkids.org

tim-moody, 2022-01-18:
As long as the server is still connected to the internet you could give me remote access by issuing the command

/usr/bin/iiab-vpn-on

if you like.

jvonau, 2022-01-18:
```
=IIAB==========================================================================
COMMAND: /usr/bin/systemctl is-active display-manager.service    # Graphical Desktop?

active
```

jvonau, 2022-01-18:
https://github.com/iiab/iiab/issues/2456

holta, 2022-01-18:
> [#2456 (comment)](https://github.com/iiab/iiab/issues/2456#issuecomment-649807106)

Perfect, thanks @jvonau.

I just spoke live with @jackrodonnell and suggested this to him.

Tonight and later this week he will do more testing and follow up when he has more time!

holta, 2022-01-19:
@jackrodonnell what happens if you remove the index.html file from each USB stick?

Any chance this allows the other {PDF, video, css} files to then be seen within http://box/usb ?

jackrodonnell, 2022-01-19:
I've removed the index.html file from both USBs.  They are the same file.  Now the USB feature on the laptop and on the iPhone are acting exactly as they should.   In the case of USB1 all of the various subfolders can be displayed,opened and explored from the Laptop and iPhone.   I've attached the index.html file.  It is lacking the images and the CSS formatting.  I had to send it as index.txt since this feature on GitHub doesnt allow HTML to be upload
[index.txt](https://github.com/iiab/iiab/files/7893940/index.txt)

holta, 2022-01-19:
@jackrodonnell great progress.

Can you fix up an index.html customized for each USB stick — that is if your goal is a prettier web page — to showcase the PDF(s) and video files (etc) on each stick?

jackrodonnell, 2022-01-19:
@holta
Hi Adam... you suggested I delete the index.html file from the USB which did successfully me to view the other contents of the USB(a series of folders containing HTML, CSS, image, PDFs, video etc) The content are the folders for an educational digital archive. I index.html file I deleted is in fact "an index.html customized for each USB stick ........to showcase the PDF(s) and video files (etc) on each stick"
[image]

jackrodonnell, 2022-01-20:
@holta @jvonau @tim-moody 
I took the same Lenovo T440p laptop and installed Mint Mate 20.3 and then reinstalled a minimal package of IIAB. I ran "gsettings set org.mate.media-handling automount false"   The results with attached USBs were identical to what I saw in Ubuntu Mate 20.04. I experimented with making folders and testing the behavior of file types. Results on server and client are the same
I'm convince that the issue is connected with .html file types.  If a folder contains an Index.html file, the USB0 displays on IIAB but when the folder is opened it appears empty of all contents.

I took some screen shots with descriptions,  Picture are numbered 1-4
[image]
[image]
[image]
[image]

holta, 2022-01-20:
@jackrodonnell call you call me so we can get to the bottom of this efficiently?

Your index.html displays without problem for me &mdash; e.g. when placed on a USB stick &mdash; as you can see in the 2 screenshots below:

[image]

(When index.html is renamed to main.html, it is also appears in browser, as seen below...)

[image]

jvonau, 2022-01-21:
```
COMMAND: /usr/sbin/blkid    # Mount point details
/dev/sda5: UUID="386a0fa4-2a42-4939-805a-48c9b87c4361" TYPE="ext4" PARTUUID="3ce9a89c-05"
/dev/loop0: TYPE="squashfs"
/dev/loop1: TYPE="squashfs"
/dev/loop2: TYPE="squashfs"
/dev/loop3: TYPE="squashfs"
/dev/loop4: TYPE="squashfs"
/dev/loop5: TYPE="squashfs"
/dev/loop6: TYPE="squashfs"
/dev/loop7: TYPE="squashfs"
/dev/sda1: UUID="9FF4-0A75" TYPE="vfat" PARTUUID="3ce9a89c-01"
/dev/loop8: TYPE="squashfs"
/dev/loop9: TYPE="squashfs"
/dev/loop10: TYPE="squashfs"
/dev/sdb1: LABEL="LaptopImage" UUID="1ddf9f13-64af-40ae-b4f9-91642584249c" TYPE="ext4" PARTUUID="eebe4afc-01"

```
Think you are running into a filesystem type/permissions issue, vfat does not support the needed file ownership that the webserver requires.

jackrodonnell, 2022-01-21:
@holta @jvonau @tim-moody I discovered the source of my problems.  I had copied a web site that I have created and have used to the USB.  This kept giving the errors I've outlined over the last few days.  After yesterday post I recopied the site to the same USB.  Everything now works as expected.

I retrieved the old files from my Trash and replicated the errors I once saw.  When I inspected the index.html page from the Trash version found it to be totally blank

How this happened is a mystery,  I appreciate your help and regret the time I've taken from an error I caused

holta, 2022-01-21:
Eureka, thanks @jackrodonnell for updating us!

As final clarification, regarding @jvonau's theory from about 10h ago:

> vfat does not support the needed file ownership that the webserver requires

I considered that idea.

This theory doesn't hold up however.

Specifically: my test (screenshots above, 14h ago) used a VFAT USB stick, and @jackrodonnell's earlier index.html (renamed from his index.txt above, 58h ago) appeared at http://box/usb/USB0 without any problems.

holta, 2022-01-21:
(Actually let's leave this ticket open for now, to help drive awareness of the Graphical Desktop double-mounting issue, even if there's not magical answer to deal with all such Visual/Graphical Desktop OS's as this time ;-)

jvonau, 2022-01-21:
I suggest that this issue be 'pinned'

jvonau, 2022-01-21:
> Eureka, thanks @jackrodonnell for updating us!
> 
> As final clarification, regarding @jvonau's theory from about 10h ago:
> 
> > vfat does not support the needed file ownership that the webserver requires
> 
> I considered that idea.
> 
> This theory doesn't hold up however.
> 
> Specifically: my tests (screenshots above, 14h ago) used a VFAT USB stick, and @jackrodonnell's earlier index.html (renamed from his index.txt above, 58h ago) appeared successfully at http://box/usb/USB0 without problems.

How the usb stick is formatted (ie vfat, exfat, ntfs) does have an effect on what you can serve from the usbstick
https://en.wikipedia.org/wiki/Comparison_of_file_systems#OS_support

holta, 2022-01-21:
> I suggest that this issue be 'pinned'

Done, at the top of: https://github.com/iiab/iiab/issues

This OpenStreetMap issue was unpinned for now, to make room:

- https://github.com/iiab/iiab/issues/3077

holta, 2022-01-21:
> How the usb stick is formatted (ie vfat, exfat, ntfs) does have an effect on what you can serve from the usbstick https://en.wikipedia.org/wiki/Comparison_of_file_systems#OS_support

Is there an example of some files working with IIAB (while some other files are not working with IIAB) when using the very same USB stick/drive filesystem?

(In this case with @jackrodonnell, that was not the issue.)

jvonau, 2022-01-22:
vfat holding a zim that is larger that 4.3 GB comes to the top of my head as an example

jvonau, 2022-01-28:
> Yes a call would be fine 773-316-2944 Tues-Fri best Jack 
On 1/17/22 1:49 PM, A Holt wrote: Thanks @jackrodonnell for pushing. 

@jackrodonnell Do I have the right person https://www.whitepages.com/phone/1-773-316-2944 of Somerville, MA using the https://twitter.com/jackrodonnell handle from https://worldcomputerexchange.org/? I'll be trying to contact you directly later today.

jackrodonnell, 2022-01-28:
May not work;I'll be driving all day.  
Jack

jvonau, 2022-01-28:
Email me directly your preferred contact info.

jackrodonnell, 2022-01-29:
Jerry

The best contact is either ***@***.*** or 773-316-2944 via text or WhatsApp
Jack

jvonau, 2022-04-05:
close

closed 2022-04-05 by holta