iiab/iiab #3106: admin-console - No space left on drive using preset

jvonau, 2022-01-19:
### Expected behavior and actual behavior:
admin-console presets checks for available space before installing
### Steps to reproduce the problem:
Try to use preset without checking free space available first
### IIAB version, OS version, hardware details:
current master Ubuntu 20.04.3 LTS vps
### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
http://sprunge.us/YJuXUa

jvonau, 2022-01-19:
[image]
[image]

holta, 2022-01-20:
Good question.  As was quickly mentioned on today's IIAB community call (http://minutes.iiab.io)

andrewclark5, 2022-05-27:
This caused Linux Mint 20.3 to become unstable - kept refreshing the Desktop.

This issue seems to be related to disk space.

But in my case the 16GB SSD running the O/S and iiab had 2.2GB free space left on it.

However there was also an external USB carrying the library files which had 0 bytes left on it.  But I don't see why after freeing up some space the issue was still there after a reboot?

[image]

jvonau, 2022-05-27:
>This caused Linux Mint 20.3 to become unstable - kept refreshing the Desktop.

Can you expand on that a little? I'm unsure of what 'refreshing' is meant to refer to. Log in and you are brought back to the login screen?
>This issue seems to be related to disk space.

To have a better idea what we are looking at could you post the output of `dh -m` or the web link produced by iiab-diagnostic?

holta, 2022-05-27:
> web link produced by [iiab-diagnostics]

Indeed, Thank You @andrewclark5 if you can please run [sudo iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) and then reply with the **sprunge.us** URL it generates.

tim-moody, 2022-05-27:
iiab-diagnostics would be especially useful.

Admin Console fails to start because the command GET-SPACE-INFO fails. Perhaps the device with library has become corrupt.

andrewclark5, 2022-05-28:
Hi Guys

Thanks to all three of you for the quick response.

I thought I had responded to these questions yesterday but for some reason my post is not appearing here :(

I explained that under time pressures to get this working for an IIAB demo this week I had to wipe the disk and start afresh.

However I did have the presence of mind to capture the iiab-diagnostics output to an external disk and it includes the pastebin:-

Publish it to a web pastebin? [Y/n] Y

PUBLISHING TO URL... http://sprunge.us/LQd2pn 

[IIAB-Diagnostics.txt](https://github.com/iiab/iiab/files/8791006/IIAB-Diagnostics.txt)

Shanti and I have surmised that the issue was caused by the external drive running out of space, yet even when I made space on the USB drive and rebooted it did not resolve the issue.  I don't think the drive is corrupted as I can see all the module files.

The Mint Desktop kept going black then reappearing with a message offering to disable widgets etc and was completely unusable.

At the time I was trying to enable an external library on the 32GB USB for IIAB and it had 28.8GB of files on it.  However it showed 0 bytes available, (presumably due to slack space from a lot of small files).  So it may be due to what I was doing, as per the FAQ here:-

https://wiki.iiab.io/go/FAQ#Can_I_permanently_attach_an_external_USB_drive,_to_add_more_content? 

Hope that helps to clarify the issue.

Andrew

tim-moody, 2022-05-28:
>I had to wipe the disk and start afresh

and now have a working iiab?

from the diagnostics I see:

```
COMMAND: /usr/bin/df -h    # Disk usage

Filesystem      Size  Used Avail Use% Mounted on
udev            1.8G     0  1.8G   0% /dev
tmpfs           363M  3.1M  360M   1% /run
/dev/sda5        58G   18G   38G  32% /
tmpfs           1.8G     0  1.8G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           1.8G     0  1.8G   0% /sys/fs/cgroup
/dev/sda1       511M  4.0K  511M   1% /boot/efi
tmpfs           363M   20K  363M   1% /run/user/1000
/dev/sdb1        29G  9.6G   18G  36% /media/usb0
```

assuming you followed the FAQ you cite that should mean you had
/ and /library on /dev/sda5
and /library-ext on /dev/sdb1 with a symlink to /library/www/html/modules
This would be a total space of approx 90G, which is less than the 128G indicated for a preset. Moreover, if the preset installed any zims, they would go into /library, not modules, so freeing space on /library-ext would not solve the problem, and since / and library share the same device, when library is full / is full too.

andrewclark5, 2022-05-28:
Hi Tim

I'm afraid this Newby to Linux is finding this procedure difficult to follow...

But I have now had the same error generate on booting up another identical machine T620-2 with an identical IIAB image on it (cloned using Foxclone) and grabbed photo of it (attached)
[image]

Also working on my original T620 and following the instructions on the FAQ to link an external drive I now get this on rebooting 
[image]

Does that concur with your diagnosis above?

andrewclark5, 2022-05-28:
>assuming you followed the FAQ you cite that should mean you had
/ and /library on /dev/sda5
**Yes, I think so**
and /library-ext on /dev/sdb1 with a symlink to /library/www/html/modules
**I have library-ext on /dev/sdb containing all the library files as a tree copied from a working IIAB install**
**Thus**
user1@box:~$ cd /library-ext
user1@box:/library-ext$ ls
library-ext  lost+found  usb0
user1@box:/library-ext$ ls library-ext
awstats      downloads  kolibri  wordpress  www
calibre-web  ka-lite    public   working    zims
user1@box:/library-ext$ 

This would be a total space of approx 90G, which is less than the 128G indicated for a preset. Moreover, if the preset installed any zims, they would go into /library, not modules, so freeing space on /library-ext would not solve the problem, and since / and library share the same device, when library is full / is full too.

Sorry - not sure I follow this bit!

tim-moody, 2022-05-28:
> finding this procedure difficult to follow...

not clear what procedure

>cloned using Foxclone

no experience with that, but adds complexity to an already complex implementation

>Does that concur with your diagnosis above?

no. this may or may not be related

>Sorry - not sure I follow this bit!

I'm afraid I have the same problem.

tim-moody, 2022-05-28:
>I have library-ext on /dev/sdb containing all the library files as a tree copied from a working IIAB install

OK now I follow

/library-ext doesn't replace /library, it extends it via the symlink  so there is no reason to copy anything but modules there

tim-moody, 2022-05-28:
if there is no /library then a lot of things won't work, but you haven't said what's on /library

andrewclark5, 2022-05-28:
>if there is no /library then a lot of things won't work, but you haven't said what's on /library
The default /library files are still there.  I was hoping that I could just point IIAB at a new library location but from what you say it is probably seeing them both so there will be duplication of some modules.

jvonau, 2022-05-28:
a78fc546d66f8d23b9aefdf100444126b390250b is branch 7.2 is that what is wanted?

andrewclark5, 2022-05-28:
It looks as if the link is not set up properly
[image]

andrewclark5, 2022-05-28:
> [a78fc54](https://github.com/iiab/iiab/commit/a78fc546d66f8d23b9aefdf100444126b390250b) is branch 7.2 is that what is wanted?

No.  I installed IIAB 8.0

jvonau, 2022-05-28:
> > [a78fc54](https://github.com/iiab/iiab/commit/a78fc546d66f8d23b9aefdf100444126b390250b) is branch 7.2 is that what is wanted?
> 
> No. I installed IIAB 8.0

No you didn't.

holta, 2022-05-28:
> No. I installed IIAB 8.0

1) Good choice.  This will bring 5 months of important fixes and improvements.

2) Please run **iiab-diagnostics** on the new build, and send a new **sprunge.us** URL?  Thanks Andrew!

jvonau, 2022-05-28:
> It looks as if the link is not set up properly 

Did you disable the desktop auto-mount feature?

tim-moody, 2022-05-28:
I think there has been a misunderstanding of https://wiki.iiab.io/go/FAQ#Can_I_permanently_attach_an_external_USB_drive,_to_add_more_content?

This extends, not replaces /library

I'm also confused by the message that /dev/sdb is already mounted at /library-ext

I would have expected it to be /dev/sdb1

andrewclark5, 2022-05-28:
> > No. I installed IIAB 8.0
> 
>     1. Good choice.  This will bring 5 months of important fixes and improvements.
> 
>     2. Please run **sudo iiab-diagnostics** on the new build, and reply with a new **sprunge.us** URL?  Thanks Andrew!

OMPLETE! Your diagnostics file (112K, 1937 lines) is:

   /etc/iiab/diag/8.0_linuxmint-20_2022-05-28_15:24:28_BST_andrewc

Publish it to a web pastebin? [Y/n] y

PUBLISHING TO URL... http://sprunge.us/wiR1Vk

andrewclark5, 2022-05-28:
> > > [a78fc54](https://github.com/iiab/iiab/commit/a78fc546d66f8d23b9aefdf100444126b390250b) is branch 7.2 is that what is wanted?
> > 
> > 
> > No. I installed IIAB 8.0
> 
> No you didn't.
Well, I thought i did.  Does this not suggest so...
/etc/iiab/diag/8.0_linuxmint-20_2022-05-28_15:24:28_BST_andrewc

holta, 2022-05-28:
From `iiab-diagnostics`, the output of `df -h` shows both disks as nearly full — concerning!

 ```
Filesystem      Size  Used Avail Use% Mounted on
udev            1.6G     0  1.6G   0% /dev
tmpfs           336M  2.9M  333M   1% /run
/dev/sda5        15G   12G  1.9G  87% /
tmpfs           1.7G     0  1.7G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           1.7G     0  1.7G   0% /sys/fs/cgroup
/dev/sda1       511M  4.0K  511M   1% /boot/efi
tmpfs           336M   20K  336M   1% /run/user/1000
/dev/sdb         29G   26G  1.5G  95% /media/usb0

jvonau, 2022-05-28:
> PUBLISHING TO URL... http://sprunge.us/wiR1Vk

That looks better
```
[runtime]
iiab_stage = 0
iiab_base_ver = 8.0
iiab_revision = 0
runtime_branch = master
runtime_commit = 858895c50b5b9716378a9ddc7bd4a9669f7a2153
runtime_date = 2022-05-27T08:19:39Z
```

The "running in fallback mode" is the lack of 3D support within the video subsystem that 'cinnamon' requires. The picture of the failure "unable to mount IIAB-LIBRARY-EXT" above is the desktop auto-mount failing, that is sort of normal and should be disabled. Need smoke and a coffee but in the mean time how is '/library-ext' being mounted? Via an entry in /etc/fstab?

andrewclark5, 2022-05-28:
Yes.  These little thin client devices are designed to have a minimal O/S on their 16GB SSD.  Mint and IIAB just fit with 1.9GB to spare, and more when I eventually delete all the modules in the default build of /library.  We have around 400 of them so we really want to make these work as IIAB servers, with an internal USB drive of up to 128GB providing the library.
The instructions in the FAQ did not mention mounting the USB drive - which surprised me, just the LABEL command line which I copied and pasted in fstab.
Where do I go to disable the automount facility?
And does the ln -s command persist over a reboot?

holta, 2022-05-28:
> We have around 400 of them so we really want to make these work as IIAB servers

@andrewclark5 are you aware that today's http://sprunge.us/wiR1Vk?en shows no Wi-Fi hardware in the machine?

(How exactly are you intending student machines to connect their devices to IIAB?)

holta, 2022-05-28:
> The instructions in the FAQ did not mention mounting the USB drive - which surprised me, just the LABEL command line which I copied and pasted in fstab.

Please read about `/etc/fstab` — that's precisely its purpose (to mount filesystems).

> Where do I go to disable the automount facility?

Linux Mint documentation.  Or talk to @shanti-bhardwa if he has more Linux Mint experience in this area?

> And does the ln -s command persist over a reboot?

Yes.

Carefully set up your symlink(s) and verify they're pointing in the right place, by running:

```
ls -l /library

jvonau, 2022-05-28:
```
=IIAB==========================================================================
COMMAND: /usr/bin/lsblk    # Partition mount points

NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda      8:0    0 14.9G  0 disk 
├─sda1   8:1    0  512M  0 part /boot/efi
├─sda2   8:2    0    1K  0 part 
└─sda5   8:5    0 14.4G  0 part /
sdb      8:16   1 28.7G  0 disk /media/usb0

=IIAB==========================================================================
COMMAND: /usr/sbin/blkid    # Mount point details

/dev/sda1: UUID="59C8-75C8" TYPE="vfat" PARTUUID="9cc66386-01"
/dev/sda5: UUID="b1859720-a50b-4db9-a02e-cd629bf9fa75" TYPE="ext4" PARTUUID="9cc66386-05"
/dev/sdb: LABEL="IIAB-LIBRARY-EXT" UUID="a4a26931-7897-440c-8b1a-38e1e26897a4" TYPE="ext4"
```

Interesting that the usb device at /dev/sdb is not partitioned, not the end of the world just not a standard layout.
>The instructions in the FAQ did not mention mounting the USB drive - which surprised me, just the LABEL command line which I copied and pasted in fstab.

Could you post the complete output of `mount` there should be two entries for /dev/sdb, '/ext-library' and '/media/USB0'

>Where do I go to disable the automount facility?

No clue, don't use 'cinnamon', search engines are your friend.
>And does the ln -s command persist over a reboot?

Yes

holta, 2022-05-28:
@andrewclark5 Wikipedia has quite a good article about `/etc/fstab` that you or @shanti-bhardwa should consider reading quickly, to help you get oriented here:

https://en.wikipedia.org/wiki/Fstab

andrewclark5, 2022-05-28:
> > We have around 400 of them so we really want to make these work as IIAB servers
> 
> @andrewclark5 are you aware that today's http://sprunge.us/wiR1Vk?en shows no Wi-Fi hardware in the machine?
> 
> (How exactly are you intending student machines to connect their devices to IIAB?)

We are installing wired networks in all classrooms.  (150 Malawi schools to date with labs installed but not all have networks yet).

holta, 2022-05-28:
> > Where do I go to disable the automount facility?
> 
> Linux Mint documentation. Or talk to @shanti-bhardwa if he has more Linux Mint experience in this area?

As Jerry hinted, Cinnamon is a bit overweight for such "thin clients" being repurposed as thin servers.

But if you stick with Mint's Cinnamon, its "NEMO" file manager >> click 'edit' at the top >> select 'preferences' at the bottom of the drop down menu ...apparently has a setting to turn off "Automatically mount removable media when inserted and on startup" :

[image]

(According to the [Mint 19 commentariat](https://unix.stackexchange.com/a/469847) anyway :-)

andrewclark5, 2022-05-28:
@jvonau 
user1@box:/$ mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
udev on /dev type devtmpfs (rw,nosuid,noexec,relatime,size=1672584k,nr_inodes=418146,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,noexec,relatime,size=343872k,mode=755)
/dev/sda5 on / type ext4 (rw,relatime,errors=remount-ro)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
pstore on /sys/fs/pstore type pstore (rw,nosuid,nodev,noexec,relatime)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/hugetlb type cgroup (rw,nosuid,nodev,noexec,relatime,hugetlb)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/memory type cgroup (rw,nosuid,nodev,noexec,relatime,memory)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/rdma type cgroup (rw,nosuid,nodev,noexec,relatime,rdma)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=28,pgrp=1,timeout=0,minproto=5,maxproto=5,direct,pipe_ino=17722)
mqueue on /dev/mqueue type mqueue (rw,nosuid,nodev,noexec,relatime)
hugetlbfs on /dev/hugepages type hugetlbfs (rw,relatime,pagesize=2M)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,relatime)
tracefs on /sys/kernel/tracing type tracefs (rw,nosuid,nodev,noexec,relatime)
configfs on /sys/kernel/config type configfs (rw,nosuid,nodev,noexec,relatime)
fusectl on /sys/fs/fuse/connections type fusectl (rw,nosuid,nodev,noexec,relatime)
/dev/sda1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=343868k,mode=700,uid=1000,gid=1000)
gvfsd-fuse on /run/user/1000/gvfs type fuse.gvfsd-fuse (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)
/dev/sdb on /library-ext type ext4 (rw,relatime)
/dev/sdb on /media/usb0 type ext4 (rw,nodev,noexec,noatime,nodiratime)
portal on /run/user/1000/doc type fuse.portal (rw,nosuid,nodev,relatime,user_id=1000,group_id=1000)

andrewclark5, 2022-05-28:
> @andrewclark5 Wikipedia has quite a good article about `/etc/fstab` that you or @shanti-bhardwa should consider reading quickly, to help you get oriented here:
> 
> https://en.wikipedia.org/wiki/Fstab

Excellent.  Thanks for that.

One thing I notice in our default fstab is that drives and partitions are referenced by UUIDs.  Since devices can change according to when they are plugged in would it be a good idea to adopt that?  Here is my fstab as modified in line with the FAQ to add the last line
*****************************
# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
# / was on /dev/sda5 during installation
UUID=b1859720-a50b-4db9-a02e-cd629bf9fa75 /               ext4    errors=remount-ro 0       1
# /boot/efi was on /dev/sda1 during installation
UUID=59C8-75C8  /boot/efi       vfat    umask=0077      0       1
/swapfile                                 none            swap    sw              0       0
#For External IIAB library
LABEL=IIAB-LIBRARY-EXT  /library-ext  ext4  defaults,nofail  0  0
*******************************

andrewclark5, 2022-05-28:
Rats.  The "#" in the fstab have caused formatting problems in the post above.   Sorry about that.

andrewclark5, 2022-05-28:
@holta
> As Jerry hinted, Cinnamon is a bit overweight for such "thin clients" being repurposed as thin servers.
>
> But if you stick with Mint's Cinnamon, its "NEMO" file manager >> click 'edit' at the top >> [...]

Indeed it does.  I have unchecked that box and we will see what happens at the next reboot.
So, apart from that setting which command is mounting the USB beforehand?

tim-moody, 2022-05-28:
https://github.com/iiab/iiab-admin-console/pull/490 just warn for now. This is relevant to the original issue, but not necessarily to recent postings.

andrewclark5, 2022-05-28:
@jvonau 

>Interesting that the usb device at /dev/sdb is not partitioned, not the end of the world just not a standard layout.
Well spotted. I had not noticed that.  Would it be better to create a partition?

tim-moody, 2022-05-28:
>LABEL=IIAB-LIBRARY-EXT /library-ext

just mounting on /library is easier than mounting on /library-ext and then having to create a symlink from /library to it. We used to do this on OLPC XOs that also had small rootfs devices.

holta, 2022-05-28:
> Since devices can change according to when they are plugged in would it be a good idea to adopt that?

Personally I would not.  (A LABEL is human-readable, making preparation and management more foolproof/failsafe in my opinion.)

> So, apart from that setting which command is mounting the USB beforehand?

OS internals do not much matter in this situation.  What matters is that `/etc/fstab` is built into every Linux/UNIX to take care of that during OS boot-up especially (as the Wikipedia article explains well).

PS I personally recommend mounting the entire `/library` as @tim-moody advises (avoiding all `ln -s` symlinks, bonus) when your internal/main disks are puny (e.g. 16GB) as appears to be the case!

jvonau, 2022-05-28:
> > Since devices can change according to when they are plugged in would it be a good idea to adopt that?
> 
> Personally I would not. (A LABEL is human-readable, making preparation and management more foolproof/failsafe in my opinion.)
>

It's a bit more than that, each copy of the usb-drive could have the same label but each copy would have a different uuid, makes using the uuid a real PITA when mass duplication is the goal.
 
> OS internals do not much matter in this situation. What matters is that `/etc/fstab` is built into every Linux/UNIX to take care of that during OS boot-up especially (as the Wikipedia article [...]

Much more straight forward but the fstab work should be done before starting the IIAB installation. Now when cloning there are 2 parts to the clone, the SSD for the OS and the usb-drive for the Data.

tim-moody, 2022-05-28:
>Much more straight forward but the fstab work should be done before starting the IIAB installation. Now when cloning there are 2 parts to the clone, the SSD for the OS and the usb-drive for the Data.

Completely agree, except the usb-drive can now be a copy instead of a clone.

holta, 2022-05-28:
> the fstab work should be done before starting the IIAB installation

Definitely this is the easiest way, Yes!

tim-moody, 2022-05-28:
and if you want to get fancy you can have a dummy /library on the sd card and mark the mounting of the drive in fstab no fail so the boot doesn't hang if the drive is missing.

jvonau, 2022-05-28:
> > Much more straight forward but the fstab work should be done before starting the IIAB installation. Now when cloning there are 2 parts to the clone, the SSD for the OS and the usb-drive for the Data.
> 
> Completely agree, except the usb-drive can now be a copy instead of a clone.

As long as one re-labels the new drive you can use copying in place of using `dd`

> and if you want to get fancy you can have a dummy /library on the sd card and mark the mounting of the drive in fstab no fail so the boot doesn't hang if the drive is missing.

The creation of '/library' on the device that booted must be done, that will be the mounting point that fstab will use and must exist. 

@andrewclark5 The fstab line that was used is fine, just change  '/library-ext' to be just '/library' and ensure /library is created before starting the IIAB installation.

jvonau, 2022-05-28:
The existing instructions are geared more for when you have ran out of space and don't want to start completely over because one could be looking at having way more data around, like hitting the limit on a 256GB storage device. @holta update for new instructions to match this thread.

tim-moody, 2022-05-28:
>The existing instructions are geared more for when you have ran out of space and don't want to start completely over because one could be looking at having way more data around, like hitting the limit on a 256GB storage device. @holta update for new instructions to match this thread.

>I think there has been a misunderstanding of https://wiki.iiab.io/go/FAQ#Can_I_permanently_attach_an_external_USB_drive,_to_add_more_content?

as I said 6 hours ago.

jvonau, 2022-05-28:
> from mount above
/dev/sdb on /library-ext type ext4 (rw,relatime)
/dev/sdb on /media/usb0 type ext4 (rw,nodev,noexec,noatime,nodiratime)

> COMMAND: /usr/bin/lsblk    # Partition mount points
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda      8:0    0 14.9G  0 disk 
├─sda1   8:1    0  512M  0 part /boot/efi
├─sda2   8:2    0    1K  0 part 
└─sda5   8:5    0 14.4G  0 part /
sdb      8:16   1 28.7G  0 disk /media/usb0

@tim-moody Note that `lsblk` only shows the last mount point out of the 2 that were performed, one by fstab and the other by usbmount. That might cause issues as noted in #3121 and is used in [get_storage_info](https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/files/iiab-cmdsrv3.py#L1535) that might be causing the function to fail like above.

andrewclark5, 2022-05-28:
OK.  So if I adopt the suggested strategy of making the USB /library can someone kindly spell out all the steps I now need to take to make it work that way (bearing in mind I know nothing about the internal workings of IIAB and how it points to where to find its data!).

jvonau, 2022-05-28:
fresh install no IIAB yet:
`sudo mkdir /library`
edit fstab 
reboot
check that /library was mounted.

holta, 2022-05-28:
> The existing instructions are geared more for when you have ran out of space and don't want to start completely over because one could be looking at having way more data around, like hitting the limit on a 256GB storage device. @holta update for new instructions to match this thread.

I agree that explaining both scenarios is likely something we need to add to http://FAQ.IIAB.IO

People are looking for a "cut + paste" solution naturally...and we can offer a few "quick + dirty" example options for the impatient.

_(But in the end they also have to understand https://en.wikipedia.org/wiki/Fstab if they want a resilient solution that does not do a disservice to their schools!)_

andrewclark5, 2022-05-28:
>`fresh install no IIAB yet:

OK.  No problem I can try that tomorrow

>sudo mkdir /library
Where is that to be created?  On the USB or in the root File System

>edit fstab
To put in what?

reboot
check that /library was mounted.`

Sorry to be so pedantic guys but if I don't understand what I'm doing I need idiot-proof instructions!

jvonau, 2022-05-28:
> > `fresh install no IIAB yet:
> 
> OK. No problem I can try that tomorrow
> 
> > sudo mkdir /library
> > Where is that to be created?  On the USB or in the root File System
>

On the root file system....
> > edit fstab
> > To put in what?
>

The same entry you used before except use /library and not /library-ext.
> reboot check that /library was mounted.`
> 
> Sorry to be so pedantic guys but if I don't understand what I'm doing I need idiot-proof instructions!

That is what a retainer provides, knowledgeable expert help. Remember a search engine can be your friend when you need to learn.

andrewclark5, 2022-05-28:
OK Guys.  It's getting late here in Edinburgh and my brain is fading.  I'll give it a go and report back to you.
Thanks for all your assistance in this Voyage!

jvonau, 2022-05-28:
cheat sheet

Edit /etc/fstab `sudo nano /etc/fstab`
add this line: LABEL=IIAB-LIBRARY-EXT  /library  ext4  defaults,nofail  0  0
save the file with `ctrl o` and exit with `ctrl w`
insert the usb-drive
`sudo mount -a`
check if /dev/sdb was mounted with 'sudo mount' should be shown similar to as above  '/dev/sdb on /library type ext4 (rw,relatime)'
shutdown/power off the machine, remove the usb-drive, boot the machine, after booted insert the usb-drive and check if it was mounted correctly with 'sudo mount'
reboot with the usb-drive inserted and recheck that /library was auto-mounted on boot with 'sudo mount'

andrewclark5, 2022-05-28:
@jvonau 
Thanks Jerry!  

@tim-moody 
https://www.foxclone.com/ Brilliant little tool to make Linux system disk images and clone disks to build lots of machines very quickly.
I reckon it takes about 2 hrs 20mins to fully build and test a Mint/IIAB system.  Once built and cloned I can copy it to another identical machine in around 5 - 10 minutes.  (This is necessary for a production line such as we operate at the Turing Trust - we send around 1000 PCs or laptops with each container shipment).

andrewclark5, 2022-05-28:
Just realised if I can leave this installing IIAB overnight it will be ready by morning... :(
> Edit /etc/fstab `sudo nano /etc/fstab` add this line: LABEL=IIAB-LIBRARY-EXT /library ext4 defaults,nofail 0 0 save the file with >`ctrl o` and exit with `ctrl w` insert the usb-drive `sudo mount [...]
```
Result:
`user1@HP-t620-Dual-Core-TC-2:~$ sudo cp /etc/fstab fstab.bac
[sudo] password for user1:  
Edited fstab and pasted LABEL=line
user1@HP-t620-Dual-Core-TC-2:~$ sudo nano /etc/fstab
user1@HP-t620-Dual-Core-TC-2:~$ sudo mount -a
mount: /library: mount point does not exist.
```
Used another machine to change USB folder name to "library"
still no joy

 shutdown/power off the machine, remove the usb-drive, boot the machine, after booted insert the usb-drive and check if it was mounted correctly with 'sudo mount' reboot with the usb-drive inserted and recheck that /library was auto-mounted on boot with 'sudo mount'

andrewclark5, 2022-05-28:
Rebooted.
Nearly there... I can see library now in Filesystem.

Only problem - it has a library folder UNDER library folder - and then all my required folders/files under that!

What went wrong here...?

holta, 2022-05-28:
@andrewclark5 are you familiar with the `lsblk` and `blkid` commands?

Both can be incredible useful &mdash; I recommend you get to know both &mdash; to help you tailor `/etc/fstab` to your exact needs.

holta, 2022-05-28:
> Rebooted. Nearly there... I can see library now in Filesystem.
> 
> Only problem - it has a library folder UNDER library folder - and then all my required folders/files under that!
> 
> What went wrong here...?

Right!  Two choices:

1) Do away with the /library directory on your USB drives.

2) Create a symlink from the main drive's /library to the USB drive's subservient /library directory &mdash; similar to the example provided at https://wiki.iiab.io/go/FAQ#Can_I_permanently_attach_an_external_USB_drive,_to_add_more_content%3F

[127 more replies]

closed 2022-05-29 by jvonau