iiab/iiab #3109: Can we hide the "Power Off" button when it's not meant to work?

holta, 2022-01-21:
Might there be any pragmatic way that IIAB could auto-hide or obfuscate the "Power Off" button when it's not meant to work?

e.g. when IIAB implementers wants the URL http://box/js-menu/menu-files/services/power_off.php fully disabled &mdash; to avoid students abusing this.

Context: I just did a standard fresh install of IIAB, taking all defaults, and the "Power Off" button appears on the bottom-left of the resulting IIAB page (at http://box.lan).  It really shouldn't, in these circumstances, if we can possibly find a way for it not to appear in these common cases?

Background: `apache_allow_sudo: False` is set as a result of these changes discussed (http://minutes.iiab.io) earlier today:

- PR #3104
- PR iiab/iiab-admin-console#458

tim-moody, 2022-01-21:
just turn it off in menu properties

holta, 2022-01-21:
The question is how to auto-hide it.

(But possibly that's not easy.)

holta, 2022-01-21:
Definitely good enough for now:

- https://github.com/iiab/iiab-admin-console/pull/459

Thanks @tim-moody

closed 2022-01-21 by holta