iiab/iiab #3125: Regression: Using preset does not install zims defined in content.json

jvonau, 2022-02-23:
### Expected behavior and actual behavior:
Install what zim files are listed in content.json for en-medical when using preset, the zim jobs are not present in "display job status" only 26 jobs as opposed to 32. the 6 zim jobs when are not included in the listing. Prior installs of iiab-admin-console at efaf6e4194dd921b016d3285e23cce1da428a320 installed the zims.
### Steps to reproduce the problem:
use preset, note if zims are installed.
### IIAB version, OS version, hardware details:
latest master

jvonau, 2022-02-23:
There is something wrong with the pick list, the older commit shows
[image]
The newer version is lacking the sub-headings shows just with nothing current.
[image]

jvonau, 2022-02-23:
Might be the cause of #3124?

jvonau, 2022-02-23:
```
Downloading Catalogs and Building Local Data Files.
Traceback (most recent call last):
  File "/opt/admin/cmdsrv/scripts/get_kiwix_catalog", line 11, in <module>
    from bs4 import BeautifulSoup
ModuleNotFoundError: No module named 'bs4'
Getting the Kiwix Catalog Failed.
Please run iiab-get-kiwix-cat again later.
SUCCESSWaiting to run iiab-make-kiwix-lib.py
```

jvonau, 2022-02-23:
`sudo apt install python3-bs4` is the fix

jvonau, 2022-02-23:
https://github.com/iiab/iiab-admin-console/pull/464

jvonau, 2022-02-23:
still unable to install any zims, but the pick list has the sub-menus now.

jvonau, 2022-02-23:
Further to can't install zims, select zim to install: 
[image]
press 'Install Selected Zims'
[image]
observer error flashed to screen with the job not being added to the job queue.
[image]

tim-moody, 2022-02-23:
[image]

jvonau, 2022-02-23:
ran `sudo /opt/admin/cmdsrv/scripts/get_kiwix_catalog -v` then "Refresh Kixix Catalog" and the pick list now works while presets don't populate the pick list.

tim-moody, 2022-02-25:
there is a second bug. url attribute missing from root.xml catalog

jvonau, 2022-02-26:
@holta I gather the statement "Pre-releases of Internet-in-a-Box (IIAB) undergo continuous QA / continuous integration / continuous deployment and are strongly recommended!" is no longer true?

> ```
> Downloading Catalogs and Building Local Data Files.
> Traceback (most recent call last):
>   File "/opt/admin/cmdsrv/scripts/get_kiwix_catalog", line 11, in <module>
>     from bs4 import [...]

The installer should of stopped at this point of the install, when the underlying cause was already known and is fatal but becomes hidden because the bash script did not stop. The underling cause of this behavior is rooted in the lack of use of the '-e' flag within https://github.com/iiab/iiab-admin-console/blob/master/install

tim-moody, 2022-02-26:
>I gather the statement "Pre-releases of Internet-in-a-Box (IIAB) undergo continuous QA / continuous integration / continuous deployment and are strongly recommended!" is no longer true?

which pre-release are we in?

holta, 2022-02-26:
> @holta I gather the statement "Pre-releases of Internet-in-a-Box (IIAB) undergo continuous QA / continuous integration / continuous deployment and are strongly recommended!" is no longer true?

QA is of course a human endeavor, with everyone attempting their best.

This ticket is extremely helpful.

Likewise @tim-moody has made really incredible progress thanks to @bernardroth in recent days:

- #3124

It's a powerful testament to the hard work of QA that some blithely take for granted, but so many others dedicate themselves to contributing to help each other.

holta, 2022-02-26:
> which pre-release are we in?

DRAFT release notes for pre-release IIAB 8.0 are here:

https://github.com/iiab/iiab/wiki/IIAB-8.0-Release-Notes

It's great that 2022 is yielding real progress month by month.

tim-moody, 2022-02-26:
so my understanding of the agreement at the last release was that we now enter a period where master is in flux and those who want stability should install from the most recent release up to the point where a pre-release is declared

holta, 2022-02-26:
There is a lot of vital volunteer-contributed testing of many different kinds, happening almost every single week, and definitely every single month &mdash; and not infrequently every single day.

This brings a new pre-release almost every single week (and sometimes every single day).  Occasionally one of these pre-releases is highlighted to attract even more testing in particular situations, e.g. IIAB 8.0 Beta 1 about a month ago: https://github.com/iiab/iiab/releases/tag/8.0-beta-1

Also occasionally IIAB's master branch contains a particularly nasty/intricate issue that's slipped by all of us.  It happens.  As QA is a human process after all.  The sky does not fall.  And when this occasionally happens, we are all human beings doing our best &mdash; probing + fixing as efficiently as we can in those case.

In essense, the CI/CD isn't perfect (nothing is) but it works extremely well &mdash; and certainly far better than the brittle/bureaucratic machinery of a waterfall release process.

Of course, classical releases _DO_ still have an important outreach/marketing purpose (e.g. to give people something to Tweet about, etc).

tim-moody, 2022-02-27:
This was probably a mistake as a failed oer2go catalog download will now break the entire install, which was the reason I didn't do this in the first place.

holta, 2022-02-27:
> This was probably a mistake as a failed oer2go catalog download will now break the entire install, which was the reason I didn't do this in the first place.

Good point.

If necessary, subtasks of any bash script can be made to appear to succeed &mdash; e.g. using syntax like:

```
program-that-fails || true
```

jvonau, 2022-02-28:
> This was probably a mistake as a failed oer2go catalog download will now break the entire install, which was the reason I didn't do this in the first place.

There can be any number of reasons for a download to fail, as strangeness is noted in #3092. At this point in the 'install' all the support files that admin-console/cmd-server use are in place, the part that is failing is priming the data source as not to fail when GET-OER2GO-STAT or GET-ZIM-STAT is called when entering the admin-console page, but the catalogs can be retrieved using the supplied 'retrieve catalog' buttons, hardly a fatal error but now a PITA requiring user intervention anyway. I'd suggest that the four lines at https://github.com/iiab/iiab-admin-console/blob/master/install#L48-L51 be moved to it's own sub-script as not to have to re-run the entire install for a failed catalog download.

closed 2022-02-28 by jvonau