iiab/iiab #3131: Calibre & essential commands like calibredb are no longer installable on Raspberry Pi OS Bullseye

holta, 2022-03-10:
This has become a much more serious issue since the desktop version of Raspberry Pi OS released stopped including Calibre on 2021-10-31. with Raspberry Pi OS's Bullseye release.

The reason is that schools wanting to organize their book collections can no longer user the calibredb command, e.g. to import folders of books into Calibre-Web etc.

BACKGROUND: With Raspberry Pi OS's prior (Buster) release, the Calibre application was installed and available within the "Raspberry" menu in the top-left of the screen), and more importantly critical commands like calibredb were available at the OS command-line.

_Do schools that care about books now need to avoid Raspberry Pi OS, considering Ubuntu instead?_

Below is the error on the desktop version of Raspberry Pi OS 2022-01-28 &mdash; just in case this is possible resolvable ??

```
root@raspberrypi:~# apt install calibre
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 calibre : Depends: calibre-bin (>= 5.12.0+dfsg-1+deb11u1) but it is not installable
           Depends: python3-pyqt5.qtwebengine (>= 5.12.1-4+b1) but it is not installable
           Recommends: python3-dnspython (>= 1.6.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

holta, 2022-03-11:
Alternative Options:

- 64-bit Raspberry Pi OS (looks good since [2022-01-28](https://www.raspberrypi.com/news/raspberry-pi-os-64-bit/), at long last!)
- 64-bit Ubuntu 21.10 on Raspberry Pi (while it lasts, EOL July 2022!)

See:

https://github.com/iiab/iiab/wiki/IIAB-Platforms#operating-systems

holta, 2022-03-17:
Conclusion: IIAB's roles/calibre Ansible playbook definitely needs to be cleaned up and clarified &mdash; given this changing support within upstream OS's:

https://github.com/iiab/iiab/tree/master/roles/calibre

(In coming weeks or days, if someone has time!)

jvonau, 2022-03-31:
If you are looking to use an 64-bit Ubuntu desktop I suggest [mate](https://releases.ubuntu-mate.org/impish/arm64/ubuntu-mate-21.10-beta1-desktop-arm64+raspi.img.xz), should be able to do an in place upgrade from 21.10 to 22.04 when released.

holta, 2022-03-31:
> If you are looking to use an 64-bit Ubuntu desktop I suggest [mate](https://releases.ubuntu-mate.org/impish/arm64/ubuntu-mate-21.10-beta1-desktop-arm64+raspi.img.xz), should be able to do an in place upgrade from 21.10 to 22.04 when released.

Its very brief changelog announcement from 2021-11-06 is here:
https://ubuntu-mate.org/raspberry-pi/changelog/#6-november-2021---2110

jvonau, 2022-04-07:
Have you filed an up stream bug about the broken apt repo?

holta, 2022-04-07:
> Have you filed an up stream bug about the broken apt repo?

I have not.  After half a year now, it seems the Calibre community on RasPiOS is likely moving on to 64-bit RasPiOS.

jvonau, 2022-04-08:
Use the same stop gap measure as was tried in the past, introduce the equivalent debian armhf repo into the apt mix. It's very risky and should be a one way upgrade, leave the repo in place.

holta, 2022-05-23:
Closing this seems best.

64-bit Raspberry Pi OS is the pragmatic (and very effective) solution.

closed 2022-05-23 by holta