iiab/iiab #3132: Lokole won't install with Python 3.9 on 64-bit RasPiOS on Raspberry Pi 4

holta, 2022-03-11:
Whether or not this is easily fixable, just FYI here's the error:

> TASK [lokole : Create Lokole admin user with password, for http://box/lokole] ******************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "while read envvar; do export \"$envvar\"; done < /home/lokole/state/settings.env\n/library/lokole/venv/bin/manage.py createadmin --name='admin' --password='changeme'\n", "delta": "0:00:00.473349", "end": "2022-03-10 20:24:36.573274", "msg": "non-zero return code", "rc": 1, "start": "2022-03-10 20:24:36.099925", "stderr": "Traceback (most recent call last):\n  File \"/library/lokole/venv/bin/manage.py\", line 9, in <module>\n    from flask_migrate import MigrateCommand\n  File \"/library/lokole/venv/lib/python3.9/site-packages/flask_migrate/__init__.py\", line 6, in <module>\n    from flask import current_app\n  File \"/library/lokole/venv/lib/python3.9/site-packages/flask/__init__.py\", line 19, in <module>\n    from . import json\n  File \"/library/lokole/venv/lib/python3.9/site-packages/flask/json/__init__.py\", line 15, in <module>\n    from itsdangerous import json as _json\nImportError: cannot import name 'json' from 'itsdangerous' (/library/lokole/venv/lib/python3.9/site-packages/itsdangerous/__init__.py)", "stderr_lines": ["Traceback (most recent call last):", "  File \"/library/lokole/venv/bin/manage.py\", line 9, in <module>", "    from flask_migrate import MigrateCommand", "  File \"/library/lokole/venv/lib/python3.9/site-packages/flask_migrate/__init__.py\", line 6, in <module>", "    from flask import current_app", "  File \"/library/lokole/venv/lib/python3.9/site-packages/flask/__init__.py\", line 19, in <module>", "    from . import json", "  File \"/library/lokole/venv/lib/python3.9/site-packages/flask/json/__init__.py\", line 15, in <module>", "    from itsdangerous import json as _json", "ImportError: cannot import name 'json' from 'itsdangerous' (/library/lokole/venv/lib/python3.9/site-packages/itsdangerous/__init__.py)"], "stdout": "", "stdout_lines": []}

CC: @nzola

Refs:

- #2833
- #2986

holta, 2022-03-11:
Another failure was reported overnight, on 32-bit RasPiOS on her RPi 4:

- #3133

(Whereas this #3132 occurred on 64-bit RasPiOS on my RPi 4.)

holta, 2022-03-28:
This issue appears likely solve by...

- PR #3149

But @jvonau can confirm.

jvonau, 2022-03-29:
@nzola mentioned a successful install, just wondering where the iiab-diagnostics info is at.

nzola, 2022-03-29:
COMPLETE! Your diagnostics file (164K, 2354 lines) is:

   /etc/iiab/diag/8.0_raspbian-11_2022-03-29_02:02:15_BST_Nzola

Publish it to a web pastebin? [Y/n] y

PUBLISHING TO URL... http://sprunge.us/muPIRd

jvonau, 2022-03-29:
Looks good at a glance, issue should be closed, revert when fixed upstream. @holta Might want to clone the upstream repo in place of using what is available on pypi for the install source.

holta, 2022-03-29:
> revert when fixed upstream.

Ok

> @holta Might want to clone the upstream repo in place of using what is available on pypi for the install source.

Possibly wait until Lokole itself is revised?

holta, 2022-03-29:
Thanks @nzola and @jvonau for pushing this forward.

And the new person too, if they choose to participate.

(We have enough orphans in this world already!)

jvonau, 2022-03-30:
> > @holta Might want to clone the upstream repo in place of using what is available on pypi for the install source.
> 
> Possibly wait until Lokole itself is revised?

No, I don't think upstream will ever be released on pypi again until somebody tags a new release so the most recent changes in git will not be used in the install. I would always just clone the git source bypassing the need for released packaged versions to be updated.

holta, 2022-03-30:
@nzola is your new person going to fix the requirements.txt file soon, as Jerry outlined on ascoderu/lokole#585 here?

- https://github.com/ascoderu/lokole/issues/585 ?

> I would always just clone the git source bypassing the need for released packaged versions to be updated.

Quite possibly IIAB should clone from ascoderu/lokole's master branch (during IIAB installation) in general, to tighten the feedback loop allowing for more responsive Lokole QA.  I'd be in favor if @jvonau wants to go that route.

If so, all 3 of IIAB's pip installs of Lokole would presumably be commented out hereunder?

https://github.com/iiab/iiab/blob/4ede465c232c94862370e39bd8d875535a24e071/roles/lokole/tasks/install.yml#L38-L67

jvonau, 2022-03-30:
No, just don't pin the install to a git hash, tinker with line 43 (use other roles as a model) and comment out 47 <-> 67

holta, 2022-03-30:
To the new person working with @nzola:

If I'm understanding @jvonau correctly, he may be suggesting changing...

https://github.com/iiab/iiab/blob/4ede465c232c94862370e39bd8d875535a24e071/roles/lokole/tasks/install.yml#L43 to something like...

```
    name: "git+https://github.com/ascoderu/lokole.git#egg=opwen_email_client"
```

holta, 2022-04-07:
During today's http://minutes.iiab.io weekly call, @jvonau recommends closing this for now.

Awaiting the upstream fix here soon hopefully:

- ascoderu/lokole#585

closed 2022-04-07 by holta

jvonau, 2022-06-30:
#3276