iiab/iiab #3133: Brand new install keeps failing

susanrm, 2022-03-11:
### Expected behavior and actual behavior:

### Steps to reproduce the problem:

### IIAB version, OS version, hardware details:

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
Multiple attempts to install on brand new Rasp Pi 400 with new SD card and new install of fresh Raspian keep failing. http://sprung.us/IzFa7F

Errors: 
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "while read envvar; do export \"$envvar\"; done < /home/lokole/state/settings.env\n/library/lokole/venv/bin/manage.py createadmin --name='admin' --password='changeme'\n", "delta": "0:00:00.462953", "end": "2022-03-10 21:38:39.888088", "msg": "non-zero return code", "rc": 1, "start": "2022-03-10 21:38:39.425135", "stderr": "Traceback (most recent call last):\n  File \"/library/lokole/venv/bin/manage.py\", line 9, in <module>\n    from flask_migrate import MigrateCommand\n  File \"/library/lokole/venv/lib/python3.9/site-packages/flask_migrate/__init__.py\", line 6, in <module>\n    from flask import current_app\n  File \"/library/lokole/venv/lib/python3.9/site-packages/flask/__init__.py\", line 19, in <module>\n    from . import json\n  File \"/library/lokole/venv/lib/python3.9/site-packages/flask/json/__init__.py\", line 15, in <module>\n    from itsdangerous import json as _json\nImportError: cannot import name 'json' from 'itsdangerous' (/library/lokole/venv/lib/python3.9/site-packages/itsdangerous/__init__.py)", "stderr_lines": ["Traceback (most recent call last):", "  File \"/library/lokole/venv/bin/manage.py\", line 9, in <module>", "    from flask_migrate import MigrateCommand", "  File \"/library/lokole/venv/lib/python3.9/site-packages/flask_migrate/__init__.py\", line 6, in <module>", "    from flask import current_app", "  File \"/library/lokole/venv/lib/python3.9/site-packages/flask/__init__.py\", line 19, in <module>", "    from . import json", "  File \"/library/lokole/venv/lib/python3.9/site-packages/flask/json/__init__.py\", line 15, in <module>", "    from itsdangerous import json as _json", "ImportError: cannot import name 'json' from 'itsdangerous' (/library/lokole/venv/lib/python3.9/site-packages/itsdangerous/__init__.py)"], "stdout": "", "stdout_lines": []}

holta, 2022-03-11:
Hi @susanrm

Apologies this is an unfortunate issue with LARGE-sized IIAB installs on Python 3.9+ &mdash; also documented here:

- #3132

Can you change [Lines 235 and 236](https://github.com/iiab/iiab/blob/master/vars/local_vars_large.yml#L234-L236) of [/etc/iiab/local_vars.yml](https://wiki.iiab.io/go/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it?) so they appear as follows?

```
# Lokole (email for rural communities) from https://ascoderu.ca
lokole_install: False
lokole_enabled: False
```

And then continue with your IIAB install, by running:

```
sudo iiab
```

Thanks for your patience until a better solution hopefully emerges soon!

susanrm, 2022-03-11:
> Apologies this is an unfortunate issue with LARGE-sized IIAB installs on Python 3.9+ — also documented here:

Thank you! Will try this when I get back home.

holta, 2022-03-11:
> Thank you! Will try this when I get back home.

You should be all set.

But keep in touch if you have any other questions, Thanks!

PS If you can recommend a Python professional to help us fix [Lokole (offline/rural email)](https://github.com/iiab/iiab/tree/master/roles/lokole#lokole-readme) properly, we would be very grateful indeed.

holta, 2022-03-11:
@susanrm please if you can _also_ consider the [64-bit version of Raspberry Pi OS](https://downloads.raspberrypi.org/raspios_arm64/images/) instead of the 32-bit version you're currently using?

- ADVANTAGE: Calibre e-book software can be installed, making essential commands like [calibredb](https://manual.calibre-ebook.com/generated/en/calibredb.html) and [ebook-convert](https://manual.calibre-ebook.com/generated/en/ebook-convert.html) available to schools building thoughtful e-book collections (whereas it's [no longer possible](https://www.mobileread.com/forums/showthread.php?t=343422) to run `sudo apt install calibre` on 32-bit Raspberry Pi OS "Bullseye").
- DISADVANTAGE: 64-bit microSD cards are not portable (insertable) into 32-bit machines like the $10 Raspberry Pi Zero W and $15 Zero 2 W.

Anyway, you have options &mdash; both work!

As summarized in the **What's New** section describing IIAB 8.0's pre-releases:
https://github.com/iiab/iiab/wiki/IIAB-8.0-Release-Notes#whats-new

susanrm, 2022-03-12:
Thank you! I installed the 64-bit OS and disabled Lokole. Now it's working (mostly). Not sure if I should open another ticket but I searched and couldn't find an answer... I tried to change the iiab-admin password in the console (had left it default) and keep getting an error of wrong password when I enter the default (g0adm1n) and just in case, my general Pi password. A bit stuck here. Thanks!

holta, 2022-03-12:
> Not sure if I should open another ticket but I searched and couldn't find an answer... I tried to change the iiab-admin password in the console (had left it default) and keep getting an error of [...]

This definitely needs more investigation, to confirm what's happening (on which OS's, if this is indeed a bug?)

Thanks @susanrm and please follow up here:

- #3134

closed 2022-03-12 by holta