iiab/iiab #3134: "Unable to change iiab-admin password using Admin Console"

holta, 2022-03-12:
@susanrm writes:
> I tried to change the iiab-admin password in the console (had left it default) and keep getting an error of wrong password when I enter the default (g0adm1n) and just in case, my general Pi password. A bit stuck here. Thanks!

FYI the OS is 64-bit Raspberry Pi OS and the hardwre is Raspberry Pi 400.

More background/context:

- #3133

holta, 2022-03-12:
I can confirm that this fails on 64-bit Raspberry Pi OS:

[image]

FYI it certainly works with Ubuntu 22.04 &mdash; there appears to be a problem with some OS's and not others.

holta, 2022-03-12:
@susanrm thanks for catching this bug affecting newer OS's &mdash; apparently all OS's based on Debian Bullseye from August 2022 &mdash; that use newer/safer password hashing schemes.

@tim-moody likely this means that our fix of #2985 six months ago was incomplete?

If so the fix might be to replace this code:
https://github.com/iiab/iiab-admin-console/blob/master/roles/cmdsrv/files/iiab-cmdsrv3.py#L2809-L2813

With code like this:
https://github.com/iiab/iiab-admin-console/blob/master/roles/console/templates/cmd-service3.wsgi.j2#L179-L189

Ref:

- #2985
- PR iiab/iiab-admin-console#428

holta, 2022-03-12:
Tangentially, this typo should also be fixed:

- PR iiab/iiab-admin-console#470

holta, 2022-03-12:
@susanrm thanks again for the report!

A tested fix is here, that @tim-moody can hopefully review and merge if he has time overnight:

- PR iiab/iiab-admin-console#471

susanrm, 2022-03-13:
I guess I just wait for the fix, and then do what? Reinstall the whole thing? Thanks!

holta, 2022-03-13:
@susanrm you'll be able to quickly upgrade Admin Console, by running:

```
cd /opt/iiab/iiab-admin-console
sudo git pull
sudo ./install
```

Meantime you can also change your password at the command-line today, if you prefer, by running:

```
sudo passwd iiab-admin
```

susanrm, 2022-03-13:
Excellent, thank you!!

holta, 2022-03-17:
> Excellent, thank you!!

@susanrm can you try Admin Console's password-changing functionality if you have time?  Thanks!

(You can quickly upgrade Admin Console, by running...)

```
cd /opt/iiab/iiab-admin-console
sudo git pull
sudo ./install
```

Which will incorporate these 2 improvements:

- PR iiab/iiab-admin-console#470
- PR iiab/iiab-admin-console#472

holta, 2022-03-23:
Let's declare victory:

Thanks @susanrm and @tim-moody and all for keeping an eye on this!

closed 2022-03-23 by holta