iiab/iiab #3140: roles/internetarchive repeatably fails to install on Ubuntu Server 20.04 -- possibly due to "fatal: remote error: ", " The unauthenticated git protocol on port 9418 is no longer supported."

holta, 2022-03-16:
Hi @mitra42,

Would you happen to know why this is happening, or have any suggestions?  Thanks!

> TASK [internetarchive : Run 'yarn add @internetarchive/dweb-mirror' to download/populate /opt/iiab/internetarchive/node_modules (CAN TAKE ~5 MINUTES)] *****************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "yarn config set child-concurrency 1 && yarn add @internetarchive/dweb-mirror", "delta": "0:00:20.538391", "end": "2022-03-16 00:02:54.284715", "msg": "non-zero return code", "rc": 128, "start": "2022-03-16 00:02:33.746324", "stderr": "warning @internetarchive/dweb-mirror > multihashes > multibase@0.7.0: This module has been superseded by the multiformats module\nwarning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.4 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.\nwarning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!\nerror Command failed.\nExit code: 128\nCommand: git\nArguments: ls-remote --tags --heads git://github.com/mitra42/readable-stream-clone.git\nDirectory: /opt/iiab/internetarchive\nOutput:\nfatal: remote error: \n  The unauthenticated git protocol on port 9418 is no longer supported.\nPlease see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.", "stderr_lines": ["warning @internetarchive/dweb-mirror > multihashes > multibase@0.7.0: This module has been superseded by the multiformats module", "warning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.4 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.", "warning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!", "error Command failed.", "Exit code: 128", "Command: git", "Arguments: ls-remote --tags --heads git://github.com/mitra42/readable-stream-clone.git", "Directory: /opt/iiab/internetarchive", "Output:", "fatal: remote error: ", "  The unauthenticated git protocol on port 9418 is no longer supported.", "Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information."], "stdout": "yarn config v1.22.17\nsuccess Set \"child-concurrency\" to \"1\".\nDone in 0.04s.\nyarn add v1.22.17\ninfo No lockfile found.\n[1/4] Resolving packages...\n[2/4] Fetching packages...\ninfo Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command.", "stdout_lines": ["yarn config v1.22.17", "success Set \"child-concurrency\" to \"1\".", "Done in 0.04s.", "yarn add v1.22.17", "info No lockfile found.", "[1/4] Resolving packages...", "[2/4] Fetching packages...", "info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command."]}

Code is here:
https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/install.yml#L45-L49

Summary Diagnostics:
This is part of a fresh/clean install of the latest pre-release of IIAB 8.0

This VM's Ubuntu Server 20.04.4 has all apt updates applied, and is running the following kernel:

> root@u20-srv:~# uname -a
Linux box 5.4.0-104-generic #118-Ubuntu SMP Wed Mar 2 19:02:41 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

> root@u20-srv:~# yarn --version
1.22.17

> root@u20-srv:~# node -v
v16.14.0

> root@u20-srv:~# npm -v
8.3.1

I've verified this is not an issue of disk or RAM,

mitra42, 2022-03-16:
This appears to be a problem in our install, due to a change in GIT - to provide more inconvenience, oops I mean more security.   I'm going to have to look into it, and at the moment I'm in the middle of supporting the flood recovery whr I am (search on mullumbimby floods) so it might take me a few weeks to get to this.

mitra42, 2022-03-16:
Do you know if anyone is currently using the internet archive as part of the services they offer, it would help to get an idea of how many people impacted ?

holta, 2022-03-16:
> This appears to be a problem in our install, due to a change in GIT - to provide more inconvenience, oops I mean more security. I'm going to have to look into it, and at the moment I'm in the middle [...]

Thanks @mitra42 if you can in coming weeks.

In response to your other question: There appears to be many communities per month installing this, as informal support requests specifically for Internet Archive as part of Internet-in-a-Box occur regularly.  However we're unfortunately lacking exact usage numbers across all offline communities worldwide, in all cases, as our privacy-centric approach has unfortunately prevented this to date :/

mitra42, 2022-03-16:
Thanks - it would be good to know if its being used, our feedback was that the materials on the Archive what not in the right languages, and not current enough for the mostly educational uses envisioned, so the project was finished, though I'm still happy to help (when we aren't in the middle of a state-of-emergency)

holta, 2022-03-16:
Coincidentally today:

```
root@u20-srv:/opt/iiab/iiab# apt list --upgradable
Listing... Done
nodejs/unknown 16.14.1-deb-1nodesource1 amd64 [upgradable from: 16.14.0-deb-1nodesource1]
yarn/stable,stable 1.22.18-1 all [upgradable from: 1.22.17-1]
```

But unfortunately installing the above two packages does not improve the situation (I also rebooted first, to be sure!)

holta, 2022-03-17:
Just FYI I've confirmed this same problem occurs on Ubuntu 22.04 (Jammy Jellyfish) pre-releases, who final release is expected in 5 weeks from today.

jvonau, 2022-03-21:
```
TASK [internetarchive : Run 'yarn add @internetarchive/dweb-mirror' to download/populate /opt/iiab/internetarchive/node_modules (CAN TAKE ~5 MINUTES)] *****************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "yarn config set child-concurrency 1 && yarn add @internetarchive/dweb-mirror", "delta": "0:00:20.538391", "end": "2022-03-16 00:02:54.284715", "msg": "non-zero return code", "rc": 128, "start": "2022-03-16 00:02:33.746324", "stderr": "warning @internetarchive/dweb-mirror > multihashes > multibase@0.7.0: This module has been superseded by the multiformats module\n
warning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.4 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.\n
warning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!\n
error Command failed.\n
Exit code: 128\n
Command: git\n
Arguments: ls-remote --tags --heads git://github.com/mitra42/readable-stream-clone.git\n
Directory: /opt/iiab/internetarchive\n
Output:\n
fatal: remote error: \n
 The unauthenticated git protocol on port 9418 is no longer supported.\n
 Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.", "stderr_lines": ["warning @internetarchive/dweb-mirror > multihashes > multibase@0.7.0: This module has been superseded by the multiformats module", "warning @internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.4 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.", "warning @internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!", "error Command failed.", "Exit code: 128", "Command: git", "Arguments: ls-remote --tags --heads git://github.com/mitra42/readable-stream-clone.git", "Directory: /opt/iiab/internetarchive", "Output:", "fatal: remote error: ", " The unauthenticated git protocol on port 9418 is no longer supported.", "Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information."], "stdout": "yarn config v1.22.17\nsuccess Set "child-concurrency" to "1".\n
 Done in 0.04s.\n
 yarn add v1.22.17\n
 info No lockfile found.\n
 [1/4] Resolving packages...\n
 [2/4] Fetching packages...\n
 info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command.", "stdout_lines": ["yarn config v1.22.17", "success Set "child-concurrency" to "1".", "Done in 0.04s.", "yarn add v1.22.17", "info No lockfile found.", "[1/4] Resolving packages...", "[2/4] Fetching packages...", "info Visit https://yarnpkg.com/en/docs/cli/add for documentation about this command."]}
```
https://github.blog/2021-09-01-improving-git-protocol-security-github/
Think [yarn.lock](https://github.com/internetarchive/dweb-mirror/blob/master/yarn.lock#L2672-L2674) and/or [package.json](https://github.com/internetarchive/dweb-mirror/blob/master/package.json#L34) needs to be altered to use git+https:// in place of just git://

willbeonekenobi, 2022-03-22:
FYI This also happens in 20.04.

I was able to manually clone 'readable-stream-clone' manually without issue. It seems that both the [Yarn.lock](https://github.com/internetarchive/dweb-mirror/blob/master/yarn.lock#L2672-L2674) and maybe also the [package.json](https://github.com/internetarchive/dweb-mirror/blob/master/package.json#L34) need to be altered.

mitra42, 2022-03-22:
I saw your pull requests - were you able to test them, or do you need me to merge before you can test (I don't currently have a test environment).   I'd actually be fine with a merge before test, since currently the package won't install so there would be no regression.

willbeonekenobi, 2022-03-22:
I pulled the previous pull requests, (had bit of a typo) and resubmitted.

mitra42, 2022-03-22:
Ok - I merged them, they look good, but am unable to test till I can get a test environment back up again 
- @holta can you let me know if that clears the overall problem or if there is something else?

holta, 2022-03-23:
FWIW this continues to fail on both Ubuntu 20.04 and 22.04:

```
sudo su -
rm -rf /opt/iiab/internetarchive
cd /opt/iiab/iiab
./runrole internetarchive
```

Error appears to be the same as a week ago:

```
TASK [internetarchive : Run 'yarn add @internetarchive/dweb-mirror' to d
ownload/populate /opt/iiab/internetarchive/node_modules (CAN TAKE ~5 MIN
UTES)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "yarn config set
 child-concurrency 1 && yarn add @internetarchive/dweb-mirror", "delta":
 "0:00:08.582506", "end": "2022-03-23 03:40:43.564885", "msg": "non-zero
 return code", "rc": 128, "start": "2022-03-23 03:40:34.982379", "stderr
": "warning @internetarchive/dweb-mirror > multihashes > multibase@0.7.0
: This module has been superseded by the multiformats module\nwarning @i
nternetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react
> fbjs > core-js@1.2.7: core-js@<3.4 is no longer maintained and not rec
ommended for usage due to the number of issues. Because of the V8 engine
 whims, feature detection in old core-js versions could cause a slowdown
 up to 100x even if nothing is polyfilled. Please, upgrade your dependen
cies to the actual version of core-js.\nwarning @internetarchive/dweb-mi
rror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34:
This is a stub types definition for localforage (https://github.com/loca
lForage/localForage). localforage provides its own type definitions, so
you don't need @types/localforage installed!\nerror Command failed.\nExi
t code: 128\nCommand: git\nArguments: ls-remote --tags --heads git://git
hub.com/mitra42/readable-stream-clone.git\nDirectory: /opt/iiab/internet
archive\nOutput:\nfatal: remote error: \n  The unauthenticated git proto
col on port 9418 is no longer supported.\nPlease see https://github.blog
/2021-09-01-improving-git-protocol-security-github/ for more information
.", "stderr_lines": ["warning @internetarchive/dweb-mirror > multihashes
 > multibase@0.7.0: This module has been superseded by the multiformats
module", "warning @internetarchive/dweb-mirror > @internetarchive/dweb-a
rchive-dist > react > fbjs > core-js@1.2.7: core-js@<3.4 is no longer ma
intained and not recommended for usage due to the number of issues. Beca
use of the V8 engine whims, feature detection in old core-js versions co
uld cause a slowdown up to 100x even if nothing is polyfilled. Please, u
pgrade your dependencies to the actual version of core-js.", "warning @i
nternetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @type
s/localforage@0.0.34: This is a stub types definition for localforage (h
ttps://github.com/localForage/localForage). localforage provides its own
 type definitions, so you don't need @types/localforage installed!", "er
ror Command failed.", "Exit code: 128", "Command: git", "Arguments: ls-r
emote --tags --heads git://github.com/mitra42/readable-stream-clone.git"
, "Directory: /opt/iiab/internetarchive", "Output:", "fatal: remote erro
r: ", "  The unauthenticated git protocol on port 9418 is no longer supp
orted.", "Please see https://github.blog/2021-09-01-improving-git-protoc
ol-security-github/ for more information."], "stdout": "yarn config v1.2
2.18\nsuccess Set \"child-concurrency\" to \"1\".\nDone in 0.05s.\nyarn
add v1.22.18\ninfo No lockfile found.\n[1/4] Resolving packages...\n[2/4
] Fetching packages...\ninfo Visit https://yarnpkg.com/en/docs/cli/add f
or documentation about this command.", "stdout_lines": ["yarn config v1.
22.18", "success Set \"child-concurrency\" to \"1\".", "Done in 0.05s.",
 "yarn add v1.22.18", "info No lockfile found.", "[1/4] Resolving packag
es...", "[2/4] Fetching packages...", "info Visit https://yarnpkg.com/en
/docs/cli/add for documentation about this command."]}
```

Is there a need to clear out anything stale that roles/internetarchive might have installed or cached a week ago?

jvonau, 2022-03-23:
Try: `sudo yarn cache clean dweb-mirror`

holta, 2022-03-23:
> Try: `sudo yarn cache clean dweb-mirror`

```
root@box:~# yarn cache clean dweb-mirror                   
yarn cache v1.22.18
error An unexpected error occurred: "There should only be one folder in
a package cache (got  in /usr/local/share/.cache/yarn/v6/npm-readable-st
ream-clone-0.0.4-b8827df9a52ae5d278995ab25548c0206ea0c210/node_modules)"
.
info If you think this is a bug, please open a bug report with the infor
mation provided in "/opt/iiab/iiab/yarn-error.log".
info Visit https://yarnpkg.com/en/docs/cli/cache for documentation about
 this command.
```

I ran `rm -rf /usr/local/share/.cache/yarn` which allowed the above command to be re-run without error.

Regrettably however, the same error still appears, when running `./runrole internetarchive` :

> "error Command failed.", "Exit code: 128", "Command: git", "Arguments: ls-remote --tags --heads git://github.com/mitra42/readable-stream-clone.git", "Directory: /opt/iiab/internetarchive", [...]

@mitra42 can you tell where this command `ls-remote --tags --heads git://github.com/mitra42/readable-stream-clone.git` is arising from?

jvonau, 2022-03-23:
Side question, wouldn't "version:" in package.json need to be bumped for npm/yarn to be able to know some dependency in the package changed?

jvonau, 2022-03-23:
@holta just start over on a new image to verify the fix is good moving forward, something is caching the old version with no point in whining about spilled milk.

jvonau, 2022-03-23:
The package is listed in the npm [registry](https://www.npmjs.com/package/@internetarchive/dweb-mirror) so that is where the package is pulled from by yarn. Try this by hand, this is what #3143 did without adding 'git' and 'remote' and pulls directly from github
```
sudo su -
rm -rf /opt/iiab/internetarchive
mkdir /opt/iiab/internetarchive
cd /opt/iiab/internetarchive
yarn add https://github.com/internetarchive/dweb-mirror
```

jvonau, 2022-03-23:
```
ubuntu@multipass1:/opt/iiab/IA2$ cat package.json
{
  "dependencies": {
    "@internetarchive/dweb-mirror": "https://github.com/internetarchive/dweb-mirror"
  }
}
```
```
ubuntu@multipass1:/opt/iiab/IA2$ ls -l node_modules/@internetarchive/
total 20
drwxr-xr-x 10 root root 4096 Mar 23 10:26 bookreader
drwxr-xr-x  9 root root 4096 Mar 23 10:26 dweb-archive-dist
drwxr-xr-x  4 root root 4096 Mar 23 10:26 dweb-archivecontroller
drwxr-xr-x  5 root root 4096 Mar 23 10:26 dweb-mirror
drwxr-xr-x  3 root root 4096 Mar 23 10:26 dweb-transports

```

holta, 2022-03-23:
> The package is listed in the npm [registry](https://www.npmjs.com/package/@internetarchive/dweb-mirror) so that is where the package is pulled from by yarn. Try this by hand, this is what #3143 did [...]

I confirm that the above manual steps do work on Ubuntu 20.04

jvonau, 2022-03-23:
```
ubuntu@multipass1:/opt/iiab/IA2$ grep clone yarn.lock 
    readable-stream-clone "https://github.com/mitra42/readable-stream-clone.git"
"readable-stream-clone@https://github.com/mitra42/readable-stream-clone.git":
  resolved "https://github.com/mitra42/readable-stream-clone.git#b8827df9a52ae5d278995ab25548c0206ea0c210"
```

holta, 2022-03-23:
> I confirm that the above manual steps do work on Ubuntu 20.04

Just FYI here's the output and its 3 warnings:

> root@box:/opt/iiab/internetarchive# yarn add https://github.com/internetarchive/                                                                                        dweb-mirror
yarn add v1.22.18
info No lockfile found.
[1/4] Resolving packages...
warning https://github.com/internetarchive/dweb-mirror > multihashes > multibase@0.7.0: This module has been superseded by the multiformats module
warning https://github.com/internetarchive/dweb-mirror > @internetarchive/dweb-archive-dist > react > fbjs > core-js@1.2.7: core-js@<3.4 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Please, upgrade your dependencies to the actual version of core-js.
warning https://github.com/internetarchive/dweb-mirror > @futurepress/epubjs-reader > epubjs > @types/localforage@0.0.34: This is a stub types definition for localforage (https://github.com/localForage/localForage). localforage provides its own type definitions, so you don't need @types/localforage installed!
[2/4] Fetching packages...
[3/4] Linking dependencies...
[4/4] Building fresh packages...
success Saved lockfile.
success Saved 232 new dependencies.
info Direct dependencies
└─ @internetarchive/dweb-mirror@0.2.89
info All dependencies
├─ @ampproject/remapping@2.1.2
├─ @babel/compat-data@7.17.7
...
├─ wide-align@1.1.5
└─ yallist@4.0.0
Done in 13.96s.

jvonau, 2022-03-23:
> > I confirm that the above manual steps do work on Ubuntu 20.04
> 
> Just FYI here's the output and its 3 warnings:
> 
Just dependencies that should be fixed, does ansible puke with these warnings present?

jvonau, 2022-03-23:
> Ansible does not stop: I posted them for the record, in case some of these warnings may bite us in future?
> 

Just part of the ever changing nodejs dependencies that need to be maintained upstream, at some point the warnings might become fatal errors.

jvonau, 2022-03-29:
Found in muPIRd in another issue.
>Mar 29 01:48:51 raspberrypi internetarchive[83476]: 2022-03-29T00:48:51.047Z dweb-transports:httptools p_httpfetch: https://dweb.me/info ''
Mar 29 01:48:51 raspberrypi internetarchive[83476]: 2022-03-29T00:48:51.849Z dweb-transports:httptools Fetch of https://dweb.me/info opened

Seem strange for an offline server, opening the connection every 30 seconds. Is this normal or a configuration error somewhere?

mitra42, 2022-03-29:
It was intended as part of a mechanism to know how many peers online. Nothing will block if its offline.

jvonau, 2022-03-30:
Wonder how big the log file becomes recording every 30 seconds.. Is there any reason to keep this issue open?

mitra42, 2022-03-30:
No need to keep it open that I'm aware of, since @holta  has said it tests ok - sorry I couldn't jump on it quicker, I've been volunteering full time in disaster relief the last 4 weeks.

closed 2022-03-30 by holta

holta, 2022-03-30:
Many thanks @jvonau, @willbeonekenobi and @mitra42 for pulling together to solve this.