iiab/iiab #3147: small imager image does not enlarge file system into expanded partition

georgejhunt, 2022-03-24:
Experienced on https://d.iiab.io/8.0/iiab-8.0beta1-220203-SMALL-raspios-11-lite-g6598f3e1.img.zip

fixed by logging on as root and doing "resize2fs /dev/mmcblk0p2"

holta, 2022-03-24:
@georgejhunt I did a complete test and was unable to reproduce the problem.

For me, the partition /dev/mmcblk0p2 expands to its fullsize about 30 seconds after I logged in (which was about 60 seconds after boot).

Question: Roughly how long did you wait after booting?

As an aside, the [code that expands IIAB's rootfs](https://github.com/iiab/iiab/blob/master/roles/1-prep/templates/iiab-expand-rootfs) has since been completely replaced:

- PR #3121 (Feb 17/18)
- PR #3137 (Mch 15)

georgejhunt, 2022-03-24:
I will repeat and wait for 5 min.  I  did not realize that I needed to wait. And there was what I interpreted as a stall during the initial boot.

georgejhunt, 2022-03-25:
I failed to replicate when I waited.  It was resized in 2 or 3 minutes. But was really sluggish during the resize process.

closed 2022-03-25 by georgejhunt