iiab/iiab #3159: New resizing logic requires an extra reboot

tim-moody, 2022-04-02:
write image from latest master prepared with min and cp sd to sd card and boot up the image

in the old version card resizing would begin and could be seen in progress in ssh

now reboot after first boot is required to start resizing.

holta, 2022-04-02:
Apparently `raspi-config --expand-rootfs` would seem to now require an additional reboot.

[Line 17 of 1-prep/templates/iiab-expand-rootfs](https://github.com/iiab/iiab/blob/212d1fa765887ba3830099dd6bb8f7ffc289f5b1/roles/1-prep/templates/iiab-expand-rootfs#L17) tries to delegate the details to RasPiOS so they can manage all the evolving details year-by-year.

But if a reboot is too annoying, we have an alternative: [Lines 20-34](https://github.com/iiab/iiab/blob/212d1fa765887ba3830099dd6bb8f7ffc289f5b1/roles/1-prep/templates/iiab-expand-rootfs#L20-L34) suddenly works far better (as a result of PR #3137 2.5 weeks ago) on most Debuntu-style OS's.  So 2 months later we can consider reverting to this older approach (now refined to deal with external boot disks, etc...)

Refs:

- PR #3121
- PR #3137 
- #3147

holta, 2022-04-02:
Resolved by:

- PR #3160

closed 2022-04-02 by holta

jvonau, 2022-04-02:
Guess someone missed the part of raspi-config that uses an older fdisk editing [method](https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L160-L196) where the partition needs to be unmounted while editing when you liberated some of the underling functions when working up 2131.  A reboot is not that bad and is what RaspiOS does when booting a 'fresh' image out of the box, just a change in behavior when using the builtin tools. Using growpart is way more flexible, and is the standard on Ubuntu based arm64 images. Now the question becomes should growpart be installed on all installations to take advantage of the revised root device detection that should work on all /dev/sd* root filesystems? That would be a bonus if cloning an intel based machine could be achieved with the tools within iiab-factory.

holta, 2022-04-02:
> should growpart be installed on all installations

Yes.  I agree.  Clarified by 4e2e4c8.

(With respect to `raspi-config --expand-rootfs` whose open source code has allowed all this progress over the past 2 months!)