iiab/iiab #3166: Can't disable remote.it once registered.

jvonau, 2022-04-05:
### Expected behavior and actual behavior:
be able to turn off remoteit once registered via local_vars _enabled: False
### Steps to reproduce the problem:
Claim the device and setup remote ssh access
### IIAB version, OS version, hardware details:
current master

>ubuntu@multipass1:/opt/iiab/iiab$ ls /etc/systemd/system/multi-user.target.wants/remoteit*
/etc/systemd/system/multi-user.target.wants/remoteit@80:00:01:7F:7E:00:B0:DD.service
/etc/systemd/system/multi-user.target.wants/remoteit@80:00:01:7F:7E:00:B0:E7.service
/etc/systemd/system/multi-user.target.wants/remoteit@80:00:01:7F:7E:00:B1:01.service

edit local_vars to be remoteit_enabled: False
`sudo ./runrole remoteit`

```
TASK [0-init : STAGE 0 HAS COMPLETED ======================================] **********************************************

TASK [remoteit : Assert that "remoteit_install is sameas true" (boolean not string etc)] **********************************
ok: [127.0.0.1]

TASK [remoteit : Assert that "remoteit_enabled | type_debug == 'bool'" (boolean not string etc)] **************************
ok: [127.0.0.1]

TASK [remoteit : Install remoteit if 'remoteit_installed' not defined, e.g. in /etc/iiab/iiab_state.yml] ******************
skipping: [127.0.0.1]

TASK [remoteit : include_tasks] *******************************************************************************************
included: /opt/iiab/iiab/roles/remoteit/tasks/enable-or-disable.yml for 127.0.0.1

TASK [remoteit : Enable & Restart remote.it "parent" service connectd, which exits after spawning 2 "child" services/daemons below] ***
skipping: [127.0.0.1]

TASK [remoteit : Enable remote.it daemon schannel ("Remote tcp command service") -- try to avoid contention with connectd which auto-spawns it as nec (just above)] ***
skipping: [127.0.0.1]

TASK [remoteit : Disable & Stop remote.it services {connectd, schannel}] **************************************************
changed: [127.0.0.1] => (item=connectd)
changed: [127.0.0.1] => (item=schannel)

TASK [remoteit : Identify remoteit "Remote tcp connection service" unit file name, including uuid, e.g. remoteit@80:00:01:7F:7E:00:56:36.service] ***
changed: [127.0.0.1]

TASK [remoteit : Disable & Stop the actual service: remoteit@80:00:01:7F:7E:00:B0:DD.service
remoteit@80:00:01:7F:7E:00:B0:E7.service
remoteit@80:00:01:7F:7E:00:B1:01.service] ***
ok: [127.0.0.1]

TASK [remoteit : Add 'remoteit' variable values to /etc/iiab/iiab.ini] ****************************************************
ok: [127.0.0.1] => (item={'option': 'name', 'value': 'remote.it'})
ok: [127.0.0.1] => (item={'option': 'description', 'value': '"https://remote.it can help you remotely maintain an IIAB.  Some benefits include: crossing multiple NATs/firewalls using a single TCP port, without requiring router port forwarding, and reducing your network\'s vulnerability."'})
ok: [127.0.0.1] => (item={'option': 'remoteit_install', 'value': True})
changed: [127.0.0.1] => (item={'option': 'remoteit_enabled', 'value': False})

PLAY RECAP ****************************************************************************************************************
127.0.0.1                  : ok=35   changed=10   unreachable=0    failed=0    skipped=8    rescued=0    ignored=0   
```
> ubuntu@multipass1:/opt/iiab/iiab$ ls /etc/systemd/system/multi-user.target.wants/remoteit*
/etc/systemd/system/multi-user.target.wants/remoteit@80:00:01:7F:7E:00:B0:DD.service
/etc/systemd/system/multi-user.target.wants/remoteit@80:00:01:7F:7E:00:B0:E7.service
/etc/systemd/system/multi-user.target.wants/remoteit@80:00:01:7F:7E:00:B1:01.service

```
ubuntu@multipass1:/opt/iiab/iiab$ systemctl status remoteit* --no-pager
● remoteit@80:00:01:7F:7E:00:B1:01.service - Remote tcp connection service 80:00:01:7F:7E:00:B1:01
     Loaded: loaded (/lib/systemd/system/remoteit@.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-04-05 14:20:24 CDT; 48min ago
   Main PID: 1651 (connectd)
      Tasks: 1 (limit: 4579)
     Memory: 236.0K
        CPU: 1.017s
     CGroup: /system.slice/system-remoteit.slice/remoteit@80:00:01:7F:7E:00:B1:01.service
             └─1651 /usr/share/remoteit/connectd -e bWF4X2RlcHRoIDM1CmFwcGxpY2F0aW9uX3R5cGUgMjgKcHJveHlfZGVzdF9pcCAxMjcuMC…

Apr 05 14:20:26 multipass1 remoteit.sh[1651]:   Using Server Channel Port 5980
Apr 05 14:20:26 multipass1 remoteit.sh[1651]: primary local ip = 192.168.122.242
Apr 05 14:20:26 multipass1 remoteit.sh[1651]: local IP address found 192.168.122.242
Apr 05 14:20:26 multipass1 remoteit.sh[1651]: Secret Provided (hash a9:80)
Apr 05 14:20:26 multipass1 remoteit.sh[1651]: initialize TCP target 127.0.0.1 port 22
Apr 05 14:20:27 multipass1 remoteit.sh[1651]: Command Processor now active.
Apr 05 14:20:28 multipass1 remoteit.sh[1651]: 57561> sending req auth to 52.38.107.102:5959
Apr 05 14:20:28 multipass1 remoteit.sh[1651]: 57562> Status redirect to  44.224.165.129:5964
Apr 05 14:20:28 multipass1 remoteit.sh[1651]: 57562> sending req auth to 44.224.165.129:5964
Apr 05 14:20:32 multipass1 remoteit.sh[1651]: 57600> sending req auth to 44.224.165.129:5964

● remoteit@80:00:01:7F:7E:00:B0:DD.service - Remote tcp connection service 80:00:01:7F:7E:00:B0:DD
     Loaded: loaded (/lib/systemd/system/remoteit@.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-04-05 14:20:24 CDT; 48min ago
   Main PID: 1648 (connectd)
      Tasks: 1 (limit: 4579)
     Memory: 228.0K
        CPU: 1.007s
     CGroup: /system.slice/system-remoteit.slice/remoteit@80:00:01:7F:7E:00:B0:DD.service
             └─1648 /usr/share/remoteit/connectd -e bWF4X2RlcHRoIDM1CmFwcGxpY2F0aW9uX3R5cGUgMzUKcHJveHlfZGVzdF9pcCAxMjcuMC…

Apr 05 14:20:28 multipass1 remoteit.sh[1648]:   Using Server Channel Port 5980
Apr 05 14:20:28 multipass1 remoteit.sh[1648]: primary local ip = 192.168.122.242
Apr 05 14:20:28 multipass1 remoteit.sh[1648]: local IP address found 192.168.122.242
Apr 05 14:20:28 multipass1 remoteit.sh[1648]: Secret Provided (hash fd:4)
Apr 05 14:20:28 multipass1 remoteit.sh[1648]: initialize TCP target 127.0.0.1 port 65535
Apr 05 14:20:29 multipass1 remoteit.sh[1648]: Command Processor now active.
Apr 05 14:20:29 multipass1 remoteit.sh[1648]: 57571> sending req auth to 52.38.107.102:5959
Apr 05 14:20:29 multipass1 remoteit.sh[1648]: 57572> Status redirect to  52.43.176.61:5962
Apr 05 14:20:29 multipass1 remoteit.sh[1648]: 57572> sending req auth to 52.43.176.61:5962
Apr 05 14:20:33 multipass1 remoteit.sh[1648]: 57610> sending req auth to 52.43.176.61:5962

● remoteit@80:00:01:7F:7E:00:B0:E7.service - Remote tcp connection service 80:00:01:7F:7E:00:B0:E7
     Loaded: loaded (/lib/systemd/system/remoteit@.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2022-04-05 14:20:24 CDT; 48min ago
   Main PID: 1650 (connectd)
      Tasks: 1 (limit: 4579)
     Memory: 228.0K
        CPU: 1.123s
     CGroup: /system.slice/system-remoteit.slice/remoteit@80:00:01:7F:7E:00:B0:E7.service
             └─1650 /usr/share/remoteit/connectd -e bWF4X2RlcHRoIDM1CmFwcGxpY2F0aW9uX3R5cGUgMjgKcHJveHlfZGVzdF9pcCAxMjcuMC…

Apr 05 14:20:28 multipass1 remoteit.sh[1650]:   Using Server Channel Port 5980
Apr 05 14:20:28 multipass1 remoteit.sh[1650]: primary local ip = 192.168.122.242
Apr 05 14:20:28 multipass1 remoteit.sh[1650]: local IP address found 192.168.122.242
Apr 05 14:20:28 multipass1 remoteit.sh[1650]: Secret Provided (hash bc:a5)
Apr 05 14:20:28 multipass1 remoteit.sh[1650]: initialize TCP target 127.0.0.1 port 22
Apr 05 14:20:29 multipass1 remoteit.sh[1650]: Command Processor now active.
Apr 05 14:20:29 multipass1 remoteit.sh[1650]: 57571> sending req auth to 52.38.107.102:5959
Apr 05 14:20:29 multipass1 remoteit.sh[1650]: 57572> Status redirect to  52.43.176.61:5964
Apr 05 14:20:29 multipass1 remoteit.sh[1650]: 57572> sending req auth to 52.43.176.61:5964
Apr 05 14:20:33 multipass1 remoteit.sh[1650]: 57610> sending req auth to 52.43.176.61:5964

```

holta, 2022-04-05:
Important insight — that remote.it Device Package on IIAB can spawn multiple "child" services of the form `/etc/systemd/system/multi-user.target.wants/remoteit*`

Thanks for catching this.

holta, 2022-04-06:
Resolved by:

- PR #3167

closed 2022-04-06 by holta