iiab/iiab #3170: Test + Streamline 10.10.10.10 instead of 172.18.96.1

holta, 2022-04-06:
This can happen in 2023 or possibly beforehand as part of IIAB 8.0 much sooner, e.g. depending on popular demand (e.g. if more and more educators keep transitioning from RACHEL to Internet-in-a-Box, who all seem to want that!)

In short, whether http://10.10.10.10 should become mainline of not remains to be seen.

This would certainly depend on many things.  Including for example (a) testing with internal and external hotspots, (b) sanity-checking all current uses of 172.18.96.1 and associated masks etc &mdash; and (c) validating that any new arrangement works well with standard short URL's like:

1. `box/`
2. http://box
3. http://box.lan

jvonau, 2022-04-07:
This was pretty much soft-coded from the start, should be straight forward.

jvonau, 2022-07-08:
#3264

holta, 2022-07-20:
Done:

- PR #3281
- PR #3299

closed 2022-07-20 by holta