iiab/iiab #3188: Gitea install downloads extremely slowly

holta, 2022-04-20:
Even though it's only 103MB, Gitea 16.x (URL below) downloads around 1 Mbit/s or less, every time I'm tried it today:

https://dl.gitea.io/gitea/1.16/gitea-1.16-linux-amd64

This creates an unecessary ~15min delay for people explore LARGE-sized IIAB installs:

https://github.com/iiab/iiab/blob/1fb41333cee4b85bba530ca5435667b73ccba7b5/roles/gitea/tasks/install.yml#L46-L51

Hopefully Gitea is aware that this is a snail's pace, giving a bad 1st impression &mdash; and as such they might fix up their infra soon?

holta, 2022-04-20:
A very ugly workaround (to bypass https://dl.gitea.io 's slow downloads) would be to download Gitea point releases from GitHub.

But I'd rather NOT revert to that inefficient and inappropriate practice of hard-coding to a new Gitea point release every few weeks (that we abandoned for a very good reason in July 2021!)
In favor of the current practice (of specifying a Gitea branch every year or so, on each major release of theirs) which makes far more sense:

https://github.com/iiab/iiab/blob/0b4892b9152f7e24404c1ae84954351488a74c95/roles/gitea/defaults/main.yml#L12

tim-moody, 2022-04-20:
we have inventoried packages in the past and could do so for this

holta, 2022-04-20:
It's an option.  Certainly locking Gitea to any particular release (e.g. 1.16.5 from 2022-03-23) would instantly speed up download from GitHub URL's like...

https://github.com/go-gitea/gitea/releases/download/v1.16.5/gitea-1.16.5-linux-amd64 (103 MB)
https://github.com/go-gitea/gitea/releases/download/v1.16.5/gitea-1.16.5-linux-arm64 (99 MB)
https://github.com/go-gitea/gitea/releases/download/v1.16.5/gitea-1.16.5-linux-arm-6 (94.5 MB)

...but this band-aid fix is not a great road to head down &mdash; if it leads to unnecessarily stale versions of Gitea and/or manual maintenance burdens every few weeks (i.e. the need to manually update the Gitea version number almost constantly). :fearful:

The Good News is https://dl.gitea.io downloads are back into gear at end of the day today (about 300X faster than earlier today!) so I'll try to keep an eye on this horse race in coming weeks.  Hoping Gitea's infra gets back on track rather permanently... :horse_racing:

(Absolute worst case we could hypothetically extract the latest Gitea version number [1.16.5 for the moment] programmatically using GitHub's API or similar, for use in very fast URL's like those above.)

holta, 2022-04-21:
Gitea 1.16.6 was about 2h ago coincidentally, so I took to opportunity to confirmed their own downloads remain fast: (for now, so far so good!)

https://dl.gitea.io/gitea/1.16/gitea-1.16-linux-amd64 (103 MiB)
https://dl.gitea.io/gitea/1.16/gitea-1.16-linux-arm64 (99 MiB)
https://dl.gitea.io/gitea/1.16/gitea-1.16-linux-arm-6 (95 MiB)

holta, 2022-05-25:
> if the problem is fixed close the issue

The problem seemed ongoing but intermittent.

Let's close this ticket in the coming month if the problem is indeed rare.

holta, 2022-06-23:
Gitea's install download remains occasionally very slow at times, similar to Nextcloud.

(But we may have to live with it!)