iiab/iiab #3190: MongoDB 4.4, 5.0 & 6.0RC1 don't run on Ubuntu 22.04 (preventing Sugarizer install)

holta, 2022-04-22:
FYI @llaske & All,

Here's the reason Sugarizer won't install on Ubuntu 22.04 &mdash; regardless whether trying to install MongoDB 4.4 or 5.0 or 6.0RC1:

```
# apt-key list mongodb
Warning: apt-key is deprecated. Manage keyring files in trusted.gpg.d instead [of /etc/apt/trusted.gpg] (see apt-key(8)).
pub   rsa4096 2021-02-16 [SC] [expires: 2026-02-15]
      F567 9A22 2C64 7C87 527C  2F8C B00A 0BD1 E2C6 3C11
uid           [ unknown] MongoDB 5.0 Release Signing Key <packaging@mongodb.com>

# cat /etc/apt/sources.list.d/mongodb-org.list
deb [ arch=amd64,arm64 ] https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/5.0 multiverse

# apt install mongodb-org mongodb-org-server
...
The following packages have unmet dependencies:
 mongodb-org-mongos : Depends: libssl1.1 (>= 1.1.1) but it is not installable
 mongodb-org-server : Depends: libssl1.1 (>= 1.1.1) but it is not installable
 mongodb-org-shell : Depends: libssl1.1 (>= 1.1.1) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

( As mentioned 13 days ago at https://www.linuxshelltips.com/install-mongodb-ubuntu/#comment-9117 )

_Does this suggest we may need to wait for the release of MongoDB 5.4 or 6.0 or 6.1 ?_

If so, it might be many weeks away &mdash; given their latest "testing" and "development" branches (below) have the exact same `libssl1.1 (>= 1.1.1)` dependency problem:

- https://repo.mongodb.org/apt/ubuntu/dists/focal/mongodb-org/testing/multiverse/binary-amd64/mongodb-org-shell_6.0.0~rc1_amd64.deb
- https://repo.mongodb.org/apt/ubuntu/dists/focal/mongodb-org/development/multiverse/binary-amd64/mongodb-org-shell_6.0.0~latest_amd64.deb

Related:

- PR #3189

holta, 2022-04-22:
Preliminary workarounds ideas are being discussed here today:

https://askubuntu.com/questions/1403619/mongodb-install-fails-on-ubuntu-22-04-depends-on-libssl1-1-but-it-is-not-insta

e.g. try force impish's [21.10] apt package `libssl1.1` onto jammy [22.04] instead of the new apt package `libssl3` if you dare (-:

holta, 2022-04-30:
> Preliminary workaround ideas are being discussed here today:
> 
> https://askubuntu.com/questions/1403619/mongodb-install-fails-on-ubuntu-22-04-depends-on-libssl1-1-but-it-is-not-insta

More active (and official) "Installing mongodb over Ubuntu 22.04" discussion 3-5 days ago:

https://www.mongodb.com/community/forums/t/installing-mongodb-over-ubuntu-22-04/159931

llaske, 2022-05-07:
@holta just upgraded my laptop from Ubuntu 21.10 to Ubuntu 22.04.
Sugarizer work on my side with Mongodb v5.0.8.

holta, 2022-05-07:
> @holta just upgraded my laptop from Ubuntu 21.10 to Ubuntu 22.04. Sugarizer work on my side with Mongodb v5.0.8.

Great news.  Can I ask how installed MongoDB 5.0.8 ?

Ubuntu 22.04 isn't supporting MongoDB (as of now) so did you download some-or-all of the 7 .deb installers from https://repo.mongodb.org/apt/ubuntu/dists/focal/mongodb-org/5.0/multiverse/binary-amd64/ ?

llaske, 2022-05-08:
> Great news. Can I ask how installed MongoDB 5.0.8 ?

Yes, installed from https://repo.mongodb.org/apt/ubuntu

holta, 2022-05-08:
> > Great news. Can I ask how installed MongoDB 5.0.8 ?
> 
> Yes, installed from https://repo.mongodb.org/apt/ubuntu

Thanks.  Good to know 👍

We might be forced likewise, i.e. to hard-code this version of MongoDB (5.0.8) for people installing Internet-in-a-Box onto Ubuntu 22.04 and similar OS's.

I'd much prefer that IIAB auto-install the latest QA'd version of MongoDB [*], but this appears likely too complicated in this particular case?

[*] Similar to Nextcloud and WordPress etc, which offer latest.tar.gz links on their ftp/download sites.

holta, 2022-05-08:
Just FYI on its "testing" branch, MongoDB posts and regularly updates "latest" links, e.g. for MongoDB 6.0 Release Candidates especially:

https://repo.mongodb.org/apt/ubuntu/dists/focal/mongodb-org/testing/multiverse/

There are also a few "latest" links for MongoDB 5.1, 5.2 and 5.3 pre-releases as well.

_(But MongoDB appears to remove most of these "latest" links after their official product releases, so these are unfortunately of limited value to IIAB.)_

holta, 2022-05-08:
fyi MongoDB 5.0.8 does not install on a fresh Ubuntu 22.04 for me:

- PR #3214

(It seems we need to wait a few more weeks/months &mdash; until MongoDB figures out how to work with fresh installs of Ubuntu 22.04.)

holta, 2022-06-10:
Just FYI IIAB's MongoDB install routine was overhauled &mdash; automatically installing MongoDB 3.0 or 4.4 for older CPU's &mdash; and offering the option of MongoDB 5+ [*] for the wealthier side of the planet that has modern CPU's:

- PR #3241

[*] MongoDB 5.x has abandoned support for "pre-2011" CPU's that are very common in low-income schools, as summarized here: https://github.com/iiab/iiab/blob/f7d3411df11a64e4a00114ab6051ab28e37fae04/roles/mongodb/tasks/install.yml#L150-L157

holta, 2022-07-19:
MongoDB 6.0 was just released:

We should monitor its official support for Ubuntu 22.04 in coming days...

(...to see if they've made progress supporting libssl3 instead of libssl1.1 ?)

holta, 2022-07-23:
> More active (and official) "Installing mongodb over Ubuntu 22.04" discussion 3-5 days ago:
>
> https://www.mongodb.com/community/forums/t/installing-mongodb-over-ubuntu-22-04/159931

Apparently still no luck &mdash; even with MongoDB 6.0.0 released earlier this week:
https://www.mongodb.com/community/forums/t/mongodb-6-0-0-is-released/175750

(More than 3 months after Ubuntu 22.04 with libssl3 was released April 21st!)

holta, 2022-09-26:
> More active (and official) "Installing mongodb over Ubuntu 22.04" discussion 3-5 days ago:
>
> https://www.mongodb.com/community/forums/t/installing-mongodb-over-ubuntu-22-04/159931

FYI MongoDB (just above) posted:

> MongoDB 4.0.x is no longer supported on any platform. That version reached end of life in April of this year. Check out the [MongoDB Software Lifecycle Schedules 1](https://www.mongodb.com/support-policy/lifecycles) page to see how long each version will be supported for.

https://www.mongodb.com/community/forums/t/installing-mongodb-over-ubuntu-22-04/159931/52

holta, 2022-09-27:
FYI Internet-in-a-box now forcibly installs libssl1.1 on Ubuntu 22.04 or higher &mdash; as a workaround to allow Sugarizer to run on more modern OS's like Mint 21 (Vanessa), Ubuntu 22.10 (Kinetic Kudu), etc:

- PR #3380

This ticket should likely be closed in coming weeks, if no serious issues flare up :leaves: :maple_leaf: :leafy_green: :fallen_leaf: :four_leaf_clover:

closed 2022-10-13 by holta

holta, 2022-10-25:
MongoDB 6.2 apparently will contain a fix (for Ubuntu 22.04's libssl3 presumably?) that may be backported to MongoDB 6.0, according to this 2022-09-29 "official" gossip below:

https://github.com/iiab/iiab/blob/5c9fe083032d804df035804ad1c9acbb9f68eb5a/roles/mongodb/tasks/install.yml#L109-L114

mkbhru, 2023-01-19:
it worked for me on ubuntu 22.04
```
wget -qO - https://www.mongodb.org/static/pgp/server-6.0.asc |  gpg --dearmor | sudo tee /usr/share/keyrings/mongodb.gpg > /dev/null
echo "deb [ arch=amd64,arm64 signed-by=/usr/share/keyrings/mongodb.gpg ] https://repo.mongodb.org/apt/ubuntu jammy/mongodb-org/6.0 multiverse" | sudo tee /etc/apt/sources.list.d/mongodb-org-6.0.list
sudo apt update
sudo apt install mongodb-org
```

and its slightly more secure

mkbhru, 2023-01-19:
> it worked for me on ubuntu 22.04
>
> ```
> wget -qO - https://www.mongodb.org/static/pgp/server-6.0.asc |  gpg --dearmor | sudo tee /usr/share/keyrings/mongodb.gpg > /dev/null
> echo "deb [ [...]

Sorry its installed but now working right now!!!!!!

holta, 2023-01-20:
@mkbhru thanks for the suggestion.  MongoDB 6.0.3 now works with libssl3 OS's like Ubuntu and Debian 12.  So this is now incorporated within Internet-in-a-Box here:

- PR #3469

Let us know details if you face further issues.  Regards.

mkbhru, 2023-02-12:
Thanks @holta for mentioning it :+1:  .

ChristianMingl3, 2023-12-31:
Hello- i am running into this problem on Ubuntu 22.04 as well now - 
TASK [mongodb : Add mongodb.org signing key (only 64-bit available) for MongoDB version 5.0] ***********
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "wget -qO - https://www.mongodb.org/static/pgp/server-5.0.asc | gpg --dearmor > /usr/share/keyrings/mongodb.gpg", "delta": "0:00:00.328933", "end": "2023-12-31 07:15:12.384374", "msg": "non-zero return code", "rc": 2, "start": "2023-12-31 07:15:12.055441", "stderr": "gpg: no valid OpenPGP data found.", "stderr_lines": ["gpg: no valid OpenPGP data found."], "stdout": "", "stdout_lines": []}

TASK [mongodb : SEE ERROR ABOVE (skip_role_on_error: False)] *******************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": ""}

TASK [sugarizer : SEE ERROR ABOVE (skip_role_on_error: False)] *****************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": ""}

PLAY RECAP *********************************************************************************************
127.0.0.1                  : ok=85   changed=13   unreachable=0    failed=1    skipped=29   rescued=2    ignored=0

can anyone give me the walk-through on the work around? or is this meant to have been fixed now?

ChristianMingl3, 2023-12-31:
i am using a Le Potato AML-S905X-CC - i tried installing this on ubuntu 22.04 but was running into this problem... which OS(compatible with Le Potato AML-S905X-CC) should i install that will make the 1 liner work smoothly and not run into this problem?

holta, 2023-12-31:
Hi @ChristianMingl3,

Just FYI Ubuntu 22.04 works on standard machines / architectures (https://wiki.iiab.io/go/FAQ) as demonstrated by:

http://sprunge.us/keWo08

(Apologies I don't know any details about Le Potato AML-S905X-CC and its specific ARM architecture.)

ChristianMingl3, 2023-12-31:
@holta i got the same errors on both ubuntu 22.04 and rasbian bookworm - 
TASK [mongodb : Add mongodb.org signing key (only 64-bit available) for MongoDB version 5.0] ***
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "wget -qO - https://www.mongodb.org/static/pgp/server-5.0.asc | gpg --dearmor > /usr/share/keyrings/mongodb.gpg", "delta": "0:00:00.403903", "end": "2023-12-31 12:43:52.862746", "msg": "non-zero return code", "rc": 2, "start": "2023-12-31 12:43:52.458843", "stderr": "gpg: no valid OpenPGP data found.", "stderr_lines": ["gpg: no valid OpenPGP data found."], "stdout": "", "stdout_lines": []}

TASK [mongodb : SEE ERROR ABOVE (skip_role_on_error: False)] *******************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": ""}

TASK [sugarizer : SEE ERROR ABOVE (skip_role_on_error: False)] *****************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": ""}

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=707  changed=429  unreachable=0    failed=1    skipped=102  rescued=2    ignored=4

holta, 2023-12-31:
@ChristianMingl3

Please consider bypassing Sugarizer, to evaluate everything else you need.

In other words, edit text file `/etc/iiab/local_vars.yml` to set these 2 lines:

```
sugarizer_install: False
sugarizer_enabled: False
```

And then run...

```
sudo IIAB
```

ChristianMingl3, 2023-12-31:
TASK [transmission : Enable & (Re)Start 'transmission-daemon' systemd service, i                         f transmission_enabled] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start servic                         e transmission-daemon: Job for transmission-daemon.service failed because the co                         ntrol process exited with error code.\nSee \"systemctl status transmission-daemo                         n.service\" and \"journalctl -xeu transmission-daemon.service\" for details.\n"}

TASK [transmission : SEE ERROR ABOVE (skip_role_on_error: False)] **************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": ""}

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=86   changed=26   unreachable=0    failed=1    s                         kipped=31   rescued=1    ignored=0

i assume i should do the same thing and disable transmission as well

holta, 2023-12-31:
> i assume i should do the same thing and disable transmission as well

Probably best, yes!

Or, another alternative (on a fresh install anyway) is to set this around Line 324 of /etc/iiab/local_vars.yml :

```
transmission_compile_latest: False
```

ChristianMingl3, 2024-01-01:
Thank you- also, now i get this error: 
TASK [matomo : Start MariaDB] **************************************************
ok: [127.0.0.1]

TASK [matomo : Create MariaDB Database for Matomo] *****************************
changed: [127.0.0.1]

TASK [matomo : Add Admin User to MariaDB Database] *****************************
changed: [127.0.0.1]

TASK [matomo : Download and Extract Matomo (~1 min)] ***************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Failure downloading ht                                                              tps://builds.matomo.org/matomo.tar.gz, Compressed file ended before the end-of-s                                                              tream marker was reached"}

TASK [matomo : SEE ERROR ABOVE (skip_role_on_error: False)] ********************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": ""}

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=730  changed=447  unreachable=0    failed=1    s                                                              kipped=110  rescued=1    ignored=4

so i am disabling that as well - i dont know if that one failed from a network error or something that messed up the download etc, but i didnt need it anyway so i just turned it off and it went fine

ChristianMingl3, 2024-01-06:
@holta sorry to bother again, is wifi_up_down: True meaning that i can both use Wifi to connect to the internet while also creating an Access Point? I am connected to wifi and it gave me this message:
TASK [network : Restart hostapd when WiFi is present but not when using WiFi as                                                                                     gateway with wifi_up_down False] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start servic                                                                                    e hostapd: Job for hostapd.service failed because the control process exited wit                                                                                    h error code.\nSee \"systemctl status hostapd.service\" and \"journalctl -xeu ho                                                                                    stapd.service\" for details.\n"}

is this saying to turn wifi_up_down to False, or that it is on False? I set it to True in the local_vars.yml so i dont think that is the case? 
or is this saying to use ethernet to set it up, but if will allow for wifi use and also create an AP after initial setup?

holta, 2024-01-06:
> is `wifi_up_down: True` meaning that i can both use Wifi to connect to the internet while also creating an Access Point?

Whether the above is possible or not depends on your WiFi hardware and WiFi firmware.  Please more about this:

https://wiki.iiab.io/go/FAQ#Can_I_create_a_Wi-Fi_hotspot_using_an_old_laptop%3F

The above might explain why your `hostapd` service is not restarting here:

https://github.com/iiab/iiab/blob/42d580ef6520b2fba9d5bef78f7977fb50ee49af/roles/network/tasks/restart.yml#L31-L36