iiab/iiab #3199: remoteit role fails on LinuxMint

jvonau, 2022-04-29:
From @shanti-bhardwa 
>TASK [remoteit : Disable & Stop remote.it services {connectd, schannel}] *******
failed: [127.0.0.1] (item=connectd) => {"ansible_loop_var": "item", "changed": false, "item": "connectd", "msg": "Could not find the requested service connectd: host"}
failed: [127.0.0.1] (item=schannel) => {"ansible_loop_var": "item", "changed": false, "item": "schannel", "msg": "Could not find the requested service schannel: host"}
PLAY RECAP *********************************************************************
127.0.0.1                  : ok=42   changed=8    unreachable=0    failed=1    skipped=15   rescued=0    ignored=0

holta, 2022-04-29:
Thank you @shanti-bhardwa and @jvonau for this bug report.

shanti-bhardwa, 2022-04-29:
got this at the end. I rebooted then ran sudo iiab and it said already installed, nothing to do]
Traceback (most recent call last):
  File "/usr/lib/python3.8/urllib/request.py", line 1354, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "/usr/lib/python3.8/http/client.py", line 1256, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1302, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1251, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.8/http/client.py", line 1011, in _send_output
    self.send(msg)
  File "/usr/lib/python3.8/http/client.py", line 951, in send
    self.connect()
  File "/usr/lib/python3.8/http/client.py", line 1425, in connect
    self.sock = self._context.wrap_socket(self.sock,
  File "/usr/lib/python3.8/ssl.py", line 500, in wrap_socket
    return self.sslsocket_class._create(
  File "/usr/lib/python3.8/ssl.py", line 1040, in _create
    self.do_handshake()
  File "/usr/lib/python3.8/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
TimeoutError: [Errno 110] Connection timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/admin/cmdsrv/scripts/get_kiwix_catalog", line 84, in <module>
    url_handle = urllib.request.urlopen(iiab_zim_cat_url)
  File "/usr/lib/python3.8/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.8/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/usr/lib/python3.8/urllib/request.py", line 542, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/usr/lib/python3.8/urllib/request.py", line 502, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.8/urllib/request.py", line 1397, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "/usr/lib/python3.8/urllib/request.py", line 1357, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno 110] Connection timed out>
Getting the Kiwix Catalog Failed.
Please run iiab-get-kiwix-cat again later.

I did run it later on with the folloing result

sb01@box:~$ sudo iiab-get-kiwix-cat
Starting xml download from Kiwix
Reading Our Catalog
Parsing xml downloads from Kiwix
Starting of processing xml download from Kiwix to /etc/iiab/kiwix_catalog.json
Starting of processing our zim catalog into /etc/iiab/kiwix_catalog.json
Ready to write /etc/iiab/kiwix_catalog.json
Finished writing to /etc/iiab/kiwix_catalog.json

holta, 2022-04-29:
remote.it refuses to install on the Linux Mint unfortunately.

We could force it, but possibly it's better to respect their recommendation.

In any case, this is easily resolved one way or the other, and root cause is visible here:

```
root@mint20:~# curl -L https://downloads.remote.it/remoteit/install_agent.sh | sh
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  8332  100  8332   licensekey:   0      0 --:--:-- --:--:-- --:--:--     0
 0     0   162k      0 --:--:-- --:--:-- --:--:--  162k
No R3_LICENSE_KEY or R3_REGISTRATION_CODE variable set for new install so look for CLAIM CODE
* Distribution: Linux

* OS:

Your OS or distribution is not supported by this install script.root@mint20:~#
```

```
root@mint20:~# more /etc/issue
Linux Mint 20.3 Una \n \l
```

```
root@mint20:~# more /etc/os-release
NAME="Linux Mint"
VERSION="20.3 (Una)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 20.3"
VERSION_ID="20.3"
HOME_URL="https://www.linuxmint.com/"
SUPPORT_URL="https://forums.linuxmint.com/"
BUG_REPORT_URL="http://linuxmint-troubleshooting-guide.readthedocs.io/en/latest/"
PRIVACY_POLICY_URL="https://www.linuxmint.com/"
VERSION_CODENAME=una
UBUNTU_CODENAME=focal
```

```
root@mint20:~# more /etc/lsb-release
DISTRIB_ID=LinuxMint
DISTRIB_RELEASE=20.3
DISTRIB_CODENAME=una
DISTRIB_DESCRIPTION="Linux Mint 20.3 Una"
```

holta, 2022-04-29:
Now resolved by:

- PR #3201

So IIAB now has clean & consistent support for remote.it across all our 4 mainline OS's {Raspberry Pi OS, Ubuntu, Mint, Debian} :heavy_check_mark:

closed 2022-04-29 by holta