iiab/iiab #3211: IIAB install fails: "HTTP Error 404: Not Found" with https://download.kiwix.org/release/kiwix-android/kiwix.apk

holta, 2022-05-06:
When `kiwix_incl_apk: True` is set, e.g. with https://github.com/iiab/iiab/blob/master/vars/local_vars_medical.yml the error is:

> TASK [kiwix : Download kiwix.apk to /library/www/html/software/kiwix] **********
fatal: [127.0.0.1]: FAILED! => {"changed": false, "dest": "/library/www/html/software/kiwix", "elapsed": 0, "gid": 0, "group": "root", "mode": "0755", "msg": "Request failed", "owner": "root", "response": "HTTP Error 404: Not Found", "size": 4096, "state": "directory", "status_code": 404, "uid": 0, "url": "https://download.kiwix.org/release/kiwix-android/kiwix.apk"}

As this URL no longer works:

https://github.com/iiab/iiab/blob/1049d94d4c35875974d84b32f8eafc67a532e7e3/vars/default_vars.yml#L473

holta, 2022-05-06:
> https://github.com/iiab/iiab/blob/1049d94d4c35875974d84b32f8eafc67a532e7e3/vars/default_vars.yml#L473

Should the above be replaced with https://download.kiwix.org/release/kiwix-android/kiwix-3.4.5.apk (from 2021-10-09) for now?

holta, 2022-05-07:
@tim-moody can you suggest a recommendation or fix here in coming days?

tim-moody, 2022-05-07:
>Should the above be replaced with https://download.kiwix.org/release/kiwix-android/kiwix-3.4.5.apk (from 2021-10-09) for now?

only answer I can see

holta, 2022-05-07:
Let's re-open this ticket as Kiwix's Emmanuel Engelhart says the apk successor is https://download.kiwix.org/nightly/2022-05-07/dev-universal-debug.apk

'We are working on 3.5.0 and this is almost over'

Hopefully that will be posted to https://download.kiwix.org/release/kiwix-android/ within the month, and IIAB can then encode that link at that time?

(Certainly the above "nightly" link will disappear after ~30 days!)

holta, 2022-07-20:
Good enough:

- PR #3310
- PR #3311

closed 2022-07-20 by holta