iiab/iiab #3227: How do we fix Kolibri Channel Import failures with http://box/kolibri configured by KOLIBRI_URL_PATH_PREFIX=/kolibri ?

holta, 2022-05-20:
Thanks to @shanti-bhardwa who helped spell out this longstanding issue, and also @jvonau, @tim-moody who both reconfirmed!

_The solution might not be simple (?) but let's try to find a way forward here together &mdash; as there are a quickly growing number of Internet-in-a-Box communities around the world using Kolibri:_

PROBLEM SUMMARY: If you try to import a Kolibri channel using Kolibri > Channels > Import > Select a source > "Local network or internet", it will fail as **the other Kolibri machine's** `KOLIBRI_URL_PATH_PREFIX=/kolibri` is somehow being ignored:

- The problem occurs when The Other Kolibri Machine has the form http://box/kolibri
- Whereas if The Other Kolibri Machine is manually hacked to instead run Kolibri on http://box:8009, the channel import then works.

BACKGROUND: `KOLIBRI_URL_PATH_PREFIX=/kolibri` is installed within [/etc/systemd/system/kolibri.service](https://github.com/iiab/iiab/blob/master/roles/kolibri/templates/kolibri.service.j2) whenever you install [Internet-in-a-Box](https://internet-in-a-box.org), as seen on Line 10 below.

```
[Unit]
Description=Kolibri

[Service]
Type=forking
RemainAfterExit=yes
Environment=KOLIBRI_USER=kolibri
Environment=KOLIBRI_HOME=/library/kolibri
Environment=KOLIBRI_HTTP_PORT=8009
Environment=KOLIBRI_URL_PATH_PREFIX=/kolibri
User=kolibri
Group=www-data
# 2020-10-03: Kolibri was timing out on RaspiOS & Ubuntu 20 NUC: iiab/iiab#2555
TimeoutStartSec=1200
# The following is the systemd default, which is too much for most teachers in
# low-electricity environments (30-60 sec is about all they can handle before
# pulling the power cord, typically every hour at the end of class).  But since
# 90 seconds is the Linux / systemd default, it's prob good enough for now:
# TimeoutStopSec=90
# 2020-04-18 @jvonau: comment out both timeouts for now, in favor of 90 seconds
# or whatever systemd / Kolibri favor? https://github.com/iiab/iiab/issues/2318
# TimeoutStartSec=infinity
# TimeoutStopSec=10
ExecStart=/usr/bin/kolibri start
ExecStop=/usr/bin/kolibri stop

[Install]
WantedBy=multi-user.target
```

GENERAL KOLIBRI QUESTION (PERHAPS UNRELATED?) `/etc/kolibri/daemon.conf` currently always contains the single line `KOLIBRI_HOME="/library/kolibri"` on Internet-in-a-Box machines, so a suggestion was made &mdash; _should this really contain all 4 variables above including `KOLIBRI_URL_PATH_PREFIX=/kolibri` ?_  As background, here's the current /etc/kolibri configuration of all Internet-in-a-Box machines:

```
root@box:/etc/kolibri# ls
conf.d  daemon.conf  README  username

root@box:/etc/kolibri# ls conf.d

root@box:/etc/kolibri# more username
kolibri

root@box:/etc/kolibri# more daemon.conf
KOLIBRI_HOME="/library/kolibri"
```

CONTEXT: Internet-in-a-Box [installs](https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/install.yml#L34-L39) Kolibri's latest .deb (https://learningequality.org/r/kolibri-deb-latest) according to the instructions provided by many helpful Learning Equality staff over the years (e.g. Benjamin Bach and others!)  Then it runs...

```
export KOLIBRI_HOME=/library/kolibri &&
/usr/bin/kolibri manage provisiondevice \
--facility Kolibri-in-a-Box \
--superusername Admin \
--superuserpassword changeme \
--preset formal \
--language_id en
```

...according to https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/install.yml#L70-L80

DOC: https://kolibri.readthedocs.io/en/latest/manage/resources.html#import-from-a-local-network-or-internet

Tangentially related:

- learningequality/kolibri#5414
  - PR learningequality/kolibri#5415
- PR learningequality/kolibri#8164
- PR learningequality/kolibri#8366

jvonau, 2022-05-25:
Close https://github.com/learningequality/kolibri/issues/9455#issuecomment-1136062000

holta, 2022-05-25:
If Kolibri is able to solve this via learningequality/kolibri#9455 that would certainly be best ✅ 

We should continue to monitor evolution here regardless (hopefully almost everything is solved by/around July!)

Pragmatic + Friendly content sync'ing (in varying field conditions) is often a Much harder problem than "in the lab" as operators need a clean UX/UI even in the best of Third World Conditions.

Possibly another ticket needs to be opened on this important/broad topic...

holta, 2022-07-20:
@shanti-bhardwa @andrewclark5

Great news https://github.com/learningequality/kolibri/releases/tag/v0.15.6 was released (2 days early in fact!) and *should* hopefully fix peer importing of Kolibri channels, i.e. from "Local network or internet".

(Let's close this ticket as soon as that's verified!)

holta, 2022-08-15:
Appears fixed!  Thanks @rtibbles here:

- https://github.com/learningequality/kolibri/issues/9533

closed 2022-08-15 by holta