iiab/iiab #3231: Admin Console's GET-SPACE-AVAIL sometimes fails

tim-moody, 2022-05-30:
This caused Linux Mint 20.3 to become unstable - kept refreshing the Desktop.

This issue seems to be related to disk space.

But in my case the 16GB SSD running the O/S and iiab had 2.2GB free space left on it.

However there was also an external USB carrying the library files which had 0 bytes left on it.  But I don't see why after freeing up some space the issue was still there after a reboot?

[image]

_Originally posted by @andrewclark5 in https://github.com/iiab/iiab/issues/3106#issuecomment-1139715863_

tim-moody, 2022-05-30:
https://github.com/iiab/iiab/issues/3106#issuecomment-1140787150

tim-moody, 2022-05-30:
https://github.com/iiab/iiab/issues/3106#issuecomment-1141095253

tim-moody, 2022-05-30:
https://github.com/iiab/iiab/issues/3106#issuecomment-1140529868

tim-moody, 2022-05-30:
GET-STORAGE-INFO used lsblk and was notoriously unreliable when parsed. GET-SPACE-AVAIL is its replacement and only uses df. The former is still available in utilities, but is just displayed and not parsed.

tim-moody, 2022-05-30:
As far as I know df only generates an error on Mint and only there if the desktop is used as root. I am not clear if double mounting will cause a problem for parsing df output, but I would avoid it.

tim-moody, 2022-05-30:
I tried df -a and df -m. Both gave the same error.

tim-moody, 2022-06-02:
df -x fuse.portal may solve the problem. put into LinuxMint.yml

jvonau, 2022-06-02:
`mount | grep portal` to check

holta, 2022-06-02:
> df -x fuse.portal may solve the problem. put into [Linux%20Mint.yml](https://github.com/iiab/iiab-admin-console/blob/master/vars/Linux%20Mint.yml)

> `mount | grep portal` to check

Thanks @andrewclark5 for helping us get much closer to solving this &mdash; during IIAB's community call (http://minutes.iiab.io) over the past 2 h.

jvonau, 2022-06-02:
https://github.com/codemicro/bar/commit/5675c40295d6edeeb742786f129cb609cc1508e1

tim-moody, 2022-06-02:
>Tangentially related:

one has to do with adm cons and the other with menuing, one with df and the other with ajax. why clutter up this issue with that one?

holta, 2022-06-02:
> one has to do with adm cons and the other with menuing

Yes.  These are the 2 issues @andrewclark5 care about most for Malawi and that's noteworthy for some of us.

holta, 2022-06-04:
@andrewclark5 FYI this `df` workaround was merged 35min ago:

- PR iiab/iiab-admin-console#495

_So if you install the very latest IIAB 8.0 pre-release onto Mint 20.3, we strongly suspect the problem is now solved._

holta, 2022-06-11:
Progress is ongoing:

- #3247
- iiab/iiab-admin-console#502
- PR iiab/iiab-admin-console#503

tim-moody, 2022-06-21:
https://github.com/iiab/iiab-admin-console/pull/503 merged

closed 2022-06-21 by tim-moody