iiab/iiab #3236: Sugarizer won't start due to "Illegal instruction /usr/bin/mongod --repair --dbpath /library/dbdata/mongodb"

holta, 2022-06-06:
Thanks to @nzola who uncovered this:

```
1958 Jun  6 13:40:18 raspberrypi iiab-mongodb-repair-if-no-lock[76019]: /usr/bin/iiab-mongodb-repair-if-no-lock: line 7: 76020 Illegal instruction     /usr/bin/mongod --repair --dbpath /library/dbdata/mongodb
1959 Jun  6 13:40:18 raspberrypi systemd[1]: mongodb.service: Control process exited, code=exited, status=132/n/a
1960 Jun  6 13:40:18 raspberrypi systemd[1]: mongodb.service: Failed with result 'exit-code'.
1961 Jun  6 13:40:18 raspberrypi systemd[1]: Failed to start High-performance, schema-free document-oriented database.
```

From http://sprunge.us/LbT5zf?en on Raspberry Pi OS on RPi4, but I assume this applies to all OS's.

Likely as a result of MongoDB 5.x (released 11 months ago) which presumably no longer supports the legacy syntax / command used by [iiab-mongodb-repair-if-no-lock](https://github.com/iiab/iiab/blob/master/roles/mongodb/templates/iiab-mongodb-repair-if-no-lock.j2) :

- PR #3214

In any case, with MongoDB 6.0 now imminent, it's time for IIAB to catch up.  I'll investigate.

holta, 2022-06-06:
Just FYI IIAB's 241-line [template for /etc/mongod.conf](https://github.com/iiab/iiab/blob/master/roles/mongodb/templates/mongod.conf.j2) is archaic to the point where it's become highly misleading.

MongoDB 5.0 rewrites it as this 42-line file: _(which we might consider as IIAB's future template)_

```
# mongod.conf

# for documentation of all options, see:
#   http://docs.mongodb.org/manual/reference/configuration-options/

# Where and how to store data.
storage:
  dbPath: /var/lib/mongodb
  journal:
    enabled: true
#  engine:
#  wiredTiger:

# where to write logging data.
systemLog:
  destination: file
  logAppend: true
  path: /var/log/mongodb/mongod.log

# network interfaces
net:
  port: 27017
  bindIp: 127.0.0.1

# how the process runs
processManagement:
  timeZoneInfo: /usr/share/zoneinfo

#security:

#operationProfiling:

#replication:

#sharding:

## Enterprise-Only Options:

#auditLog:

#snmp:
```

holta, 2022-06-07:
A couple more FYIs:

1) As mentioned above, IIAB's `/etc/mongod.conf` needs to be fixed and cleaned up &mdash; even if that's not quite enough to solve this problem @nzola identified (#3236).

2) MongoDB's `mongod.lock` file is very poorly named, after all these years it's really a misnomer.  At this point, a clean shutdown of MingoDB **does not delete the file.**  Nor does existence of the file prevent starting MongDB!

   Rather: a clean shutdown of MongoDB removes the PID from this file (i.e. setting the file length to zero, but not deleting the file).  _Long Story Short: in recent versions of MongoDB (and certainly MongoDB 5.0.9) this file is really just about storing MongoDB's process ID._

   So IIAB's [/usr/bin/iiab-mongodb-repair-if-no-lock](https://github.com/iiab/iiab/blob/master/roles/mongodb/templates/iiab-mongodb-repair-if-no-lock.j2) (at the core of this bug) needs to be redesigned regardless!

3) I was wrong about this problem (#3236) affecting all OS's in the same way.  While it does affect 64-bit Raspberry Pi OS on RPi 4 (I'll investigate that next) it does not appear to affect 64-bit Ubuntu 20.04 &mdash; even after rebooting etc &mdash; example output from `journalctl -u sugarizer.service` here:

   ```
   Jun 06 18:21:24 box systemd[1]: Started Node.js Sugarizer Server.
   Jun 06 18:21:27 box sugarizer[700]:    _____                        _
   Jun 06 18:21:27 box sugarizer[700]:   / ____|                      (_)
   Jun 06 18:21:27 box sugarizer[700]:  | (___  _   _  __ _  __ _ _ __ _ _______ _ __
   Jun 06 18:21:27 box sugarizer[700]:   \___ \| | | |/ _` |/ _` | '__| |_  / _ \ '__|
   Jun 06 18:21:27 box sugarizer[700]:   ____) | |_| | (_| | (_| | |  | |/ /  __/ |
   Jun 06 18:21:27 box sugarizer[700]:  |_____/ \__,_|\__, |\__,_|_|  |_/___\___|_|
   Jun 06 18:21:27 box sugarizer[700]:                 __/ |
   Jun 06 18:21:27 box sugarizer[700]:                |___/
   Jun 06 18:21:27 box sugarizer[700]: Server for Sugarizer application v1.4.0
   Jun 06 18:21:27 box sugarizer[700]: node v18.3.0
   Jun 06 18:21:27 box sugarizer[700]: Settings file './env/sugarizer.ini'
   Jun 06 18:21:27 box sugarizer[700]: Presence is listening on port 8039...
   Jun 06 18:21:27 box sugarizer[700]: API is listening on port 8089...
   ```

holta, 2022-06-07:
A couple more FYI's &mdash; that we might as well take into account:

1) MongoDB's new shutdown process (changed as of MongoDB 5.0) no longer works well with IIAB's `mongodb.service` [killall](https://github.com/iiab/iiab/blob/4ddc7c4461f00a7c1dd0a3170f71b306cae264f8/roles/mongodb/templates/mongodb.service.j2#L15) procedure:

   ```
   Jun 06 19:36:08 box killall[2986]: mongod: no process found
   Jun 06 19:36:08 box systemd[1]: mongodb.service: Control process exited, code=exited, status=1/FAILURE
   Jun 06 19:36:08 box systemd[1]: mongodb.service: Failed with result 'exit-code'.
   ```

2) MongoDB's shutdown now (claims to) implement a 15-second ["quiesce"](https://en.wikipedia.org/wiki/Quiesce) period by default, as explained here:

   - https://www.mongodb.com/docs/manual/reference/method/db.shutdownServer/
   - https://www.mongodb.com/docs/v5.3/reference/command/shutdown/#std-label-shutdown-cmd-timeoutSecs

holta, 2022-06-07:
Another FYI &mdash; and an important one near the very core &mdash; towards cleaning up all these bugs:

- On 64-bit OS's with MongoDB 4.x until a month ago (and likewise with MongoDB 5.x since a month ago) IIAB's [/usr/bin/iiab-mongodb-repair-if-no-lock](https://github.com/iiab/iiab/blob/master/roles/mongodb/templates/iiab-mongodb-repair-if-no-lock.j2) (at the core of this bug) is working with the **wrong database** (`/library/dbdata/mongodb/` &mdash; instead of `/var/lib/mongodb/` specified in `/etc/mongod.conf`).

   Which means the `mongo.lock` file it's working with is doubly irrelevant[*] (it's working with the wrong directory!) so it's more than a bit surprising MongoDB works at all with Sugarizer on OS's like Ubuntu 20.04 (!!)

[*] Recap: `mongod.lock` is NOT really a lock file, despite its name &mdash; as I've re-confirmed with MongoDB 4.4.14 on RPi 4 &mdash; just like with MongoDB 5.0.9 on Ubuntu 20.04: https://github.com/iiab/iiab/issues/3236#issuecomment-1148119542

holta, 2022-06-07:
Returning to the core of this bug report:

1) MongoDB 5.0.9 simply does not run on Raspberry Pi 4:

   ```
   root@box:~# /usr/bin/mongod -f /etc/mongod.conf
   Illegal instruction

   root@box:~# uname -a
   Linux box 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
   ```

2) The underlying issue is MongoDB 5.x not being compiled for RPi 4:

   > I’ve tracked down this issue to a change in the [arm64 microarchitecture requirements 37](https://docs.mongodb.com/manual/release-notes/5.0/#minimum-microarchitecture) for MongoDB 5.0 packages. MongoDB 5.0 requires ARM v8.2-A or later and the Raspberry Pi 4 uses an [ARM Cortex-A72 2](https://en.wikipedia.org/wiki/ARM_Cortex-A72) which is ARM v8-A.
   > 
   > Unfortunately this means the pre-built packages for MongoDB 5.0 will not support Raspberry Pi 4.
   > 
   > However, you should be able to build from source by adding CCFLAGS-march=armv8 to the SCons invocation. See [Building MongoDB 123](https://github.com/mongodb/mongo/blob/master/docs/building.md) for full instructions.
   > 
   > If building from source is a daunting prospect, an alternative would be to install the latest ARM 64 package of MongoDB 4.4.

https://www.mongodb.com/community/forums/t/core-dump-on-mongodb-5-0-on-rpi-4/115291/14

3) Andy Felong pushed this further in January 2022, compiling MongoDB 5.0.5 binaries for Raspberry Pi 4:

   - https://andyfelong.com/downloads/raspbian_mongodb_5.0.5.gz
   - https://andyfelong.com/2021/08/mongodb-4-4-under-raspberry-pi-os-64-bit-raspbian64/

holta, 2022-06-07:
@nzola this PR below has been merged &mdash; to help you get back on track: :steam_locomotive:

- PR #3238 

(While likely not the final word, please let us know if this at least gets Sugarizer working for you on Raspberry Pi 4, Thanks!)

holta, 2022-06-07:
I'm thinking of keeping this ticket open for a few more days or even longer?

In Short: all looks good &mdash; thanks to @nzola and @tim-moody (iiab/iiab-admin-console#499) helping with testing on RPi 4 &mdash; but further improvements might well make sense!

nzola, 2022-06-07:
I will more testing when I return home tonight. I will try to use Sugarizer, get the results and then I will wipe up the SD card to re-install iiab from scratch. I would like to see it run through all the way to the end without any errors.

holta, 2022-06-23:
Good enough.

Resolved by:

- PR #3238

closed 2022-06-23 by holta