iiab/iiab #3242: > > This protects the user when first installing, but does not protect later changes done by admin-console

jvonau, 2022-06-08:
> > This protects the user when first installing, but does not protect later changes done by admin-console
> 
> Yes.
> 
> This was done on purpose so implementers who really want to can try again 10/20/30 days later — if they really want to try installing apps/services.
> 
> Using either `sudo ./runrole <app>` or Admin Console.

This 'workaround' has effectively broken the preset functionality other than for the medical one.
>ubuntu@exclusive-limpkin:/opt/iiab/iiab-admin-console$ grep next roles/cmdsrv/files/presets/*/vars.yml
roles/cmdsrv/files/presets/en-school-256-base/vars.yml:nextcloud_install: True
roles/cmdsrv/files/presets/en-school-256-base/vars.yml:nextcloud_enabled: True
roles/cmdsrv/files/presets/en-school/vars.yml:nextcloud_install: True
roles/cmdsrv/files/presets/en-school/vars.yml:nextcloud_enabled: True
roles/cmdsrv/files/presets/es-school/vars.yml:nextcloud_install: True
roles/cmdsrv/files/presets/es-school/vars.yml:nextcloud_enabled: True
roles/cmdsrv/files/presets/fr-school/vars.yml:nextcloud_install: True
roles/cmdsrv/files/presets/fr-school/vars.yml:nextcloud_enabled: True
 
>ubuntu@exclusive-limpkin:/opt/iiab/iiab-admin-console$ grep sugar roles/cmdsrv/files/presets/*/vars.yml
roles/cmdsrv/files/presets/en-school/vars.yml:sugarizer_install: True
roles/cmdsrv/files/presets/en-school/vars.yml:sugarizer_enabled: True
roles/cmdsrv/files/presets/es-school/vars.yml:sugarizer_install: True
roles/cmdsrv/files/presets/es-school/vars.yml:sugarizer_enabled: True

And sets up the user for a fail should the user attempt to use ICO with a medium or larger install without even changing anything.

[image]

ubuntu@exclusive-limpkin:/opt/iiab/iiab$ `cat adm-run-roles-tmp.yml`
```
---
- hosts: all
  become: yes

  vars_files:
  - vars/default_vars.yml
  - vars/{{ ansible_local.local_facts.os_ver }}.yml
  - /etc/iiab/local_vars.yml
  - /etc/iiab/iiab_state.yml

  roles:
    - { role: 0-init }
    - { role: kolibri }
    - { role: nextcloud }
    - { role: sugarizer }
    - { role: network }
```

_Originally posted by @jvonau in https://github.com/iiab/iiab/issues/3189#issuecomment-1114501001_

holta, 2022-06-08:
On the bright side Nextcloud (PHP 8.1 issue) and Kolibri (Python 3.10 issue) have thankfully now both been solved &mdash; each now works well on Ubuntu 22.04.

Still remaining are MongoDB (libssl1.1 -> libssl3 issue) and Moodle (PHP 8.1 issue) which are still not yet solved upstream.  Which is somewhat surprising that such incredibly famous packages still do not work on Ubuntu 22.04 6 weeks later, but hopefully they will update us on their progress shortly:

- #3189
- #3190

jvonau, 2022-06-08:
Think it is a simple as calling GET-VARS prior to running ICO, the data is stale.

tim-moody, 2022-06-08:
>This protects the user

I actually have no idea what the issue is here. I assume we are talking about keeping ansible from running roles that are not supported on a given platform. IMHO putting patches in all the callers is not the right approach.

>Think it is a simple as calling GET-VARS prior to running ICO, the data is stale.

This sounds like a comment on #3243

jvonau, 2022-06-08:
> Rather than making every invocation of ansible try to figure out whether a given role can run in the current environment it would be better if roles had conditionals that cause them to exit and report when they are incompatible with the current environment. 

Isn't that exactly what valiadate.yml does, pre-check recommended user editable files for errors prior to reaching the role?

jvonau, 2022-06-08:
> > This protects the user
> 
> I actually have no idea what the issue is here. I assume we are talking about keeping ansible from running roles that are not supported on a given platform. IMHO putting patches in all the callers is not the right approach.
>

Given 'the fix' is currently in the meta (6-7) stages' main.yml files and the move away from iiab-from-console.yml there is no common ground between 'iiab' and 'admin-console', that leaves 0-init as the common place to react to the current environment beteen the two code paths.

> > Think it is a simple as calling GET-VARS prior to running ICO, the data is stale.
> 
> This sounds like a comment on #3243

Yup, different issue.

tim-moody, 2022-06-08:
>Isn't that exactly what valiadate.yml does

I didn't write that and have never cared for it, but it is not quite the same. It checks for the internal consistency of global variables, not whether they are valid for a given environment, or at least that's what I thought it did and it certainly should not do more than that.

Imagine I have f2, f3, f4, ... that call f1 with parameters p1, p2. I would not support an approach that made every caller check that f1 could handle the parameters. That is the job of f1.

tim-moody, 2022-06-08:
>'the fix' is currently in the meta (6-7) stages' main.yml 

sounds like it isn't a fix, either for Adm Console or runrole.

It's really quite simple. If a role can't run in a given environment it should simply exit and say so. I also don't think it has to kill the whole ansible run.

jvonau, 2022-06-08:
> > 'the fix' is currently in the meta (6-7) stages' main.yml
> 
> sounds like it isn't a fix, either for Adm Console or runrole.
>

That is my point...
 
> It's really quite simple. If a role can't run in a given environment it should simply exit and say so. I also don't think it has to kill the whole ansible run.

A silent skip is OK? That could be do-able with a simple override set_facts within validate.yml as the common ground in place of the meta stages.

holta, 2022-06-23:
This issue (#3242) is resolved by:

- PR #3255
- PR #3256
- PR #3262

More work is needed for:

- #3182
- #3190
- #3244

closed 2022-06-23 by holta

jvonau, 2022-06-29:
This issue remains just defers and papers over the issue in the hopes that a git pull is not required in the future to unlock the failing roles.