iiab/iiab #3243: Think this might be an admin-console issue, the tick box uncheck should of only written 'enabled False' and left as 'install True' in local_vars.yml

jvonau, 2022-06-08:
Going back to un-tick kolibri, save, ico.

[image]

2022-05-02 00:15:21,252 p=92605 u=root n=ansible | failed: [127.0.0.1] (item=kolibri) => {"ansible_loop_var": "item", "assertion": "kolibri_install or kolibri_installed is undefined", "changed": false, "evaluated_to": false, "item": "kolibri", "msg": "DISALLOWED: 'kolibri_install: False' (e.g. in /etc/iiab/local_vars.yml) WHEN 'kolibri_installed' is defined (e.g. in /etc/iiab/iiab_state.yml) -- IIAB DOES NOT SUPPORT UNINSTALLS -- please verify those 2 files especially, and other places variables are defined?"}

Think this might be an admin-console issue, the tick box uncheck should of only written 'enabled False' and left as 'install True' in local_vars.yml

ubuntu@exclusive-limpkin:/opt/iiab/iiab$ cat adm-run-roles-tmp.yml
```
---
- hosts: all
  become: yes

  vars_files:
  - vars/default_vars.yml
  - vars/{{ ansible_local.local_facts.os_ver }}.yml
  - /etc/iiab/local_vars.yml
  - /etc/iiab/iiab_state.yml

  roles:
    - { role: 0-init }
    - { role: nextcloud }
    - { role: sugarizer }
    - { role: network }
```

_Originally posted by @jvonau in https://github.com/iiab/iiab/issues/3189#issuecomment-1114509740_

jvonau, 2022-06-08:
What I think occurs here is that reading of iiab-state.yml is not refreshed prior to the second run of ansible and has written data based on the first run. In this case Kolibri was not initially installed(Install False), then the first run of ansible in admin-console installed kolibri(install: True), the second run of admin-console should of seen that kolibri was installed but did not and wrote data based on the first run (install: False).

tim-moody, 2022-06-08:
>What I think occurs here is that reading of iiab-state.yml is not refreshed prior to the second run of ansible and has written data based on the first run.

Not sure what the point is. Vars are not set when ansible is run; they are set prior to running ansible. So getting state before running ansible would have no impact.

When Adm Cons saves the state of the vars it has to deal with what to do when a previously checked service is unchecked, and it tries to decide whether it checked install or inherited install. The first thing it does is read the state file. If the service is installed it assumes it inherited the install true and leaves it. Otherwise it sets it to false.

What is not being done I think is rejecting a save vars if ansible is currently running, so there could be a race condition.

jvonau, 2022-06-09:
> Not sure what the point is. Vars are not set when ansible is run; they are set prior to running ansible. So getting state before running ansible would have no impact.

Something is not firing to pick up the change to iiab_state,yml that was part of installing kolibri in this case.
> When Adm Cons saves the state of the vars it has to deal with what to do when a previously checked service is unchecked, and it tries to decide whether it checked install or inherited install. The [...]

I have demonstrated above that is not always the case. 

> What is not being done I think is rejecting a save vars if ansible is currently running, so there could be a race condition.

There is no race condition, the logging above clearly shows 20 minutes in between the 2 runs of ansible.

tim-moody, 2022-06-09:
tell me what to do to reproduce the problem

tim-moody, 2022-06-09:
>There is no race condition, the logging above clearly shows 20 minutes in between the 2 runs of ansible.

the two that both failed?

jvonau, 2022-06-09:
> > There is no race condition, the logging above clearly shows 20 minutes in between the 2 runs of ansible.
> 
> the two that both failed?

The first failure at restarting the service of the role, the role did in fact get installed and wrote 'installed: True' to iiab_state.yml.

> I believe this is the intended behavior

It is, but not what was observed above when ICO was pressed after Save Configuration, did you try to run ansible after the second save?

tim-moody, 2022-06-09:
>did you try to run ansible after the second save?

No. what specifically would I be looking for if I did that?

tim-moody, 2022-06-09:
ran it. what were you expecting?

```
PLAY RECAP *********************************************************************
127.0.0.1 : ok=102 changed=23 unreachable=0 failed=0 skipped=106 rescued=0 ignored=0

```
```
root@box:~# cat /etc/iiab/local_vars.yml
# turn off defaults
remoteit_install: False
openvpn_install: False
kalite_install: False
kalite_enabled: False
kiwix_install: True
kiwix_enabled: False
osm_vector_maps_install: False
awstats_install: False
awstats_enabled: False
captiveportal_install: False

############################################################################
# IIAB -- following variables are first set by browser via the Admin Console
#  They may be changed via text editor, or by the Admin Console.

gui_desired_network_role: Appliance
gui_static_wan: False
gui_static_wan_ip: 192.168.3.162
gui_static_wan_netmask: 255.255.255.0
gui_static_wan_gateway: 192.168.3.1
gui_static_wan_nameserver: 192.168.3.1
dansguardian_enabled: False

```

tim-moody, 2022-06-24:
iiab/iiab-admin-console#504 has two parts:

- block changing local vars if ansible is running
- reread install state (including state.yml) on the front end before calculating which vars to send to the backend

The first was done because changing local vars while ansible is running seems like a bad idea, but I am not sure what if any impact it has, and it was not mentioned in the writeup of this issue. .

The second was done because the front end assumes variables sent to the server were set and so could become out of sync with the backend if Save Configuration is run without refreshing adm cons, resulting in a changed variable not being resent. This is the opposite of the problem in this issue.

Since I am unable to reproduce the reported problem by the means reported, I don't know if it is a problem and what would fix it.

The above attempt to reproduce was done with Kiwix. I will repeat with Kolibri on the chance it is a special case.

tim-moody, 2022-06-24:
Initial state:

```
root@box:/opt/iiab/iiab-admin-console# cat   /etc/iiab/local_vars.yml
# turn off defaults
remoteit_install: False
openvpn_install: False
kalite_install: True
kalite_enabled: True
kiwix_install: True
kiwix_enabled: True
osm_vector_maps_install: False
awstats_install: False
awstats_enabled: False
captiveportal_install: False

############################################################################
# IIAB -- following variables are first set by browser via the Admin Console
#  They may be changed via text editor, or by the Admin Console.

gui_desired_network_role: Appliance
gui_static_wan: False
gui_static_wan_ip: 192.168.3.162
gui_static_wan_netmask: 255.255.255.0
gui_static_wan_gateway: 192.168.3.1
gui_static_wan_nameserver: 192.168.3.1
dansguardian_enabled: False

root@box:/opt/iiab/iiab-admin-console# cat   /etc/iiab/iiab_state.yml
sshd_installed: True
iiab_admin_installed: True
mysql_installed: True
nginx_installed: True
www_base_installed: True
pylibs_installed: True
usb_lib_installed: True
www_options_installed: True
kiwix_installed: True

```

tim-moody, 2022-06-24:
now check Kolibri and save

root@box:/opt/iiab/iiab-admin-console# cat   /etc/iiab/local_vars.yml
# turn off defaults
remoteit_install: False
openvpn_install: False
kalite_install: True
kalite_enabled: True
kiwix_install: True
kiwix_enabled: True
osm_vector_maps_install: False
awstats_install: False
awstats_enabled: False
captiveportal_install: False

############################################################################
# IIAB -- following variables are first set by browser via the Admin Console
#  They may be changed via text editor, or by the Admin Console.

gui_desired_network_role: Appliance
gui_static_wan: False
gui_static_wan_ip: 192.168.3.162
gui_static_wan_netmask: 255.255.255.0
gui_static_wan_gateway: 192.168.3.1
gui_static_wan_nameserver: 192.168.3.1
dansguardian_enabled: False
kolibri_enabled: True
kolibri_install: True

root@box:/opt/iiab/iiab-admin-console# cat   /etc/iiab/iiab_state.yml
sshd_installed: True
iiab_admin_installed: True
mysql_installed: True
nginx_installed: True
www_base_installed: True
pylibs_installed: True
usb_lib_installed: True
www_options_installed: True
kiwix_installed: True

tim-moody, 2022-06-24:
now rerun ICO (succeeded)

```
root@box:/opt/iiab/iiab-admin-console# cat   /etc/iiab/local_vars.yml
# turn off defaults
remoteit_install: False
openvpn_install: False
kalite_install: True
kalite_enabled: True
kiwix_install: True
kiwix_enabled: True
osm_vector_maps_install: False
awstats_install: False
awstats_enabled: False
captiveportal_install: False

############################################################################
# IIAB -- following variables are first set by browser via the Admin Console
#  They may be changed via text editor, or by the Admin Console.

gui_desired_network_role: Appliance
gui_static_wan: False
gui_static_wan_ip: 192.168.3.162
gui_static_wan_netmask: 255.255.255.0
gui_static_wan_gateway: 192.168.3.1
gui_static_wan_nameserver: 192.168.3.1
dansguardian_enabled: False
kolibri_enabled: False
kolibri_install: True

root@box:/opt/iiab/iiab-admin-console# cat   /etc/iiab/iiab_state.yml
sshd_installed: True
iiab_admin_installed: True
mysql_installed: True
nginx_installed: True
www_base_installed: True
pylibs_installed: True
usb_lib_installed: True
www_options_installed: True
kiwix_installed: True
kalite_installed: True
kolibri_installed: True

```
[image]

jvonau, 2022-06-24:
> [iiab/iiab-admin-console#504](https://github.com/iiab/iiab-admin-console/pull/504) has two parts:
>
>     * block changing local vars if ansible is running
>
>     * reread install state (including [...]

Can you clarify 'without refreshing adm cons'? As in logout/login, hit reload current page in the browser(which re-prompts you for id/password), or something else? After the first failure I didn't 'refresh' just unchecked the failed role and hit SCO and then ran anisble the second time.
> This is the opposite of the problem in this issue

Seems like that explains what was the problem in this issue. Doesn't matter now the play won't stop now, one has to dig through the logs to see how the role failed to cause a rescue situation.

tim-moody, 2022-06-24:
>Can you clarify 'without refreshing adm cons'? As in logout/login, hit reload current page in the browser(which re-prompts you for id/password), or something else?

yes, but only relevant if save failed. I had to introduce this after I allowed the back end to reject a save if ansible was running.

jvonau, 2022-06-24:
In the above [comment](https://github.com/iiab/iiab/issues/3243#issuecomment-1150173421) the ICO didn't fail, that was used post ansible run that did fail and I didn't 'refresh'. Having to logout/login to 'refresh' seem less than ideal, some users might add services one at a time and forget to do the logout/login dance. Given the time passed since May 5 other code revisions might be in play here. I'll open a new issue if something like this crops up again.

closed 2022-06-24 by jvonau