iiab/iiab #3244: I'm just pointing out that adm-console does not use these 2 files that are modified with this PR making things extremely fragile within the console until these workarounds goes away.

jvonau, 2022-06-08:
When the upstream get their stuff in line most of these issues go away, I'm just pointing out that adm-console does not use these 2 files that are modified with this PR making things extremely fragile within the console until these workarounds goes away.

Maybe something like this in validate might be welcome?
```
- name: 'DISALLOW "XYZ_install: True" WHEN is_ubuntu_2204'
  assert:
    that: not {{ item }}_install and is_ubuntu_2204
    fail_msg: "DISALLOWED: '{{ item }}_install: True' (e.g. in /etc/iiab/local_vars.yml) WHEN is_ubuntu_2204"
    quiet: yes
  with_items:
    - nextcloud
    - kolibri
    - moodle
    - sugarizer

- name: 'DISALLOW "XYZ_install: True" WHEN is_raspbian'
  assert:
    that: not {{ item }}_install and not is_raspbain and not ansible_architecture == "aarch64"
    fail_msg: "DISALLOWED: '{{ item }}_install: True' (e.g. in /etc/iiab/local_vars.yml) WHEN 32bit RaspOS is used"
    quiet: yes
  with_items:
    - calibre
```
[image]

```
TASK [0-init : DISALLOW "XYZ_install: True" WHEN is_ubuntu_2204] ***************
ok: [127.0.0.1] => (item=nextcloud)
failed: [127.0.0.1] (item=kolibri) => {"ansible_loop_var": "item", "assertion": "not kolibri_install and is_ubuntu_2204", "changed": false, "evaluated_to": false, "item": "kolibri", "msg": "DISALLOWED: 'kolibri_install: True' (e.g. in /etc/iiab/local_vars.yml) WHEN is_ubuntu_2204"}
ok: [127.0.0.1] => (item=moodle)
failed: [127.0.0.1] (item=sugarizer) => {"ansible_loop_var": "item", "assertion": "not sugarizer_install and is_ubuntu_2204", "changed": false, "evaluated_to": false, "item": "sugarizer", "msg": "DISALLOWED: 'sugarizer_install: True' (e.g. in /etc/iiab/local_vars.yml) WHEN is_ubuntu_2204"}

PLAY RECAP *********************************************************************
127.0.0.1 : ok=13 changed=0 unreachable=0 failed=1 skipped=2 rescued=0 ignored=0 
```
The whole error message is visible within the slider to view the results.

_Originally posted by @jvonau in https://github.com/iiab/iiab/issues/3189#issuecomment-1114533415_

jvonau, 2022-06-08:
As an alternative to the hard stop with assert could just set_facts the _install _enabled flags to False for the roles not supported on the given combinations.

holta, 2022-10-15:
Good enough.  (The situation's dramatically improved 4+ months later.)

closed 2022-10-15 by holta