iiab/iiab #3245: IIAB's Node-RED install fails on Raspberry Pi OS: Cannot create dir /home/pi/.node-red/

holta, 2022-06-08:
Thank @nzola for uncovering this issue:

```
1658 2022-06-08 10:50:43,911 p=46640 u=root n=ansible | TASK [nodered : Create /home/pi/.node-red/ directory (rpi)] *********************************************************************************************************************************************************************************
1659 2022-06-08 10:50:44,567 p=46640 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "gid": 0, "group": "root", "mode": "0755", "msg": "chown failed: failed to look up user pi", "owner": "root", "path": "/home/pi/.node-red", "size": 4096, "state": "directory", "uid": 0}
1660 2022-06-08 10:50:44,571 p=46640 u=root n=ansible | PLAY RECAP **********************************************************************************************************************************************************************************************************************************
1661 2022-06-08 10:50:44,572 p=46640 u=root n=ansible | 127.0.0.1                  : ok=77   changed=18   unreachable=0    failed=1    skipped=30   rescued=0    ignored=0  
```

Above is excerpted from http://sprunge.us/0AnFbV?en

jvonau, 2022-06-09:
Next issue will be nodered fails to start on RasPiOS: https://github.com/iiab/iiab/blob/master/roles/nodered/templates/nodered.service.j2#L22-L25

tim-moody, 2022-06-09:
if pi is not a user does /home/pi exist?

holta, 2022-06-09:
Node-RED has changed quite a lot over the past 3 years.

I've made progress trying to get to the bottom of this and will try to post a PR overnight, addressing most of these issues.

holta, 2022-06-11:
Should now be solve by:

- PR #3248

closed 2022-06-11 by holta