iiab/iiab #3247: Best strategy for preparing multiple IIAB servers using imaging

andrewclark5, 2022-06-09:
@tim-moody that should work, but you could do '/library-ext /library bind' and not have the extra level. Think the 'copy function' would just see the unified underlying filesystem and the second target drive would still be mounted in /media/. Once /library is populated on the 'master data drive' mounted via fstab one could just copy the contents to a new target drive, then redeploy on a second machine with just the fstab entries without the need for symlinks. What needs to be on the cloned target is the library.xml from the donor to match what is on the '/library partition' that would be #3052, to save manual intervention so library.xml travels with the data in /library.

_Originally posted by @jvonau in https://github.com/iiab/iiab/issues/3106#issuecomment-1141380523_

@jvonau @tim-moody @shanti-bhardwa

I'd be grateful for final guidance on how best to configure systems for easy deployment bearing the following in mind:-

1. Maintain separate partitions/drives for Library data and OS/Programs 
2. Be able to restore separate images for each and point IIAB to where to find its /library (and set appropriate edits in fstab)
3. Ensure that all necessary Library content is contained within its image.  For example mention is made by @jvonau in the quote above that a libray.xml file must also be copied.  Which one is that?  I have found :-
a. library.xml in /library/zims
b. iiab-library.xml in /opt/iiab/iiab-admin-console/assets
( as well as a deprecated one which I assume I can ignore.)

a. is no problem as it will be copied with /library.
b. would be more problematic...

jvonau, 2022-06-09:
Perhaps you could share your understanding of what is needed and the tools that you are going to use to support your endeavor. My guidance is recorded in #3106, 'b' is not a problem if you make both data and os 'images' at the same time from the source machine.

tim-moody, 2022-06-09:
>b. iiab-library.xml in /opt/iiab/iiab-admin-console/assets

not a problem. this is the git repo and will be present as part of the install. it is also not used after the install.

I think we have come down to two strategies:

1) mount an external drive LIBRARY-EXT on /library
2) mount it on /library-ext and bind mount its internal library directory on /library

@jvonau says 2) is not viable because df will return the wrong or no size. If that is true then 1) should be used.

The only downside is that LIBRARY-EXT will be usable in Admin Console Manage Content as an internal device, but not as an external device mountable on another system.

andrewclark5, 2022-06-09:
> Perhaps you could share your understanding of what is needed and the tools that you are going to use to support your endeavor. My guidance is recorded in #3106, 'b' is not a problem if you make both data and os 'images' at the same time from the source machine.

Ideally we will have 2 separate images for /library and OS/IIAB
But it may be that /library is just held on the big TT server and the file tree dragged over with rsync onto the target server.
It would be great if a script or program could be run from within Admin Console with the Source and Target locations user-specified to keep them away from command line disasters.
We will probably use Foxclone for imaging as it seems to work well.  It can be booted from USB or possibly PXE over the network as we do with our wiping tools.
Ideally there will be a specific image for each different type of server although I have found in practice that I can get away with mis-matched image builds up to a point.  I recently restored the image from a T620 thin client to a Lenovo X240 laptop and it worked first time.  I was without an external USB mouse driver for the first 20 mins then it fixed itself!

holta, 2022-06-09:
Quick Recap from our weekly community call (http://minutes.iiab.io) than began 8h ago.

@tim-moody's post from June 4th appears to be the most immediately useful here: (for this ticket especially)

- https://github.com/iiab/iiab/issues/3106#issuecomment-1146645914

To reconfirm that the various `df` issues are all under control, @shanti-bhardwa and @andrewclark5 will try this out to verify that it works well on Mint 20.3+ on the hardware most relevant for Turing Trust &mdash; verifying that it allows for:

1) Easily mountable (easily interchangeable) external IIAB content disks.
2) With Admin Console's (http://box.lan/admin) > Install Content > [Manage Content](https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/InstContent.rst#manage-content) continuing to work.

@jvonau had suggestions regarding future use of a loopback file (loopback mount?) towards possibly streamlining `/library` content drives/directories/etc in future (much like a partition, but not quite the same).  For example in cases where 100 or 128 GB of `/library` content has already been decided upon, and the inevitable last-minute content changes calm down, allowing this "content collection" [*]  to be circulated among many similar IIAB's?

[*] Not quite Learning Bouquet or "preset", but a very similar idea!

Finally, we did not have time to complete the discussion, as there are several quite subtle + nuanced issues to review regarding what belongs "externally" on /library &mdash; versus what belongs "internally" on IIAB root filesystems.  To be discussed on a future call.

andrewclark5, 2022-06-10:
>    - #3106 (comment)
> "externally" on /library — versus what belongs "internally" on IIAB root

@Tim Moody ***@***.***>
@jvonau <https://github.com/jvonau>

Guys I was testing this on my fresh Mint 20.3 install on a laptop with a
750GB HDD as discussed and encountered problems perhaps due to my
misunderstanding of the sequence required...

I used Gparted to re-partiton the disk as per screenshot attached and
labelled /dev/sda6 to take the library, edited /etc/fstab as per finally
agreed version then

***@***.***:~$ sudo mount -a
[sudo] password for user1:
mount: /library-ext: mount point does not exist.
mount: /library: mount point does not exist.
***@***.***:~$ lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 698.7G  0 disk
├─sda1   8:1    0   513M  0 part /boot/efi
├─sda2   8:2    0   512M  0 part /media/user1/055D-46BB
├─sda3   8:3    0     1K  0 part
├─sda5   8:5    0  15.4G  0 part /
├─sda6   8:6    0 488.9G  0 part
└─sda7   8:7    0 193.4G  0 part
sdf      8:80   1   3.8G  0 disk
└─sdf1   8:81   1   3.8G  0 part
sr0     11:0    1  1024M  0 rom

***@***.***:~$ blkid
/dev/sda1: UUID="7616-1C9B" TYPE="vfat" PARTUUID="bc55cb24-01"
/dev/sda2: UUID="055D-46BB" TYPE="vfat" PARTUUID="bc55cb24-02"
/dev/sda5: UUID="df13e79b-06b1-468f-9660-27157428e78c" TYPE="ext4"
PARTUUID="bc55cb24-05"
/dev/sda6: LABEL="IIAB-LIBRARY-EXT"
UUID="581d987e-7435-4709-a25f-e9d804f170e1" TYPE="ext4"
PARTUUID="bc55cb24-06"
/dev/sda7: LABEL="Backup" UUID="ab3e4b96-cffc-4558-a748-6f72f0210ea0"
TYPE="ext4" PARTUUID="bc55cb24-07"
/dev/sdf1: UUID="BCC8-B24E" TYPE="vfat"
(last line is just a USB with the procedure docs on it)

So where am I going wrong here?

tim-moody, 2022-06-10:
instructions have gotten spread over so many locations that things easily go missing.

>mount: /library-ext: mount point does not exist.
mount: /library: mount point does not exist.

as the error messages indicate you have skipped this step

```
mkdir /library
mkdir /library-ext
```

now mount -a should work

I take it that using an internal partition is just an experiment and not the eventually intended configuration.

tim-moody, 2022-06-10:
/dev/sda6 also needs a library directory on it before mount -a will work

andrewclark5, 2022-06-10:
OK.  I wondered about that.  But in order to create the directories on that
partition I first have to mount it, right?

That was what I was attempting to do, with the mount - a command.

>I take it that using an internal partition is just an experiment and not
the eventually intended configuration.

Not exactly.  This is what I was saying yesterday - if that WOULD work for
big disks then we would have a standardization of procedure across all
server builds, big or small..

tim-moody, 2022-06-10:
if the data (library) partition is internal I would think it would not ever be mounted externally on another system

if that is the case then the bind mount is not really needed and you can mount /dev/sda6 directly on /library in fstab and comment out the other bind mount line

tim-moody, 2022-06-10:
I should add that the bind mount will still work even for an internal partition if standardization is the top priority.

andrewclark5, 2022-06-10:
Looks as if we are partly there, after a reboot.  (So sudo mount -a is not
the same as rebooting, I guess?)

/dev/sda6 is now mounted, but looks as if I need to change it to /library
and not /library-ext, which is what is appearing in the File System?

So I will now try
LABEL=IIAB-LIBRARY-EXT /library ext4 defaults,nofail 0 0
# was /library-ext/library /library none defaults,bind,nofail 0 0

tim-moody, 2022-06-10:
>/dev/sda6 is now mounted, but looks as if I need to change it to /library
and not /library-ext, which is what is appearing in the File System?

So now it looks like you have unexpectedly switched strategies to mounting the internal partition directly on /library.

Let's get the main strategy working first.

with /dev/sda6 mounted on /library-ext
mkdir /library-ext/library

with fstab as

LABEL=IIAB-LIBRARY-EXT /library-ext ext4 defaults,nofail 0 0
/library-ext/library /library none defaults,bind,nofail 0 0

mount -a

andrewclark5, 2022-06-10:
OK, did all that.

***@***.***:~$ sudo nano /etc/fstab
[sudo] password for user1:
***@***.***:~$ sudo mount -a
***@***.***:~$ blkid
/dev/sda1: UUID="7616-1C9B" TYPE="vfat" PARTUUID="bc55cb24-01"
/dev/sda2: UUID="055D-46BB" TYPE="vfat" PARTUUID="bc55cb24-02"
/dev/sda5: UUID="df13e79b-06b1-468f-9660-27157428e78c" TYPE="ext4"
PARTUUID="bc55cb24-05"
/dev/sda6: LABEL="IIAB-LIBRARY-EXT"
UUID="581d987e-7435-4709-a25f-e9d804f170e1" TYPE="ext4"
PARTUUID="bc55cb24-06"
/dev/sda7: LABEL="Backup" UUID="ab3e4b96-cffc-4558-a748-6f72f0210ea0"
TYPE="ext4" PARTUUID="bc55cb24-07"
***@***.***:~$

***@***.***:~$ lsblk
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 698.7G  0 disk
├─sda1   8:1    0   513M  0 part /boot/efi
├─sda2   8:2    0   512M  0 part
├─sda3   8:3    0     1K  0 part
├─sda5   8:5    0  15.4G  0 part /
├─sda6   8:6    0 488.9G  0 part /library
└─sda7   8:7    0 193.4G  0 part
sr0     11:0    1  1024M  0 rom
***@***.***:~$

Is that what you want?

andrewclark5, 2022-06-10:
...but I have now ended up with a directory called library-ext containing
sub-directories for library and library-ext!

jvonau, 2022-06-10:
Having to create /library-ext/library forces editing /etc/fsab and using mount -a twice adding to two /etc/fstab entries individually.  
First edit to add the mounting of /library-ext line.
`sudo mount -a` 
`sudo mkdir /library-ext/library`
Second edit to fstab adding the bindmount line.
`sudo mount -a`

I've baked this functionality into #3051

andrewclark5, 2022-06-10:
Hi Jerry

That seems like a good explanation!  I am talking to Tim on email just now as he is out so will cc your text to him by email.

andrewclark5, 2022-06-10:
Should I delete all the superfluous folders and start again with fstab?

And what are we trying to achieve with the library folder structure?

jvonau, 2022-06-10:
> Should I delete all the superfluous folders and start again with fstab?

On the OS filesystem there should be two folders /library /library-ext
On the content filesystem there should be /library, that would be /library-ext/library when LABEL= is mounted at /library-ext
> And what are we trying to achieve with the library folder structure?

Think Tim's goal is to avoid rewriting current code in admin-console to be able to extend the existing import/export functionality.

andrewclark5, 2022-06-10:
@tim-moody @jvonau @holta 
Ah.  I see.  OK, I'm just going to edit fstab, delete the other unwanted folders and reboot and see what I have then.

shanti-bhardwa, 2022-06-11:
Done some testing on HP t620 with 16GB flash drive. Results look promising. Here is a summary. 
1. Built fresh Mint 20.3. from .iso file and installed it on t620. I am using Raspberry PI 2B  USB WI-FI dongle as my WI-FI interface.
2. Updated the system using sudo apt update & sudo apt upgrade to bring everything to current levels. Added Gparted as an addition utility.
3. Created two folders on root with  sudo mkdir /library and  sudo mkdir /library-ext 
4. Formatted my external drive. This is a 32GB SSD in a SATA/USB caddy with label IIAB-LIBRARY-EXT with ext4 file format. I used these two commands
    sudo mkfs.ext4 /dev/sdb1 and sudo e2label /dev/sdb1 IIAB-LIBRARY-EXT. can also be done using Disks utility or from File Manager, but have not tested this.
5. Installed the drive and then created a folder with sudo mkdir /library-ext/library on this USB device.
6. Then added my /etc/fstab entries as follows. These two entries tell me that USB will always be mounted at /library-ext by mount at boot time &  bind makes the two directory  structures the same. IIAB puts everthing under /library so at IIAB build time only the /library-ext/library on USB is used as the real srorage,  
     LABEL=IIAB-LIBRART-EXT /library-ext ext4 defaults,nofail 0 0 
     /library-ext/library /library none defaults,bind,nofail, 0 0 
7. 
8. Install IIAB 8.0. Install went without a hitch. The sudo iiab-diagnostics is here : http:/sprunge.us/InAh1R
9. Downloaded few modules and worked fine.
10 . Thinks I have noticed. df -h shows the usb drive mounted on /media/usb0 & in on admin console cannot get system storage.
11. The machine is on vpn at 10.8.0.82. You can ssh user is sb01 and password is shanti

shanti-bhardwa, 2022-06-11:
I think this going work really well. You can see the amount of storage on the uSB drives from Manage Content  option.
My iiab-admin password is also shanti, so try adding some content if you want.
The external drive is hsown as mounted on /media/usb0. The second drive gets mounted on /media/usb1

shanti-bhardwa, 2022-06-11:
Trying to get System Storage on Admin Console  gives me this error
GET-STORAGE-INFO FAILED and reported Unexpected error in Command GET-STORAGE-INFO

shanti-bhardwa, 2022-06-11:
Apologies, the machine is on vpn **10.8.0.62.** My user id is sb01 with password shanti. Also iiab-admin password is shanti.
I will leave this machine connected for rest of the day.

shanti-bhardwa, 2022-06-11:
Next steps for me to test is to create more External drives. So I will clone the IIAB-LIBRARY-EXT.
One drive  will have Wikipedia , second Maps, third TED Video, fourth  Kolibri Channels. See how it works

shanti-bhardwa, 2022-06-11:
If you use gparted to look at the drives mounted then IIAB-LIBRARY-EXT which is /dev/sdb1 is mounted in three places /library, /library-ext and /media/usb0. You also get the used and unused storage size. So there must be other ways to get storage information without using df command.

tim-moody, 2022-06-11:
>df -h shows the usb drive mounted on /media/usb0

even after reboot?

>GET-STORAGE-INFO FAILED

yes it's a bug - not compatible with df change for linux mint.

andrewclark5, 2022-06-11:
Shanti
For reference

Shanti version - on USB
LABEL=IIAB-LIBRARY-EXT /library-ext ext4 defaults,nofail 0 0
/library-ext/library /library none defaults,bind,nofail, 0 0

Current HP g6 version - on HDD
# Added for IIAB Library - here using HDD partition /dev/sda6
# Not currently working - puts /library on sda5 File System
LABEL=IIAB-LIBRARY-EXT /library-ext ext4 defaults,nofail 0 0
/library-ext/library /library none defaults,bind,nofail 0 0

As far as I can see they are the same syntax but being treated differently
due to me using an internal partition in an effort to standardise.  But

***@***.***:~$ df
df: /run/user/1000/doc: Operation not permitted
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1696672        0   1696672   0% /dev
tmpfs             348688     3120    345568   1% /run
/dev/sda5       14793648 11812176   2160328  85% /
tmpfs            1743432        0   1743432   0% /dev/shm
tmpfs               5120        4      5116   1% /run/lock
tmpfs            1743432        0   1743432   0% /sys/fs/cgroup
/dev/sda6      503578816  1488376 476440420   1% /library-ext
/dev/sda1         524272        4    524268   1% /boot/efi
tmpfs             348684       24    348660   1% /run/user/1000

Andrew

andrewclark5, 2022-06-11:
I don't know if this throws any further light on what is happening on my HP
g6

***@***.***:/$ cd library
***@***.***:/library$ ls
awstats  calibre-web  downloads  ka-lite  kolibri  public  Test.txt
 wordpress  working  www  zims
***@***.***:/library$ cd /
***@***.***:/$ ls -l /lib*
lrwxrwxrwx  1 root root    7 Jun  9 13:29 /lib -> usr/lib
lrwxrwxrwx  1 root root    9 Jun  9 13:29 /lib32 -> usr/lib32
lrwxrwxrwx  1 root root    9 Jun  9 13:29 /lib64 -> usr/lib64
lrwxrwxrwx  1 root root   10 Jun  9 13:29 /libx32 -> usr/libx32

/library:
total 40
drwxr-x--- 2 www-data www-data 4096 Jun 11 13:40 awstats
drwxr-xr-x 3 root     www-data 4096 Jun 10 19:15 calibre-web
drwxr-xr-x 5 root     root     4096 Jun 10 18:56 downloads
drwxr-xr-x 9 root     root     4096 Jun 11 11:23 ka-lite
drwxr-xr-x 6 kolibri  www-data 4096 Jun 11 13:47 kolibri
drwxrwxrwx 2 smbuser  smbuser  4096 Jun 10 19:03 public
-rw-r--r-- 1 root     root        0 Jun 10 17:40 Test.txt
drwxrwxr-x 5 root     www-data 4096 Jun 10 19:06 wordpress
drwxr-xr-x 6 root     root     4096 Jun 10 19:18 working
drwxr-xr-x 4 root     root     4096 Jun 10 19:11 www
drwxr-xr-x 4 root     root     4096 Jun 10 19:22 zims

/library-ext:
total 24
drwxr-xr-x 12 root root  4096 Jun 10 19:14 library
drwx------  2 root root 16384 Jun 10 12:03 lost+found
-rw-r--r--  1 root root    72 Jun 10 14:31 TestDoc.txt
***@***.***:/$

I can start the VPN support if it helps.

Andrew

jvonau, 2022-06-11:
> If you use gparted to look at the drives mounted then IIAB-LIBRARY-EXT which is /dev/sdb1 is mounted in three places /library, /library-ext and /media/usb0. You also get the used and unused storage size. So there must be other ways to get storage information without using df command.

From: 10.8.0.62 snipped for size
df -a
/dev/sdb1       29702580  9673856  18496844  35% /library-ext
/dev/sdb1       29702580  9673856  18496844  35% /library
/dev/sdb1       29702580  9673856  18496844  35% /media/usb0

df
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3733716        0   3733716   0% /dev
tmpfs             756096     1640    754456   1% /run
/dev/sda2       14810456  9514724   4523688  68% /
tmpfs            3780476        0   3780476   0% /dev/shm
tmpfs               5120        4      5116   1% /run/lock
tmpfs            3780476        0   3780476   0% /sys/fs/cgroup
/dev/sdb1       29702580  9673856  18496844  35% /media/usb0
/dev/sda1         523248     5356    517892   2% /boot/efi
tmpfs             756092       28    756064   1% /run/user/1000
/dev/sdc1       30642992 16396256  12667072  57% /media/usb1
tmpfs             756092        4    756088   1% /run/user/1001

tim-moody, 2022-06-11:
I opened https://github.com/iiab/iiab-admin-console/issues/502

tim-moody, 2022-06-11:
>/dev/sdb1 29702580 9673856 18496844 35% /media/usb0

this to me is still an outlier and probably there was no reboot

tim-moody, 2022-06-11:
if the bind mount strategy is pursued, I would suggest putting the first device mount on /mnt/library-ext to make things less confusing.

jvonau, 2022-06-11:
> > /dev/sdb1 29702580 9673856 18496844 35% /media/usb0
> 
> this to me is still an outlier and probably there was no reboot

Doesn't matter if a reboot occurred, the usb device is going to get mounted on reboot anyway.

tim-moody, 2022-06-11:
>Doesn't matter if a reboot occurred, the usb device is going to get mounted on reboot anyway.

didn't for @andrewclark5 or for me

jvonau, 2022-06-11:
Internal partitions won't have this issue as the disk is not on the usb bus, however usb booted RPi will, see #2552

holta, 2022-06-11:
_Thanks everyone for a fortnight of very hard work here!_

Tidying up many things as a result, above and beyond...

- #3106 
- #3231
- PR #3251

Allowing us to clean up + clarify...

http://FAQ.IIAB.IO > `41 Can I permanently attach an external USB drive, to add more content?`

...in the coming week hopefully :heavy_check_mark:

jvonau, 2022-06-11:
> if the bind mount strategy is pursued, I would suggest putting the first device mount on /mnt/library-ext to make things less confusing.

I'll suggest using /media/library-ext, as we already littered the /media directory with usb* entries. **Should** (needs confirmation) fake out df as usbmount's path (/media/usb0) would be shorter than /media/library-ext.

jvonau, 2022-06-11:
Nope, still end up with /media/usb0 as showing up in df on Mint(10.8.0.62), displays the last mounting action first ahead of the other 2 entries that are viewable with df -a.
```
df -a | grep sd
/dev/sdb2       14810456  9540176   4498236  68% /
/dev/sdc1       29702580  9673712  18496988  35% /media/library-ext
/dev/sdc1       29702580  9673712  18496988  35% /library
/dev/sdb1         523248     5356    517892   2% /boot/efi
/dev/sdc1       29702580  9673712  18496988  35% /media/usb0
gvfsd-fuse             -        -         -    - /run/user/116/gvfs
/dev/sda1       30642992 16396256  12667072  57% /media/usb1
```
```
 df
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             3733728        0   3733728   0% /dev
tmpfs             756100     1500    754600   1% /run
/dev/sdb2       14810456  9540184   4498228  68% /
tmpfs            3780488        0   3780488   0% /dev/shm
tmpfs               5120        4      5116   1% /run/lock
tmpfs            3780488        0   3780488   0% /sys/fs/cgroup
/dev/sdc1       29702580  9673712  18496988  35% /media/usb0
/dev/sdb1         523248     5356    517892   2% /boot/efi
tmpfs             756096        8    756088   1% /run/user/116
/dev/sda1       30642992 16396256  12667072  57% /media/usb1
tmpfs             756096        4    756092   1% /run/user/1001

```
```
 findmnt -s
TARGET             SOURCE                                    FSTYPE OPTIONS
/                  UUID=d94f9d2c-42be-4128-a5a1-bdb8416fa883 ext4   errors=remount-ro
/boot/efi          UUID=F704-7064                            vfat   umask=0077
none               /swapfile                                 swap   sw
/media/library-ext LABEL=IIAB-LIBRARY-EXT                    ext4   defaults,nofail
/library           /media/library-ext/library                none   defaults,bind,nofail
```

shanti-bhardwa, 2022-06-12:
I tested the following:
1. Clone my 32 GB IIAB-LIBRARY-EXT to 240 GB SSD  GB & 320 HDD GB drives in a  SATA caddy.
2. Both Mount correctly. I then expanded the 32 GB partitions using gparted to fill out the full capacity of the drives.
3. I can then add extra content, from another USB mounted drive on /media/usb1 or from the normal repositories using admin console
4. Can also add Kolibri channels to cloned drives.

So we have the ability to create & share content on drives between IIAB Servers if they configured to work with external media.

My next test will be to clone this HP t620 build to other hardware. I have HP t520 & DELL Wyse thin clients  and Lenovo Notebook with 32GB flash drive. See how they all behave.

jvonau, 2022-06-12:
>> if the bind mount strategy is pursued, I would suggest putting the first device mount on /mnt/library-ext to make things less confusing.

>I'll suggest using /media/library-ext, as we already littered the /media directory with usb* entries. **Should** (needs confirmation) fake out df as usbmount's path (/media/usb0) would be shorter than /media/library-ext.

Given the 'fake out' didn't work think I prefer '/library-ext' as the mount point, saves me altering #3051 any further.

tim-moody, 2022-06-13:
>I'll suggest using /media/library-ext

I think /media should be reserved for mounting non-permanent devices, so I would not put a library device there even if it is a USB device. I still think /mnt is less confusing, but I don't feel as strongly about that as about not putting it on /media.

holta, 2022-06-13:
The official story...

- [Filesystem Hierarchy Standard 2.3 (2004)](https://www.pathname.com/fhs/pub/fhs-2.3.html#MEDIAMOUNTPOINT)
- [Filesystem Hierarchy Standard 3.0 (2015)](https://refspecs.linuxfoundation.org/FHS_3.0/fhs-3.0.html#mediaMountPoint)

...seems to be:

> **/media** is mount point for **removable media**. In other words, where system mounts removable media. This directory contains sub-directories used for mounting removable media such as CD-ROMs, floppy disks, etc.
>
> **/mnt** is for **[temporarily mounted filesystems]**. In other words, where user can mount things. This directory is generally used for mounting filessytems temporarily when needed.

https://unix.stackexchange.com/questions/252822/whats-the-difference-between-mnt-vs-media/252823#252823
https://en.wikipedia.org/wiki/Filesystem_Hierarchy_Standard

shanti-bhardwa, 2022-06-13:
Done some more testing, this time cloning the system (OS+IIAB 8.0) 16GB flash to run on HP t520 and DELL WYSE 5070 thin clients. I used Foxclone to clone the system  drive from my master build on HP t620.

 Also created various content on  IIAB-LIBRARY-EXT labelled 120 GB, 240 GB & 320 GB Hard drives. Then ported them to  HP t520  and DELL WYSE 5070 machines to run them as servers. They work fine. 

I will create the .img file of the 16GB OS+IIAB 8.0 build , upload it on Dropbox, so people can download it & try on  other INTEL hardware.

andrewclark5, 2022-06-14:
All

I picked up 2 t620 thin clients from  TT this afternoon and a couple of
Dell Latitude laptops to work on building and documentation of new images.

It would be helpful if after all our discussions someone could document the
final recommendations on the Library.  Any volunteers?

Interestinglying I had a request passed on today from our Malawi Project
Manager who wants to know if we could dual- boot our machines as Windows
Clients or IIAB Servers!  I will have to think about the challenges that
might present?

A.

holta, 2022-06-14:
> document the final recommendations on [ /library ]. Any volunteers?

Hopefully a couple of us can work on this over the coming week.

Summarizing priority/consensus recommendations, to places like:
http://FAQ.IIAB.IO -> `41 Can I permanently attach an external USB drive, to add more content?`

I can help.

andrewclark5, 2022-06-15:
I am happy to contribute but some of the finer subtleties of Linux may be
lost on me!

holta, 2022-06-15:
> I am happy to contribute

Thanks @andrewclark5!

(Standardizing our global suggestions will help a lot of people if we're able to simplify, simplify, simplify ;)

tim-moody, 2022-06-16:
>Summarizing priority/consensus recommendations, to places like:
http://faq.iiab.io/ -> 41 Can I permanently attach an external USB drive, to add more content?

This conflates two needs: 1) creating the entirety of /library on an external device 2) putting some content on an external device

andrewclark5, 2022-06-19:
Hi Guys
So I have started building 2 IIAB server systems for final testing of the install process before I make images.
1) An HP T620 Thin Client
2) A Dell latitude 7480 laptop

For 1) Having installed and updated Mint to latest version and made the changes to fstab to accommodate the library on a 32GB USB drive I made a base image and hen installed IIAB 8.0.

On reboot after successful install I immediately got a crash in Cinnamon, then "You are running in Fallback Mode".

I and sudo iiab-diagnostics and uploaded the file to a pastebin at 
PUBLISHING TO URL... http://sprunge.us/pUTlYT 

I was hoping to finish this today, so this is a bit surprising.
Most bits look normal...
```
oem@box:~$ lsblk
NAME   MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda      8:0    0 14.9G  0 disk 
├─sda1   8:1    0  512M  0 part /boot/efi
└─sda2   8:2    0 14.4G  0 part /
sdb      8:16   1 28.7G  0 disk /media/usb0
oem@box:~$ blkid
/dev/sda1: UUID="0C86-4375" TYPE="vfat" PARTLABEL="EFI System Partition" PARTUUID="ee64a491-a6e4-48e4-9784-590d968f476b"
/dev/sda2: UUID="3733af81-b887-4d62-90f6-561c46bca0a7" TYPE="ext4" PARTUUID="50402ed4-c00f-4d76-92da-67146a8101c6"
/dev/sdb: LABEL="IIAB-LIBRARY-EXT" UUID="c86e5c4e-5a94-4740-93f9-84d4cb1d179c" TYPE="ext4"
oem@box:~$ 
@jvonau @tim-moody @holta @shanti-bhardwa 
Any ideas?
```

jvonau, 2022-06-19:
From pUTlYT
> 00:01.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] Kabini [Radeon HD 8280E] [1002:9837]

Is the 'Dell latitude 7480 laptop' running a stock mint install or a image you prepared based on the 'HP T620 Thin Client' machine? Seems like you ran into the same issues on the same hardware https://github.com/iiab/iiab/issues/3106#issuecomment-1140277039

> /dev/sdb: LABEL="IIAB-LIBRARY-EXT" UUID="c86e5c4e-5a94-4740-93f9-84d4cb1d179c" TYPE="ext4"

https://github.com/iiab/iiab/issues/3106#issuecomment-1140299072 Didn't you want a partitioned device?

andrewclark5, 2022-06-19:
@jvonau 
The above relates only to the T620 on which I have made a clean build of MInt 20.3 using the 16GB M2 storage.  It was all working fine,
Adding the External library on a 32GB USB and making the changes as agreed to fstab then installing IIAB has caused it to crash Cinnamon!
In parallel I am building a dual-boot system on the 7840.

andrewclark5, 2022-06-19:
@jvonau 
I will try and think of a way to amend the M2 config without starting again...:(

jvonau, 2022-06-19:
> @jvonau The above relates only to the T620 on which I have made a clean build of MInt 20.3 using the 16GB M2 storage. It was all working fine, but I note that perhaps I should have created a partition on the M2 instead of just letting Mint do its thing. 

Pictures or a better description of 'just letting Mint do its thing' would go a long way here. After the Mint install and editing fstab for /library did you reboot to ensure the device was auto-mounting correctly with the desktop behaving correctly (did you disable the desktop automount feature?) prior to installing IIAB?
> Adding the External library on a 32GB USB and making the changes as agreed to fstab then installing IIAB has caused it to crash Cinnamon! In parallel I am building a dual-boot system on the 7840.

Petty sure the agreed instructions were to use a partitioned device for the 32GB USB drive. Think the desktop doesn't like the way /dev/sdb is setup, I recall you had good luck when the usb device was partitioned ie /dev/sdb1 and the desktop auto-mount was disabled.
> I will try and think of a way to amend the M2 config without starting again

The M2 is the 'OS disk' that looks to be OK
>sda      8:0    0 14.9G  0 disk 
├─sda1   8:1    0  512M  0 part /boot/efi
└─sda2   8:2    0 14.4G  0 part /

andrewclark5, 2022-06-19:
@jvonau 
You were absolutely right (as usual!).  I thought initially that you were referring to the main File System disk but it was the USB that was causing the problem.  
To save time I was reusing a 32GB USB that already had a library on it it, forgetting that it needed to have a partition on it with the label IIAB-LIBRARY-EXT, not the whole USB.
That gave rise to a further problem when I went to fix it in that if I plugged it into a machine that already had that label, it would not Mount (understandably).  When I finally figured that out and found a laptop with just the original structure I was able to copy the library off, reformat the USB and partition it with GParted.  I then copied the library back on and all is working on the T620 now.
It is something to keep in mind though, and at least by embarrassing myself with this mistake I might remember next time :)

jvonau, 2022-06-19:
> To save time I was reusing a 32GB USB that already had a library on it it, forgetting that it needed to have a partition on it with the label IIAB-LIBRARY-EXT, not the whole USB.

First you get good, then you get fast. I'll bet you'll remember next time. ;)

> That gave rise to a further problem when I went to fix it in that if I plugged it into a machine that already had that label, it would not Mount (understandably).

@tim-moody The above needs a bit more investigation, usbmount should of mounted any usb disk within /media anyway just not create a link in local_content or the whole dual purpose layout with /library within the top level of the usb disk doesn't work as intended to support the import function of admin-console.

andrewclark5, 2022-06-19:
@jvonau 
Let me know if you want further data on that - or I can give you VPN access.

andrewclark5, 2022-06-19:
@jvonau 
I also have another issue following my image Restore process to another T620.
I backed up the (2) system partitions from T620-1 to external media and then restored them to T620-2.
I left out the Library USB, but physically moved it to the T620-2.
The machine booted up but again crashed Cinnamon as it displayed the Desktop,
I wondered if this might be a UUID issue?
```
oem@box:~$ blkid
/dev/sda1: UUID="0C86-4375" TYPE="vfat" PARTLABEL="EFI System Partition" PARTUUID="ee64a491-a6e4-48e4-9784-590d968f476b"
/dev/sda2: UUID="3733af81-b887-4d62-90f6-561c46bca0a7" TYPE="ext4" PARTUUID="50402ed4-c00f-4d76-92da-67146a8101c6"
/dev/sdb5: LABEL="IIAB-LIBRARY-EXT" UUID="bc15607b-ece5-413a-8a4e-dc212a320bf4" TYPE="ext4" PARTUUID="05b0ec8f-05"
oem@box:~$ 
*******************
fstab reads
```
# /etc/fstab: static file system information.
#
# Use 'blkid' to print the universally unique identifier for a
# device; this may be used with UUID= as a more robust way to name devices
# that works even if disks are added and removed. See fstab(5).
#
# <file system> <mount point>   <type>  <options>       <dump>  <pass>
# / was on /dev/sda2 during installation
UUID=3733af81-b887-4d62-90f6-561c46bca0a7 /               ext4    errors=remount-ro 0       1
# /boot/efi was on /dev/sda1 during installation
UUID=0C86-4375  /boot/efi       vfat    umask=0077      0       1
/swapfile                                 none            swap    sw              0       0
#Additions to put IIAB library on separate location off system
LABEL=IIAB-LIBRARY-EXT /library-ext ext4 defaults,nofail 0 0
/library-ext/library /library none defaults,bind,nofail 0 0

but the main identifiers seem to be the same and fstab does not mention the partition UUIDs
```

```

andrewclark5, 2022-06-19:
I thought that you might like to see the scale of the operation... Hence the desire to get a reliable cloning process.
[image]

jvonau, 2022-06-19:
> @jvonau I also have another issue following my image Restore process to another T620. I backed up the (2) system partitions from T620-1 to external media and then restored them to T620-2. 

I suggest that you test your cloning process without IIAB present, before doing the /etc/fstab mods, with every other possible application closed except for your cloning process that you are able to clone the stock system with the technique you are using. A complete step by step of what you are doing might be helpful as well.  

> I left out the Library USB, but physically moved it to the T620-2. The machine booted up but again crashed Cinnamon as it displayed the Desktop, I wondered if this might be a UUID issue?
>

Maybe, depends on what the cloning tool that you are using does when writing the data, but it smells like '/' is being mounted as you get a desktop.
 
> ```
> oem@box:~$ blkid
> /dev/sda1: UUID="0C86-4375" TYPE="vfat" PARTLABEL="EFI System Partition" PARTUUID="ee64a491-a6e4-48e4-9784-590d968f476b"
> /dev/sda2: UUID="3733af81-b887-4d62-90f6-561c46bca0a7" TYPE="ext4" PARTUUID="50402ed4-c00f-4d76-92da-67146a8101c6"
> /dev/sdb5: LABEL="IIAB-LIBRARY-EXT" UUID="bc15607b-ece5-413a-8a4e-dc212a320bf4" TYPE="ext4" PARTUUID="05b0ec8f-05"
> oem@box:~$ 
> *******************
> fstab reads
> ```
> 
> > \# /etc/fstab: static file system information.
> > 
> > \# Use 'blkid' to print the universally unique identifier for a
> > \# device; this may be used with UUID= as a more robust way to name devices
> > \# that works even if disks are added and removed. See fstab(5).
> > 
> > \#                
> > \# / was on /dev/sda2 during installation
> > UUID=3733af81-b887-4d62-90f6-561c46bca0a7 /               ext4    errors=remount-ro 0       1
> > \# /boot/efi was on /dev/sda1 during installation
> > UUID=0C86-4375  /boot/efi       vfat    umask=0077      0       1
> > /swapfile                                 none            swap    sw              0       0
> > \#Additions to put IIAB library on separate location off system
> > LABEL=IIAB-LIBRARY-EXT /library-ext ext4 defaults,nofail 0 0
> > /library-ext/library /library none defaults,bind,nofail 0 0

> but the main identifiers seem to be the same and fstab does not mention the partition UUIDs

Yes looks like it should work, might be chasing a non-iiab related issue in Cinnamon, that is why I'm asking for the test case without IIAB.

(Sorry about the formatting - not sure how to fix that!)
use a \ in front of the #

jvonau, 2022-06-19:
A pic of the 'crash' might be helpful to help me understand with you are looking at.

[61 more replies]