iiab/iiab #3249: Can't auto test PRs with the curl install routine

jvonau, 2022-06-10:
curl d.iiab.io/fast.txt | sudo bash -s 3051

```
PR #3051: cd /opt/iiab/iiab
git pull --no-edit https://github.com/jvonau/iiab library-img
remote: Enumerating objects: 155, done.
remote: Counting objects: 100% (95/95), done.
remote: Compressing objects: 100% (35/35), done.
remote: Total 155 (delta 59), reused 94 (delta 59), pack-reused 60
Receiving objects: 100% (155/155), 20.10 KiB | 1.34 MiB/s, done.
Resolving deltas: 100% (86/86), completed with 6 local objects.
From https://github.com/jvonau/iiab
 * branch              library-img -> FETCH_HEAD
hint: You have divergent branches and need to specify how to reconcile them.
hint: You can do so by running one of the following commands sometime before
hint: your next pull:
hint: 
hint:   git config pull.rebase false  # merge (the default strategy)
hint:   git config pull.rebase true   # rebase
hint:   git config pull.ff only       # fast-forward only
hint: 
hint: You can replace "git config" with "git config --global" to set a default
hint: preference for all repositories. You can also pass --rebase, --no-rebase,
hint: or --ff-only on the command line to override the configured default per
hint: invocation.
fatal: Need to specify how to reconcile divergent branches.

git exit/error code: 128

Delete it from top of /etc/iiab/pr-queue and continue? [Y/n] 
```

jvonau, 2022-06-10:
Wouldn't be a bad idea to use a branch too.

holta, 2022-06-10:
Excellent question.  I'll work on this.

> You can also pass --rebase, --no-rebase, hint: or --ff-only on the command line

We need to help IIAB community testers navigate these 3 choices quickly and without confusion &mdash; when they don't understand (and don't want to understand!) these 3:

- merge
- rebase
- fast-forward

Badly exacerbated by [git 2.27 (2020-06-01) which meant well but made longstanding git corner cases even more confusing than necessary](https://stackoverflow.com/questions/62653114/).

This mess has been gradually cleaned up over the past 2 years such that [git 2.36 (2022-04-18) is now dramatically improved](https://stackoverflow.com/a/71774640).

(Of course all these ugly versions of git will be with us for ~2 years until Debian/Ubuntu/Mint/RasPiOS sweep out the cobwebs, so we have to deal with all this sloppy messaging regardless!)

> Wouldn't be a bad idea to use a branch too.

Indeed.

There's a serious challenge however: what to name the branch (especially when multiple PR's are pulled) _while trying very hard to avoid (if possible)_ an interactive UI/delay/prompt suggesting and/or asking for the branch name.

Thanks for raising both these issues.  As Simplifying Is Really Important (i.e. helping well-meaning community testers survive this messiness...of so many different versions of git over the past 2 years ;-)

holta, 2022-06-11:
ASIDE: Interesting that Git is cleaning up its byzantine + kludgy syntax finally after 17 years :sweat_smile: 

> Git developers insist on retaining this Awkward User eXperience (AUX) in the face of widespread community dissent is... enlightening. And discouraging. (Mostly discouraging.)
https://stackoverflow.com/questions/6089294/why-do-i-need-to-do-set-upstream-all-the-time#comment94399695_22933955

> With Git 2.37 (Q3 2022), a `git config --global push.autoSetupRemote true` will take care of that for you.
https://stackoverflow.com/questions/6089294/why-do-i-need-to-do-set-upstream-all-the-time/72401899#72401899

holta, 2022-06-18:
(1) This immediate issue is confirmed solved by PR iiab/iiab-factory#219

(2) Leaving this ticket open is also Ok &mdash; as we can and should make PR testing much friendlier for sincere newcomers &mdash; whenever/wherever there are opportunities :heavy_check_mark:

holta, 2022-07-01:
This issue's resolved.

Let's open a new ticket to make PR testing much friendlier for sincere newcomers (as mentioned above).

Brief discussion of this same topic...is also here:
https://github.com/iiab/iiab/issues/3266#issuecomment-1171231231

closed 2022-07-01 by holta