iiab/iiab #3252: Don't create local_content symlink for device that is permanent /library storage

tim-moody, 2022-06-13:
Even if iiab_usb_lib_show_all is True and the /library device is a USB it should not get a symlink on local_content as this is a significant security risk.

tim-moody, 2022-06-13:
Agree #2552 is related.

>The issue of http://box/usb displaying the contents of the entire Linux filesystem is related: (arguably this is a feature rather than a bug, for certain kids trying to learn Linux if they get a vote anyway, but of course not all teachers agree ;)

I certainly don't agree. Even just mounting all of /library on local content can potentially make passwords readable by anyone for wordpress or mediawiki and probably nextcloud, etc.

jvonau, 2022-06-14:
Toggle iiab_usb_lib_show_all to False, that is what the variable is intended for, selective creation of the local_content symlink.

holta, 2022-06-14:
> Toggle iiab_usb_lib_show_all to False, that is what the variable is intended for, selective creation of the local_content symlink.

That's an interesting point.  Internal mechanics are here:

https://github.com/iiab/iiab/search?q=IIAB_USB_LIB_SHOW_ALL

tim-moody, 2022-06-14:
>Toggle iiab_usb_lib_show_all to False, that is what the variable is intended for, selective creation of the local_content symlink.

This creates the risk that I accidentally expose all of /library because I have a usb with local content that I wish to display in its entirety.

Which is why I began with

>Even if iiab_usb_lib_show_all is True and the /library device is a USB it should not get a symlink on local_content as this is a significant security risk.

But because we already have logic to create the symlink selectively it should be possible to support the the case where I have /library on a USB and I have another USB with local content.

jvonau, 2022-06-14:
@tim-moody You have commit access, fix it the way you want. While the hood is up fix up the usb booted example also.

jvonau, 2022-06-17:
Is this now resolved?

tim-moody, 2022-06-17:
I believe so.

closed 2022-06-17 by tim-moody