iiab/iiab #3264: Networking refactoring & similar efforts

holta, 2022-06-24:
This is just a placemarker (pinned to the top of https://github.com/iiab/iiab/issues for now) to help draw more attention to PR's like:

- [ ] PR #3145 
- [ ] PR #3173

holta, 2022-06-24:
@deldesir: everyone recognizes that you're more than busy with other things in Haiti :exploding_head: so ignore this as appropriate.

But *if* by some miracle you have a moment in July to review progress here with me and @jvonau and others (and/or offer suggestions) that'd be really great! :partying_face:

TOTAL ASIDE: Debian 12 Bookworm's initial freeze will be 2023-01-12 so we might want to start testing IIAB installs on their [daily build](https://cdimage.debian.org/cdimage/daily-builds/daily/arch-latest/amd64/) later this year.

jvonau, 2022-06-30:
You intentionally dump stuff into the network role to break outstanding PRs and I'm the one who doesn't play nice in the sand box. wow just wow.

jvonau, 2022-07-08:
related #3281

jvonau, 2022-07-10:
You broke #3296 on purpose with #3299, couldn't just leave named/dhcpd alone knowing they're not being used and I discussed removing the support during the call.