iiab/iiab #3266: Adopt better tagging procedures

jvonau, 2022-06-25:
Could you find it in your heart to adopt the simplest changelog like practice? Before tagging could the [iiab_revision](https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L17) value be incremented each time a tagged release is going to be created? This methodology could become a good general yardstick as to the age of reported install going forward, tying the git hash to a easy to understand integer and would augment understanding changes like https://github.com/iiab/iiab/compare/8.0-preview-2...master

jvonau, 2022-06-25:
refs: #2771 https://github.com/iiab/iiab/pull/2500#issuecomment-689114958

tim-moody, 2022-06-25:
Our whole 'versioning' system needs rethinking. If I look at iiab_base_ver and iiab_revision I would think we are at release 8.0.0, but 8 has never been released. The release/tag says 8.0 Preview 2. Such releases are completely arbitrary, made up by @holta with no community input. We have never really used an n.m.p release naming convention.

Then there is the double entry:
iiab_base_ver and iiab_revision are maintained both in default vars and as a comment in [roles/0-init/defaults/main.yml](https://github.com/iiab/iiab/blob/79de65425c5dcf09ecf5435a23b5b52fa035941d/roles/0-init/defaults/main.yml) Release/tag is maintained independently.

The simplest and most consistent thing would be to automate the writing of the value of the tag to a variable in default vars and write that variable to the ini and env files, which would mess up git pull, or just go directly to the files. git describe --tags --abbrev=0 could be used at install time. This might also pick up the testing of PRs that can be other branches. (could pick up other attributes of the install like repo and branch too)

jvonau, 2022-06-26:
> Our whole 'versioning' system needs rethinking. If I look at iiab_base_ver and iiab_revision I would think we are at release 8.0.0, but 8 has never been released. 

I tend to agree, the version really is 7.2, plus a bunch of commits since release-7.2 with some periodic tagging, 27757f283fd727e02dcd3781b202ffaa68eeaca3 introduced that confusion.
> The release/tag says 8.0 Preview 2. Such releases are completely arbitrary, made up by @holta with no community input.

I don't disagree, but there is some order to the naming I would hope. 
> We have never really used an n.m.p release naming convention.

Just n.m with an extremely extended time between releases. Might of been clearer to just bump iiab_revision to one on 'master' just after the release-7.2 branch point in place of bumping iiab_base_ver. Save changing iiab_base_ver to just before branching for the new release with resetting iiab_revision to zero when needed.
> Then there is the double entry: iiab_base_ver and iiab_revision are maintained both in default vars and as a comment in [roles/0-init/defaults/main.yml](https://github.com/iiab/iiab/blob/79de65425c5dcf09ecf5435a23b5b52fa035941d/roles/0-init/defaults/main.yml)

Think that is more of a note within roles/0-init/defaults/main.yml given 5ed418338f1131f90eaadc88ede4388619c7563b was pushed 6 months before the change to vars/default_vars.yml
> Release/tag is maintained independently.

There has been 4 tags released with 8.0 in the name since the 7.2 release, after each of these tagging events iiab_revision could of been bumped up by one giving a pretty clear connection of the date/githash/tag (ie 7.2.5) that is now in play. I see @tim-moody used a n.m.p like notation within PR #3254 (7.2.4) that is the last commit before '8.0 Preview 2' was released and matches the proposed workflow I have been trying to describe. Thanks for the example.
  
> The simplest and most consistent thing would be to automate the writing of the value of the tag to a variable in default vars and write that variable to the ini and env files, which would mess up git pull, 

Can you expand on 'which would mess up git pull'? I could see a post-tagging action that might bump a variable in the code base on github but fail to see how that might cause a git pull to fail. A local change to default_vars could cause git pull to fail.
>or just go directly to the files. git describe --tags --abbrev=0 could be used at install time. This might also pick up the testing of PRs that can be other branches. (could pick up other attributes of the install like repo and branch too)

How about a code sample of what you propose?

holta, 2022-06-27:
> git describe --tags --abbrev=0

The above works very well, to display the most recent tag that is reachable from a commit.  It can be used iteratively if necessary:

```
# git describe --tags --abbrev=0 8.0-preview-2^
8.0-preview-1
```

In addition, if counting numbers are important, below is an easy to (automatically) get the number of commits since the most recent tag:

```
git log 8.0-preview-2..HEAD --oneline | wc -l
```

jvonau, 2022-06-27:
> > git describe --tags --abbrev=0
> 
> The above works very well, to display the most recent tag that is reachable from a commit. It can be used recursively if necessary, e.g.:
> 
> ```
> # git describe --tags --abbrev=0 8.0-preview-2^
> 8.0-preview-1
> ```
```
sudo git describe --tags --abbrev=0 8.0-preview-2^
fatal: Not a valid object name 8.0-preview-2^
```
That only works (using 8.0-preview-2) when the tag is already in the git tree prior to the initial clone. Consider this:
```
[version]
distribution = Ubuntu
arch = x86_64
iiab_base_ver = 8.0
iiab_branch = master
iiab_commit = be71adc6a1dfb3a125800c19a5af3bf2c0f9209f
install_date = 2022-06-10T15:26:00Z
```
`sudo git describe --tags`
```
8.0-preview-1-169-g76ae9985
```
`sudo git pull`
`sudo git -P tag`
```
6.4-RC1
6.4-RC2
6.4-RC3
6.4-RC4
6.4-RC5
6.4-RC6
6.4-RC7
6.4-RC8
6.4-RC9
6.5-RC1
6.5-RC2
6.5-RC3
6.5-RC4
6.5-RC5
6.5-RC6
6.5-preview-1
6.5-preview-10
6.5-preview-11
6.5-preview-12
6.5-preview-13
6.5-preview-14
6.5-preview-15
6.5-preview-2
6.5-preview-3
6.5-preview-4
6.5-preview-5
6.5-preview-6
6.5-preview-7
6.5-preview-8
6.5-preview-9
6.6-RC1
6.6-RC2
6.6-RC3
6.6-RC4
6.6-preview-1
6.6-preview-2
6.6-preview-3
6.6-preview-4
6.6-preview-5
6.6-preview-6
6.7-RC1
6.7-RC2
6.7-preview-1
6.7-preview-2
7.0-RC1
7.0-RC2
7.0-preview-1
7.0-preview-2
7.0-preview-3
7.1-RC1
7.1-RC2
7.1-RC3
7.1-RC4
7.1-preview-1
7.1-preview-2
7.1-preview-3
7.1-preview-4
7.1.5-premap
7.2-RC1
7.2-RC2
7.2-RC3
7.2-preview-1
7.2-preview-2
7.2-preview-3
8.0-beta-1
8.0-beta-2
8.0-preview-1
release-6.3
release-6.4
release-6.5
release-6.6
release-6.7
release-7.0
release-7.1
release-7.2
v7.0.9
``` 
Note that the most recent tag (8.0-preview-2) is missing but the repo is up to date
`sudo git log` 
```
commit 18aef95d925aea2662169a87ee7b3a639c7ecb09 (origin/master, origin/HEAD, master)
Author: A Holt <holta@users.noreply.github.com>
Date:   Fri Jun 24 14:42:32 2022 -0400

    vars/linuxmint-21.yml: python_ver: "3.10" advance fix re: PR #3260 & PR #3265
```
> In addition, if counting numbers are important, below is an easy way to (automatically) get the number of commits since the most recent tag:
> 
> ```
> git log 8.0-preview-2..HEAD --oneline | wc -l
> ```

Given 8.0-preview-2 does not exist on the local machine, the above fails
```
sudo git log 8.0-preview-2..HEAD --oneline | wc -l
fatal: ambiguous argument '8.0-preview-2..HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
0
```
However using the most recent tag or older from the local machine works
```
sudo git log 8.0-preview-1..HEAD --oneline | wc -l
169
sudo git log release-7.2..HEAD --oneline | wc -l
481
```

jvonau, 2022-06-27:
#3203 uses (7.3.2) part of 8.0-preview-1 #3232 uses (7.3.3) notation part of 8.0-preview-2

tim-moody, 2022-06-27:
In adm cons I use x.x.x.x-desc as my branch pattern, but just to keep my branches straight. I don't do many iiab commits so I use x.x.x-desc.

iiab has no real release schedule or methodology, so the numbers are pretty arbitrary. All we really need is a way to know what someone has installed in order to troubleshoot it. adding a new sub version for every commit seems like a lot of work unless it can be automated at commit time.

I'd like to see repo added to what's being captured in case someone is working on a developer's repo. Adding latest tag could be useful.

holta, 2022-06-27:
> Note that the most recent tag (8.0-preview-2) is missing but the repo is up to date

FWIW I cannot reproduce that.

In my case `git pull` brings in (1) all recent commits, and also (2) all recent tags.
(e.g. tag `8.0-preview-2` appeared immediately after I ran `git pull`, in all the expected places like `git tag` output etc.)

Any idea what's different about how you're running `git pull` ?
FYI my Raspberry Pi 4 says:

```
root@box:/opt/iiab/iiab# git remote -v
origin  https://github.com/iiab/iiab (fetch)
origin  https://github.com/iiab/iiab (push)

root@box:/opt/iiab/iiab# git --version
git version 2.30.2

root@box:/opt/iiab/iiab# uname -a
Linux box 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux

root@box:/opt/iiab/iiab# more /etc/rpi-issue
Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 226b479f8d32919c9f
e36dd5b4c20c02682f8180, stage2
```

holta, 2022-06-28:
> > Note that the most recent tag (8.0-preview-2) is missing but the repo is up to date
> 
> FWIW I cannot reproduce that.
> 
> In my case `git pull` brings in (1) all recent commits, and also (2) all recent tags.

My guess is @jvonau was running something like `git pull origin master` rather than `git pull`

If so, use `--tags` or `-t` to bring in all tags like "8.0-preview-2" :

```
git pull -t origin master
```

(Likely the `git pull remote branch` on [Line 397 of `sudo iiab`](https://github.com/iiab/iiab-factory/blob/0d6f6348e75508b54d10028fbc1faa458af7222e/iiab#L397) should do the same ?)

jvonau, 2022-06-28:
> iiab has no real release schedule or methodology, so the numbers are pretty arbitrary. 

Numbers can be incremented in a meaningful way.
>All we really need is a way to know what someone has installed in order to troubleshoot it. 

Looking up a git hash is not fun when the install date is not current.
>adding a new sub version for every commit seems like a lot of work unless it can be automated at commit time.

Who asked for with every commit? I suggested based on a tag point, so that is less frequent, I'd say periodic at best.
> I'd like to see repo added to what's being captured in case someone is working on a developer's repo. Adding latest tag could be useful.

One has no clue from where the initial clone originated from, and can't tell the difference if the dev repo was maintained in sync with the main iiab repo. 

> > > Note that the most recent tag (8.0-preview-2) is missing but the repo is up to date
> > 
> > 
> > FWIW I cannot reproduce that.
> > In my case `git pull` brings in (1) all recent commits, and also (2) all recent tags.

When was the repo cloned in your example?
> My guess is @jvonau was running something like `git pull origin master` rather than `git pull`

Yup, I can have multiple branches and/or upstream repos defined.
> If so, use `--tags` or `-t` to bring in all tags like "8.0-preview-2" :
> 
> ```
> git pull -t origin master
> ```

Why? I get the latest code without -t according to git log
> (Likely the `git pull <remote> <branch>` on [Line 397 of `sudo iiab`](https://github.com/iiab/iiab-factory/blob/0d6f6348e75508b54d10028fbc1faa458af7222e/iiab#L397) should do the same ?)

Not really, the tags would be embedded within the initial git clone, what is needed is a local branch to pull the PR into, even if the branch is named test-branch which might be better than recording 'master' and a meaningless git hash without context within iiab.ini when testing/reporting on a PR.

tim-moody, 2022-06-28:
>One has no clue from where the initial clone originated from, and can't tell the difference if the dev repo was maintained in sync with the main iiab repo.

that's exactly why I'd like repo added to the captured data

jvonau, 2022-06-28:
> > One has no clue from where the initial clone originated from, and can't tell the difference if the dev repo was maintained in sync with the main iiab repo.
> 
> that's exactly why I'd like repo added to the captured data

Wondering what the magic value would be that would reveal the repo value. 

At any rate since the concept of default_vars.yml being the starting  defaults for only local_vars.yml to help with the distinction between a user override-able values and processing defaults went out the window long ago a sharp user or downstream developer is still free to use whatever values wanted by them via local_vars for their workflows. Think I'll start doing that with my fork, just to have some fun with the reporting.

jvonau, 2022-06-28:
> > When was the repo cloned in your example?
> 
> Somewhere between "8.0-preview-1" and "8.0-preview-2"
> 
> In any case, the larger point is that commands like "git pull" (and others) effortlessly bring in meaningful tags for those that care.

Try a 'git pull' when you are on a testing|work-in-progress branch and master just changed now you want to pull those changes into the current branch. Here is a simple reproducer:
```
jvonau@kickass:/mnt/scratch/git/iiab$ git show
commit 18aef95d925aea2662169a87ee7b3a639c7ecb09 (HEAD -> master, origin/master)
Author: A Holt <holta@users.noreply.github.com>
Date:   Fri Jun 24 14:42:32 2022 -0400

    vars/linuxmint-21.yml: python_ver: "3.10" advance fix re: PR #3260 & PR #3265

diff --git a/vars/linuxmint-21.yml b/vars/linuxmint-21.yml
index a6403ed91..0135cd65f 100644
--- a/vars/linuxmint-21.yml
+++ b/vars/linuxmint-21.yml
@@ -28,4 +28,4 @@ sshd_service: ssh
 php_version: 8.1
 postgresql_version: 14
 systemd_location: /lib/systemd/system
-python_ver: 3.10
+python_ver: "3.10"
```
`git checkout -b test-demo`
```
jvonau@kickass:/mnt/scratch/git/iiab$ git pull
remote: Enumerating objects: 74, done.
remote: Counting objects: 100% (40/40), done.
remote: Total 74 (delta 40), reused 40 (delta 40), pack-reused 34
Unpacking objects: 100% (74/74), 15.99 KiB | 629.00 KiB/s, done.
From https://github.com/iiab/iiab
 * [new branch]          release-6.3    -> origin/release-6.3
 * [new branch]          release-6.4    -> origin/release-6.4
 * [new branch]          release-6.5    -> origin/release-6.5
 * [new branch]          release-6.6    -> origin/release-6.6
 * [new branch]          release-6.7    -> origin/release-6.7
 * [new branch]          release-7.0    -> origin/release-7.0
 * [new branch]          release-7.1    -> origin/release-7.1
 * [new branch]          release-7.2    -> origin/release-7.2
 * [new tag]             6.4-RC1        -> 6.4-RC1
 * [new tag]             6.4-RC2        -> 6.4-RC2
 * [new tag]             6.4-RC3        -> 6.4-RC3
 * [new tag]             6.4-RC4        -> 6.4-RC4
 * [new tag]             6.4-RC5        -> 6.4-RC5
 * [new tag]             6.4-RC6        -> 6.4-RC6
 * [new tag]             6.4-RC7        -> 6.4-RC7
 * [new tag]             6.4-RC8        -> 6.4-RC8
 * [new tag]             6.4-RC9        -> 6.4-RC9
 * [new tag]             6.5-RC1        -> 6.5-RC1
 * [new tag]             6.5-RC2        -> 6.5-RC2
 * [new tag]             6.5-RC3        -> 6.5-RC3
 * [new tag]             6.5-RC4        -> 6.5-RC4
 * [new tag]             6.5-RC5        -> 6.5-RC5
 * [new tag]             6.5-RC6        -> 6.5-RC6
 * [new tag]             6.5-preview-1  -> 6.5-preview-1
 * [new tag]             6.5-preview-10 -> 6.5-preview-10
 * [new tag]             6.5-preview-11 -> 6.5-preview-11
 * [new tag]             6.5-preview-12 -> 6.5-preview-12
 * [new tag]             6.5-preview-13 -> 6.5-preview-13
 * [new tag]             6.5-preview-14 -> 6.5-preview-14
 * [new tag]             6.5-preview-15 -> 6.5-preview-15
 * [new tag]             6.5-preview-2  -> 6.5-preview-2
 * [new tag]             6.5-preview-3  -> 6.5-preview-3
 * [new tag]             6.5-preview-4  -> 6.5-preview-4
 * [new tag]             6.5-preview-5  -> 6.5-preview-5
 * [new tag]             6.5-preview-6  -> 6.5-preview-6
 * [new tag]             6.5-preview-7  -> 6.5-preview-7
[... 41 lines cut ...]
 * [new tag]             release-6.4    -> release-6.4
 * [new tag]             release-6.5    -> release-6.5
 * [new tag]             release-6.6    -> release-6.6
 * [new tag]             release-6.7    -> release-6.7
 * [new tag]             release-7.0    -> release-7.0
 * [new tag]             release-7.1    -> release-7.1
 * [new tag]             release-7.2    -> release-7.2
 * [new tag]             v7.0.9         -> v7.0.9
There is no tracking information for the current branch.
Please specify which branch you want to merge with.
See git-pull(1) for details.

    git pull <remote> <branch>

If you wish to set tracking information for this branch you can do so with:

    git branch --set-upstream-to=<remote>/<branch> test-demo
```
Probability the first time in my life I have ever used a bare 'git pull' without \<remote> \<branch> being present as tags are not really overly useful they are just a pointer to a prior git commit hash.
```
jvonau@kickass:/mnt/scratch/git/iiab$ git pull origin master
From https://github.com/iiab/iiab
 * branch                master     -> FETCH_HEAD
Already up to date.
```

holta, 2022-06-28:
This is a good point that there exist quite different ways to test a PR, each very useful in their own right:

1) Methods like `git clone http://github.com/iiab/iiab` then `git pull <pr-remote> <pr-branch>`
  (e.g. this is what `curl iiab.io/install.txt | bash -s 361 3267` does for you)
2) Methods like `git clone <pr-remote> <pr-branch>` (e.g. working off an older branch as is, evolving it, sharing it, etc)

Recent tags obviously being much more visible when using Testing Methods similar to 1. above.  And some annoying subtleties "in between" both above Methods, for sure.

So streamlining different testing workflows _wherever possible (the most common ones anyway, and making them very clear to newcomers)_ is definitely an extremely useful track.

jvonau, 2022-06-29:
> > Wondering what the magic value would be that would reveal the repo value.
> 
> git remote -v?
```
jvonau@kickass:/mnt/scratch/git/iiab$ git remote -v
ah	https://github.com/holta/iiab (fetch)
ah	https://github.com/holta/iiab (push)
jv	https://github.com/jvonau/iiab (fetch)
jv	https://github.com/jvonau/iiab (push)
origin	https://github.com/iiab/iiab (fetch)
origin	https://github.com/iiab/iiab (push)
```
Ok that should show from where the clone originated from as long as .git/config was not overly meddled with. 
> > the concept of default_vars.yml being the starting defaults for only local_vars.yml
> 
> if you're suggesting these values don't belong in default vars I totally agree. default and local vars should be settable install parameters. get all the reportable values from local facts and write them to env or ini

Nice, we have common ground. Lets leave that for a different day, but thinking out load vars/default_vars.yml could be reworked with vars/newname_vars.yml where newname would hold what is not wanted within default_vars and newname would be added to wherever the booleans need to be used. Basically move the [boolean entries](https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L746-L786) to newname_vars.yml and add vars/newname_vars.yml to the top level playbooks 'vars_files:' entries

```
  vars_files:
    - roles/0-init/defaults/main.yml
    - vars/default_vars.yml
    - vars/newname_vars.yml
    - vars/{{ ansible_local.local_facts.os_ver }}.yml
    - /etc/iiab/local_vars.yml
    - /etc/iiab/iiab_state.yml

```
'newname' is up for suggestions.

tim-moody, 2022-06-29:
>Ok that should show from where the clone originated from as long as .git/config was not overly meddled with.

I think it will always be origin:

```
root@box:/tmp# git clone https://github.com/tim-moody/iiab-admin-console
Cloning into 'iiab-admin-console'...
remote: Enumerating objects: 11446, done.
remote: Counting objects: 100% (1910/1910), done.
remote: Compressing objects: 100% (831/831), done.
remote: Total 11446 (delta 1121), reused 1828 (delta 1062), pack-reused 9536
Receiving objects: 100% (11446/11446), 8.36 MiB | 17.16 MiB/s, done.
Resolving deltas: 100% (6290/6290), done.
root@box:/tmp# cd iiab-admin-console/
root@box:/tmp/iiab-admin-console# git remote -v
origin  https://github.com/tim-moody/iiab-admin-console (fetch)
origin  https://github.com/tim-moody/iiab-admin-console (push)

```

holta, 2022-06-29:
> I think it will always be origin:

That's typical, for sure, but iiab-diagnostics and iiab-summary commands don't make that assumption so as to verify.

They figure out the name of the remote from the branch.  Here are 2 examples:

- https://github.com/iiab/iiab/blob/master/scripts/iiab-summary#L14-L15
- https://github.com/iiab/iiab/blob/master/scripts/iiab-summary#L24-L25

tim-moody, 2022-06-29:
>but iiab-diagnostics and iiab-summary commands don't make that assumption so as to verify.

I noticed. would be better to capture the values on install and report elsewhere that to have multiple algorithm for capturing

also more accurate. if I pull and don't do an install the values could be misleading

holta, 2022-06-30:
> even if the branch is named test-branch which might be better than recording 'master'

@jvonau I agree something like is very likely best.

It needs tuning to be very understandable (making 100% clear what the branch name is, possibly even overridable with an extremely simple UX).

Devil's in the details (it might even checking that the branch doesn't already exist, without drowning in too many corner cases) but yes working something out here this year is definitely a very good idea :heavy_check_mark: