iiab/iiab #3278: Mint has traceback in networkd-dispatcher

jvonau, 2022-06-30:
https://github.com/iiab/iiab/issues/3106#issuecomment-1140557739
>Secondary issue noted, networkd-dispatcher has a traceback, I suspect the cause is the same as found https://github.com/iiab/iiab/pull/2645, need to confirm the package version installed with apt list networkd-dispatcher before suggesting the workaround be extended to LinuxMint.

From http://sprunge.us/CX089U
```
=IIAB==========================================================================
COMMAND: /usr/bin/sudo journalctl -b 0 -u networkd-dispatcher    # networkd-dispatcher log

-- Logs begin at Fri 2022-05-27 07:29:42 BST, end at Sun 2022-05-29 16:17:07 BST. --
May 29 14:45:13 box systemd[1]: Starting Dispatcher daemon for systemd-networkd...
May 29 14:45:14 box networkd-dispatcher[784]: WARNING: systemd-networkd is not running, output will be incomplete.
May 29 14:45:14 box networkd-dispatcher[582]: ERROR:Unknown state for interface NetworkctlListState(idx=1, name='lo', type='loopback', operational='n/a', administrative='unmanaged'): n/a
May 29 14:45:14 box networkd-dispatcher[582]: Traceback (most recent call last):
May 29 14:45:14 box networkd-dispatcher[582]:   File "/usr/bin/networkd-dispatcher", line 298, in trigger_all
May 29 14:45:14 box networkd-dispatcher[582]:     self.handle_state(iface_name,
May 29 14:45:14 box networkd-dispatcher[582]:   File "/usr/bin/networkd-dispatcher", line 348, in handle_state
May 29 14:45:14 box networkd-dispatcher[582]:     raise UnknownState(operational_state)
May 29 14:45:14 box networkd-dispatcher[582]: UnknownState: n/a
May 29 14:45:14 box networkd-dispatcher[582]: ERROR:Unknown state for interface NetworkctlListState(idx=2, name='eno1', type='ether', operational='n/a', administrative='unmanaged'): n/a
May 29 14:45:14 box networkd-dispatcher[582]: Traceback (most recent call last):
May 29 14:45:14 box networkd-dispatcher[582]:   File "/usr/bin/networkd-dispatcher", line 298, in trigger_all
May 29 14:45:14 box networkd-dispatcher[582]:     self.handle_state(iface_name,
May 29 14:45:14 box networkd-dispatcher[582]:   File "/usr/bin/networkd-dispatcher", line 348, in handle_state
May 29 14:45:14 box networkd-dispatcher[582]:     raise UnknownState(operational_state)
May 29 14:45:14 box networkd-dispatcher[582]: UnknownState: n/a
May 29 14:45:14 box networkd-dispatcher[582]: ERROR:Unknown state for interface NetworkctlListState(idx=3, name='wlan0', type='wlan', operational='n/a', administrative='unmanaged'): n/a
May 29 14:45:14 box networkd-dispatcher[582]: Traceback (most recent call last):
May 29 14:45:14 box networkd-dispatcher[582]:   File "/usr/bin/networkd-dispatcher", line 298, in trigger_all
May 29 14:45:14 box networkd-dispatcher[582]:     self.handle_state(iface_name,
May 29 14:45:14 box networkd-dispatcher[582]:   File "/usr/bin/networkd-dispatcher", line 348, in handle_state
May 29 14:45:14 box networkd-dispatcher[582]:     raise UnknownState(operational_state)
May 29 14:45:14 box networkd-dispatcher[582]: UnknownState: n/a
May 29 14:45:14 box systemd[1]: Started Dispatcher daemon for systemd-networkd.
```

holta, 2022-06-30:
Linux Mint communities seem to be growing every year, in Africa and elsewhere.

Even in Mint 21 does not solve every last bug when it's released in coming weeks ;)

In any case, this comment should later be amended or removed as appropriate:

https://github.com/iiab/iiab/blob/46fad7936e3db81e05c8be9cc1d2874b9a9ca5ab/roles/network/tasks/netplan.yml#L62-L77

jvonau, 2022-06-30:
The issue present today, nothing bubbling up on single interface machines, but a wifi-hotspot might be affected.

jvonau, 2022-07-16:
#3307 rxJ3B2?en
1211 =IIAB==========================================================================
1212 COMMAND: /usr/bin/sudo journalctl -b 0 -u networkd-dispatcher    # networkd-dispatcher log
1213 
1214 Jul 13 21:22:43 box systemd[1]: Starting Dispatcher daemon for systemd-networkd...
1215 Jul 13 21:22:45 box networkd-dispatcher[696]: WARNING: systemd-networkd is not running, output will be incomplete.
1216 Jul 13 21:22:45 box networkd-dispatcher[572]: ERROR:Unknown state for interface NetworkctlListState(idx=1, name='lo', type='loopback', operational='n/a', administrative='unmanaged'): n/a
1217 Jul 13 21:22:45 box networkd-dispatcher[572]: Traceback (most recent call last):
1218 Jul 13 21:22:45 box networkd-dispatcher[572]:   File "/usr/bin/networkd-dispatcher", line 298, in trigger_all
1219 Jul 13 21:22:45 box networkd-dispatcher[572]:     self.handle_state(iface_name,
1220 Jul 13 21:22:45 box networkd-dispatcher[572]:   File "/usr/bin/networkd-dispatcher", line 348, in handle_state
1221 Jul 13 21:22:45 box networkd-dispatcher[572]:     raise UnknownState(operational_state)
1222 Jul 13 21:22:45 box networkd-dispatcher[572]: UnknownState: n/a
1223 Jul 13 21:22:45 box networkd-dispatcher[572]: ERROR:Unknown state for interface NetworkctlListState(idx=2, name='enp0s3', type='ether', operational='n/a', administrative='unmanaged'): n/a
1224 Jul 13 21:22:45 box networkd-dispatcher[572]: Traceback (most recent call last):
1225 Jul 13 21:22:45 box networkd-dispatcher[572]:   File "/usr/bin/networkd-dispatcher", line 298, in trigger_all
1226 Jul 13 21:22:45 box networkd-dispatcher[572]:     self.handle_state(iface_name,
1227 Jul 13 21:22:45 box networkd-dispatcher[572]:   File "/usr/bin/networkd-dispatcher", line 348, in handle_state
1228 Jul 13 21:22:45 box networkd-dispatcher[572]:     raise UnknownState(operational_state)
1229 Jul 13 21:22:45 box networkd-dispatcher[572]: UnknownState: n/a
1230 Jul 13 21:22:45 box systemd[1]: Started Dispatcher daemon for systemd-networkd.

holta, 2022-07-18:
@jvonau Should this be close?

(Or does it need more work and/or verification?)

jvonau, 2022-07-19:
91b037940c341a115fa817a09eb760366f5b3ca5 has a note on what might be needed if someone does use the hotspot functionality and that change breaks dnsmasq somehow.