iiab/iiab #3283: IIAB install froze at "pip install 7 packages into virtual environment: /opt/iiab/jupyterhub (~217 MB)"

holta, 2022-07-03:
@georgejhunt would you have any recommendations to diagnose/mitigate ?

It's been completely stuck for more than 30 minutes now:

```
root@box:~# du -bsc /opt/iiab/jupyterhub/*
18904   /opt/iiab/jupyterhub/bin
16384   /opt/iiab/jupyterhub/etc
4096    /opt/iiab/jupyterhub/include
4793374 /opt/iiab/jupyterhub/lib
3       /opt/iiab/jupyterhub/lib64
69      /opt/iiab/jupyterhub/pyvenv.cfg
2277704 /opt/iiab/jupyterhub/share
7110534 total
```

iiab-summary (below) shows temperature is 75C, but everything else is running fine:

```
root@box:~# uptime
 19:04:12 up 56 min,  3 users,  load average: 1.07, 1.24, 1.22
```

```
root@box:~# iiab-summary
install_date = 2022-07-03T18:13:38Z   Current TZ: UTC

iiab: f2f4afb3, 19 PR's / 70 commits since tag 8.0-preview-2
    "Merge pull request #3282 from holta/iiab-summary_early_on"
    https://github.com/iiab/iiab   branch: master

iiab-admin-console: c6c6b9d, 43 PR's / 123 commits since tag v0.5.5.9
    "Merge pull request #505 from tim-moody/0.6.0.4-enh2"
    https://github.com/iiab/iiab-admin-console   branch: master

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2
/etc/debian-version: 11.3
display-manager? inactive   Arch1: arm64   Arch2: armhf
5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 unknown
Cortex-A72   CPU(s): 4   Mem: 1849 Swap: 1023
Raspberry Pi 4 Model B Rev 1.4
Temperature(s): 74984

/opt/iiab/iiab/scripts/iiab-apps-to-be-installed: line 18: /tmp/iiab-apps-list: Permission denied
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        59G  7.8G   50G  14% /      ZIMs: 0 OER2Go: 0 Apps2B: 20

lo eth0 wlan0 tun0
openvpn_enabled: True
openvpn_handle: 181-rpi4-lite-LARGE-0703
192.168.0.181 10.8.0.22
```

iiab-diagnostics: http://sprunge.us/MKNY8z?en

holta, 2022-07-03:
FYI hitting CTRL-C and then `sudo iiab` does not seem move it forward it any way &mdash; once again Ansible gets stuck at `pip install 7 packages into virtual environment: /opt/iiab/jupyterhub (~217 MB)` and not even 1 more byte is downloaded/installed here:

```
root@box:~# du -bsc /opt/iiab/jupyterhub/*
18904   /opt/iiab/jupyterhub/bin
16384   /opt/iiab/jupyterhub/etc
4096    /opt/iiab/jupyterhub/include
4793374 /opt/iiab/jupyterhub/lib
3       /opt/iiab/jupyterhub/lib64
69      /opt/iiab/jupyterhub/pyvenv.cfg
2277704 /opt/iiab/jupyterhub/share
7110534 total
```

I tried rebooting and then `sudo iiab` which results in exactly the same!

Suggestions?

georgejhunt, 2022-07-03:
> any way — once again Ansible gets stuck at pip install 7 packages into

I guess I'd do a hand install of each of the items in requirements.txt and
see which one fails.
Remember to activate the virtual environment with
"/opt/iiab/jupyterhub/bin/activate" and then run the pip installs

holta, 2022-07-03:
> I guess I'd do a hand install of each of the items in requirements.txt and see which one fails.

1. I don't see any requirements.txt

2. Probably you mean to try the 7 by hand:

https://github.com/iiab/iiab/blob/8aef9d8fc87a83214ff2f059f2f465b3c96a6d8a/roles/jupyterhub/tasks/install.yml#L36-L50

> Remember to activate the virtual environment with "/opt/iiab/jupyterhub/bin/activate" and then run the pip installs

3. That doesn't work:

```
root@box:~# /opt/iiab/jupyterhub/bin/activate
-bash: /opt/iiab/jupyterhub/bin/activate: Permission denied

root@box:~# ls -l /opt/iiab/jupyterhub/bin/activate
-rw-r--r-- 1 root root 1901 Jul  3 18:30 /opt/iiab/jupyterhub/bin/activate
```

4. I'm out for most of the rest of the day, so feel free to log in directly if you have a minute?  (10.8.0.22)

georgejhunt, 2022-07-04:
>    1. That doesn't work:
>    1. I'm out for most of the rest of the day, so feel free to log in

you need to activate as root

holta, 2022-07-04:
> you need to activate as root

User was root.  The word `source` had been missing:

```
source /opt/iiab/jupyterhub/bin/activate
```

I then ran these 7 commands:

```
pip install pip
pip install wheel
pip install ipywidgets
pip install jupyterhub
pip install jupyterlab
pip install jupyterhub_firstuseauthenticator
pip install jupyterhub-systemdspawner
```

After that `sudo iiab` ran correctly.

(So I have no idea why it failed, and why the above apparently fixed things!  But this particular installation is now back on its feet...)

georgejhunt, 2022-07-04:
> I'll leave this ticket open for now, in case this prevents other IIAB
> installs in coming weeks?
>
> (Hoping this might lead to an explanation and/or preventive measures to
> avoid this happening again?)

looking at what stalled:
- name: "pip install 7 packages into virtual environment: {{
jupyterhub_venv }} (~217 MB)"
pip:
name:
- pip
- wheel
- ipywidgets
- jupyterhub
- jupyterlab
- jupyterhub_firstuseauthenticator
- jupyterhub-systemdspawner
virtualenv: "{{ jupyterhub_venv }}" # /opt/iiab/jupyterhub
virtualenv_site_packages: no
virtualenv_command: python3 -m venv "{{ jupyterhub_venv }}" # 2021-07-29:
This works on RasPiOS 10, Debian 11, Ubuntu 20.04 and Mint 20 -- however if
you absolutely must use the older Debian 10 -- you can work around errors
"can't find Rust compiler" and "This package requires Rust >=1.41.0" if you
(1) revert this line to 'virtualenv_command: virtualenv' AND (2) uncomment
the line just below
#virtualenv_python: python3 # 2021-07-29: Was needed when above line was
'virtualenv_command: virtualenv' (generally for Python 2)
extra_args: "--no-cache-dir --pre" # 2021-11-30: The "--pre" flag should
likely be removed after JupyterHub 2.0.0 is released.

Pip and wheel are used by the python pip installer.  Maybe they should be
pulled out into separate plays -- working on the hypothesis that failing
mutual dependency, or some sort of lockout during update of installer, was
causing the stall.-- if it continues to be a problem.  You could also
comment out pip and wheel, and just live with the versions that are default
(they may be good enough).

holta, 2022-07-04:
Just to highlight 1 of @georgejhunt's above suggestions for consideration:

> comment out pip and wheel, and just live with the versions that are default
(they may be good enough)

holta, 2022-07-05:
@georgejhunt I've confirmed [roles/jupyterhub](https://github.com/iiab/iiab/tree/master/roles/jupyterhub) fails every time on fresh OS installs of 64-bit RasPiOS Lite on RPi 4.

And yet this does not affect installs on Ubuntu Server 22.04 on x86_64.

Would you have a moment this week to investigate?  Here's the latest failed install:

```
root@raspberrypi:~# iiab-summary
install_date = 2022-07-05T17:15:46Z   Current TZ: UTC

iiab: 9e39d421, 21 PR's / 76 commits since tag 8.0-preview-2
    "firmware/tasks/download.yml: Tighter iiab_download_url + with_items"
       branch: tls-ssl-downloads

/etc/iiab/pr-list-pulled:
3288 2022-07-05 17:13:55 UTC

iiab-admin-console: c6c6b9d, 43 PR's / 123 commits since tag v0.5.5.9
    "Merge pull request #505 from tim-moody/0.6.0.4-enh2"
    https://github.com/iiab/iiab-admin-console   branch: master

Raspberry Pi reference 2022-04-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 27a8050c3c06e567c794620394a8c2d74262a516, stage2
/etc/debian-version: 11.3
display-manager? inactive   Arch1: arm64   Arch2: armhf
5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 unknown
Cortex-A72   CPU(s): 4   Mem: 1849 Swap: 1023
Raspberry Pi 4 Model B Rev 1.4
Temperature(s): 67192

Filesystem      Size  Used Avail Use% Mounted on
/dev/root        59G  5.6G   52G  10% /      ZIMs: 0 OER2Go: 0 Apps2B: 6

lo eth0 wlan0 tun0
openvpn_enabled: True
openvpn_handle: 181-rpi4-lite64-SMALL-Jupyt-3288
192.168.0.181 10.8.0.22
```

holta, 2022-07-07:
> Just to highlight 1 of @georgejhunt's above suggestions for consideration:
> 
> > comment out pip and wheel, and just live with the versions that are default
> > (they may be good enough)

@jvonau suggests this approach might indeed be promising!

jvonau, 2022-07-07:
Can you confirm the install on ubuntu-22.04 actually works correctly? Think one needs to research the difference between [jupyterhub and jupyterlab](https://stackoverflow.com/questions/57067651/what-is-the-difference-between-jupyterlab-and-jupyterhub) and figure out the correct instructions to [install](https://jupyter.org/install) everything. If you are trying the one by one hand approach with `source /opt/iiab/jupyterhub/bin/activate` post some useful info along the way like `pip list` between each step. By the way you exit the virtual environment with `deactivate`

jvonau, 2022-07-07:
This role needs the same treatment as moodle & sugarizer [have](https://github.com/iiab/iiab/blob/master/roles/7-edu-apps/tasks/main.yml,) just targeting RasPiOS-arm64 because the role is part of ['large'](https://github.com/iiab/iiab/blob/master/vars/local_vars_large.yml#L244-L246). Developmental work can still be done on RasPiOS-arm64 via `runrole`

holta, 2022-07-07:
> This role needs the same treatment as moodle & sugarizer [have](https://github.com/iiab/iiab/blob/master/roles/7-edu-apps/tasks/main.yml,) just targeting RasPiOS-arm64 because the role is part of [...]

Yes.  And in this case it's even worse, as Ansible hangs (essentially forever) so it's not possible to trap the error using `skip_role_on_error`:

- PR #3262

holta, 2022-07-07:
I too have confirmed George's fix using:

```
rm -rf /opt/iiab/jupyterhub
cd /opt/iiab/iiab
sudo ./runrole --reinstall jupyterhub
```

Which works (there was a "500 internal server error" apparently due to the reinstall, which resolved itself on reboot, good enough!)

closed 2022-07-07 by holta