iiab/iiab #3286: dnsmasq failure when installing IIAB 8.0 fast

avni, 2022-07-04:
### Expected behavior and actual behavior:
Expecting successful install. Getting: 
TASK [1-prep : Don't use stock dnsmasq systemd unit file during boot but start now] ********************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service dnsmasq: Job for dnsmasq.service failed because the control process exited with error code.\nSee \"systemctl status dnsmasq.service\" and \"journalctl -xeu dnsmasq.service\" for details.\n"}

PLAY RECAP ******************************************************************************
127.0.0.1                  : ok=34   changed=13   unreachable=0    failed=1    skipped=2    rescued=0    ignored=0   

### Steps to reproduce the problem:
curl iiab.io/fast.txt | sudo bash -s 3173

### IIAB version, OS version, hardware details:
**IIAB:** 
IIAB 8.0 (http://iiab.io/fast.txt)

**OS:** 
iiab-admin@pete-SEi:~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04 LTS
Release:	22.04
Codename:	jammy

**Hardware:** 
iiab-admin@pete-SEi:~$ lshw
WARNING: you should run this program as super-user.
stow-vm4-zero-pr3173        
    description: Computer
    width: 64 bits
    capabilities: smp vsyscall32
  *-core
       description: Motherboard
       physical id: 0
     *-memory
          description: System memory
          physical id: 0
          size: 24GiB
     *-cpu
          product: Intel(R) Core(TM) i5-8279U CPU @ 2.40GHz
          vendor: Intel Corp.
          physical id: 1
          bus info: cpu@0
          version: 6.142.10
          size: 1516MHz
          capacity: 4100MHz
          width: 64 bits

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

http://sprunge.us/GAjsXP

avni, 2022-07-04:
This may be due to me running 'sudo iiab' post-install on the host machine instead of the VM (though I think technically the install should work on both).

holta, 2022-07-04:
> on the host machine instead of the VM

I've confirmed an IIAB install was accidentally attempted on the host PC instead of the VM starting 10:10 am EDT this morning!  This might seem regrettable, but in fact it's an incredible learning opportunity.  Here's why:

Very unexpectedly IIAB's install (very thankfully!!) failed on this Ubuntu Desktop 22.04 host PC &mdash; and failed **very early on** during IIAB's install (incredible convenient!!) so as to cause very little damage.

_I'd love to understand why_ :smile:

Until then, here are a couple observations + suggestions to help unroll (undo) the accidental changes to the host PC:

```
iiab-admin@pete-SEi:~$ systemctl status dnsmasq
× dnsmasq.service - dnsmasq - A lightweight DHCP and caching DNS server
     Loaded: loaded (/lib/systemd/system/dnsmasq.service; disabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Mon 2022-07-04 10:27:39 EDT; 6h ago
    Process: 22775 ExecStartPre=/etc/init.d/dnsmasq checkconfig (code=exited, status=0/SUCCESS)
    Process: 22783 ExecStart=/etc/init.d/dnsmasq systemd-exec (code=exited, status=2)
        CPU: 17ms

iiab-admin@pete-SEi:~$ sudo journalctl -u dnsmasq
Jul 04 10:27:35 stow-vm4-zero-PR3173 systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Jul 04 10:27:35 stow-vm4-zero-PR3173 dnsmasq[22503]: dnsmasq: failed to create listening socket for port 53: Address already in use
Jul 04 10:27:35 stow-vm4-zero-PR3173 dnsmasq[22503]: failed to create listening socket for port 53: Address already in use
Jul 04 10:27:35 stow-vm4-zero-PR3173 dnsmasq[22503]: FAILED to start up
Jul 04 10:27:35 stow-vm4-zero-PR3173 systemd[1]: dnsmasq.service: Control process exited, code=exited, status=2/INVALIDARGUMENT
Jul 04 10:27:35 stow-vm4-zero-PR3173 systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
Jul 04 10:27:35 stow-vm4-zero-PR3173 systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
Jul 04 10:27:39 stow-vm4-zero-PR3173 systemd[1]: Starting dnsmasq - A lightweight DHCP and caching DNS server...
Jul 04 10:27:39 stow-vm4-zero-PR3173 dnsmasq[22783]: dnsmasq: failed to create listening socket for 10.161.33.1: Address already in use
Jul 04 10:27:39 stow-vm4-zero-PR3173 dnsmasq[22783]: failed to create listening socket for 10.161.33.1: Address already in use
Jul 04 10:27:39 stow-vm4-zero-PR3173 dnsmasq[22783]: FAILED to start up
Jul 04 10:27:39 stow-vm4-zero-PR3173 systemd[1]: dnsmasq.service: Control process exited, code=exited, status=2/INVALIDARGUMENT
Jul 04 10:27:39 stow-vm4-zero-PR3173 systemd[1]: dnsmasq.service: Failed with result 'exit-code'.
Jul 04 10:27:39 stow-vm4-zero-PR3173 systemd[1]: Failed to start dnsmasq - A lightweight DHCP and caching DNS server.
```

And here are my quick suggestions for a checklist of things to try to unroll (undo) all-or-most-all traces of IIAB, prior to rebooting this host PC:

1. Remove `stow-vm4-zero-PR3173.lan`, `stow-vm4-zero-PR3173` (and maybe a few other things?) should be removed from /etc/hosts here:

   ```
   127.0.0.1     stow-vm4-zero-PR3173.lan localhost.localdomain localhost stow-vm4-zero-PR3173 box box.lan
   127.0.1.1       pete-SEi

   # The following lines are desirable for IPv6 capable hosts
   ::1     ip6-localhost ip6-loopback
   fe00::0 ip6-localnet
   ff00::0 ip6-mcastprefix
   ff02::1 ip6-allnodes
   ff02::2 ip6-allrouters
   ```

2. Run `hostnamectl hostname pete-SEi` to fully restore the PC's hostname.

3. Run `dnsdomainname` similar to the above, to try to remove domain `.lan`

4. Run `rm -f /etc/systemd/system/iiab-dnsmasq.service /etc/dnsmasq.d/dnsmasq-iiab`

5. Run `systemctl purge dnsmasq`

6. Other stuff I've likely forgotten?!

7. And after a reboot of the host PC hopefully works well...  sanity check than all its Ethernet/WiFi/VM networking works... and if so then let's clean up more completely with: `sudo rm -rf /etc/iiab /opt/iiab /usr/sbin/iiab`

jvonau, 2022-07-05:
> I'd love to understand why

From GAjsXP:
 982     inet 10.161.33.1/24 scope global mpbr0
1027 mpbr0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
1028         inet 10.161.33.1  netmask 255.255.255.0  broadcast 0.0.0.0
```
1469    1145 ?        00:00:00   daemon.start
1470    1331 ?        00:01:03     lxd
1471    1448 ?        00:00:00       dnsmasq
1472   19021 ?        00:00:00       virtiofsd
1473   19049 ?        00:00:00         virtiofsd
1474    1311 ?        00:00:00   lxcfs
```
lxd started dnsmasq binding 10.161.33.1 to that dnsmasq process.

> Other stuff I've likely forgotten?!

The creation of the iiab-admin user during curl on the main host machine?

jvonau, 2022-07-05:
> Run systemctl purge dnsmasq

systemctl purge dnsmasq
Unknown command verb purge.

Good thing you wrote 'systemctl' and not 'apt', which I think is what you wanted, as apt purge might cause other apt packages to be removed.

jvonau, 2022-07-05:
>Very unexpectedly IIAB's install (very thankfully!!) failed on this Ubuntu Desktop 22.04 host PC — and failed very early on during IIAB's install (incredible convenient!!) so as to cause very little damage.

A real install or any other PR would have done far more damage, 1-prep would of been nearly completed, than my 3173 did. I think that is a vote in favor of the revised sequencing of the network role.

holta, 2022-07-05:
@pickypet helped me restore the host PC an hour ago :heavy_check_mark: 

(It's in great shape and this was a very useful learning experience.)

jvonau, 2022-07-08:
#3264

jvonau, 2022-07-10:
The cause seems to be user error. What is the outstanding issue here?

holta, 2022-07-10:
> The cause seems to be user error. What is the outstanding issue here?

Systemic user error indeed.

Many people make this kind of mistake &mdash; when they have many terminal windows open on the screen, and are in a hurry &mdash; typically when using `multipass shell name-of-vm` especially (as they do not notice when their VM reboots, leading them to run dangerous commands on their host machine w/o realizing it...)

Best we can do is educate people who circumvent instructions (typically by running [fast.txt](https://github.com/iiab/iiab-factory/blob/master/fast.txt) or [risky.txt](https://github.com/iiab/iiab-factory/blob/master/risky.txt)) so that they understand that it's their own doing when they shoot themselves in the foot.

closed 2022-07-10 by holta