iiab/iiab #3287: Single Line Installer is Down

yuvrajmakkena, 2022-07-04:
## Single Line Installer is Down
[image]

holta, 2022-07-04:
Hi @yuvrajmakkena !

What ISP (Internet Service Provider) or mobile phone carrier are you using?

yuvrajmakkena, 2022-07-05:
Hi @holta 
It is blocked by campus network, do we have any other repository?

holta, 2022-07-05:
> Hi @holta It is blocked by campus network, do we have any other repository?

Hi @yuvrajmakkena thanks for explaining:

1) If they don't block GitHub you can also try to run:

   ```
   sudo curl https://raw.githubusercontent.com/iiab/iiab-factory/master/iiab > /usr/sbin/iiab
   sudo chmod 0744 /usr/sbin/iiab
   sudo iiab
   ```

2) However it will try to download several pieces of IIAB will from download.iiab.io so the problem is probably not fully solved!

3) Can you please ask your campus IT department to unblock https://download.iiab.io and anything that ends in *iiab.io* ?  

   What university or school is this &mdash; I ask because this is a humanitarian education project &mdash; and we should try to make sure they understand that!

jvonau, 2022-07-05:
> However it will try to download several pieces of IIAB from download.iiab.io so the problem is probably not fully solved!

Once upon a time there was code to detect and record that situation.

yuvrajmakkena, 2022-07-06:
- Observations
  - I tried to Configure Manual Proxy in Firefox which VPN users outside campus use to get into campus network.(Intranet)
           [image]

  - Its working now.
  
      [image]

jvonau, 2022-07-06:
Think you might want to export those proxy setting to your environment as curl, apt, and ansible don't know about the firefox proxy setting. Depends on the desktop used as to where the gui-tool is located 'setting' -> 'network' -> 'Network Proxy' maybe.

holta, 2022-07-07:
@yuvrajmakkena progress?

(Let us know how you make out!)

holta, 2022-07-15:
Thanks @yuvrajmakkena for making contact.

Let us know if further problems arise.

closed 2022-07-15 by holta