iiab/iiab #3289: Does Debian 'bookworm' nightlies have python2 available or not?

jvonau, 2022-07-06:
https://wiki.debian.org/Python
https://wiki.debian.org/Python/FAQ#Python_2_support
https://cdimage.debian.org/cdimage/weekly-builds/arm64/list-cd/debian-testing-arm64-netinst.list.gz 'python2' not preinstalled

This will affect kalite's role at some point, plan now.

holta, 2022-07-06:
This can be moved to the [IIAB 8.1 Milestone](https://github.com/iiab/iiab/milestone/9) later, if/as necessary.

jvonau, 2022-11-28:
#3399

jvonau, 2022-12-29:
#3450

jvonau, 2023-03-20:
Now would be a good time to harvest the kalite virtual environment by tarring them up and storing them on d.iiab.org, one each for 32bit-arm RasPiOS, 64-bit-arm RasPiOS or Ubuntu, and X86_64 from Ubuntu. Once available for download from the internet the installation could be just retrieve the archive and decompress the files into place without needing to install the virtualenv or python2 (that might disappear soon) apt packages.

holta, 2023-03-20:
This is a very important question 👍

A counterargument is that IIAB should invest what precious time we have in automating Kolibri Channel installs.

(UX & content are both far superior within Kolibri, and field communities now strongly prefer this over KA Lite.)

If we had infinite resources (!) we would of course do both (salvage KA Lite & refine the imperfect Kolibri Channel install process).

Certainly several difficult but strategically necessary choices might be necessary in 2023 🕙

tim-moody, 2023-03-20:
are Tauthon or PyPy viable as alternatives? https://www.infoworld.com/article/3365221/python-2-end-of-life-how-to-survive-the-end-of-python-2.html

jvonau, 2023-03-20:
> This is a very important question +1
> 
> A counterargument is that IIAB should invest what precious time we have in automating Kolibri Channel installs.
> 
> (UX & content are both far superior within Kolibri, and field communities now strongly prefer Kolibri over KA Lite.)
>

If ka-lite has fallen out of favor with 'field communities' then perhaps ka-lite can be omitted from the base install in MIN and maybe MEDIUM local_vars given the ease of use within admin-console to install ka-lite on demand via user selection or auto selection  via presets. 
 
> If we had infinite resources (!) we would of course do both (salvage KA Lite & refine the imperfect Kolibri Channel install process).
>

The idea above is pretty low hanging fruit, just needs a web presence to store the tar files and would have the bonus of not having compilers dragged in needlessly via the installation of python3-pip.
> Certainly several difficult but strategically necessary choices might be necessary in 2023 clock10

The clock is ticking and python2 is going to go away at some point in the future, how soon is up to the distros.

tim-moody, 2023-03-20:
I certainly support the idea of tar files as a backup. What would you tar, /library/ka-lite? I guess there's no mysql db needed.

tim-moody, 2023-03-20:
in addition to automating channels for kolibri, changes will be needed to presets and menuing

holta, 2023-03-20:
> perhaps ka-lite can be omitted

Yes.  KA Lite should completely disappear ideally, as Kolibri becomes more reliably automated.

jvonau, 2023-03-20:
I would harvest /library/ka-lite just after the current role is run prior to running 'kalite manage generate_zone' and the 'English Pack' installation as not to upset what admin-console does after the role is run as the first step. Scrapping the complete install after 'kalite manage generate_zone' and the 'English Pack' installation should be workable also, just don't want to interfere with what admin-console does. If admin-console respects the /etc/iiab/install-flags for ka-lite then those flags can be added to the archive and things should be good to go for using the full scraping of the installation.

tim-moody, 2023-03-20:
I am prepared to start work on implementing additional support for Kolibri when two teachers in the field independently tell me that they are using Kolibri for Khan Academy content and prefer that over Ka-lite.

holta, 2023-03-20:
Just FYI there are thousands of Kolibri educators who've dumped KA Lite.

I'm not sure it's IIAB's job to fix all of Kolibri's installation glitches.

But if we end up contributing here, that would certainly be a positive.

jvonau, 2023-03-20:
#3227 I believe @shanti-bhardwa is using Kolibri out in the wild and has content channels in use.

holta, 2023-03-20:
> Take the first step, stop forcing the kalite installation on every iiab install going forward.

Our Kolibri guidance (doc or video or whatever) needs to improve first.

This is not hard work (lots of people are now installing Kolibri with IIAB) but it will take a few months to cleanly articulate all these common installation tips & tricks.

jvonau, 2023-03-20:
> > Take the first step, stop forcing the kalite installation on every iiab install going forward.
> 
> Our Kolibri guidance (doc or video or whatever) needs to improve first.
>

How does needing better docs for an upstream project relate to de-emphasizing the push for kalite going forward knowing that the role doesn't currently install in the newer distros as noted in #3399 and #3450?
> This is not hard work (lots of people are now installing Kolibri with IIAB) but it will take a few months to cleanly articulate all these common installation tips & tricks.

Those better docs only come together with feedback, I would prefer positive feedback when things run right but all that is ever seen is negative feedback when things go wrong.

tim-moody, 2023-03-20:
>Just FYI there are thousands of Kolibri educators who've dumped KA Lite.

Then it should be pretty easy to name two.

holta, 2023-03-20:
> Then it should be pretty easy to name two.

Some Kolibri educators spend their time here, if you in fact want to develop relationships with them, this is a very good place to start:

https://community.learningequality.org

jvonau, 2023-03-20:
Ref cloning the kalite install https://community.learningequality.org/t/installing-ka-lite-on-inovato-quadra-device/2737/9

tim-moody, 2023-03-20:
you mean like https://community.learningequality.org/t/installing-ka-lite-on-inovato-quadra-device/2737

tim-moody, 2023-03-20:
OK @jvonau  got there first

holta, 2023-03-20:
> Other interesting treads: https://community.learningequality.org/t/kolibri-channels/2753 https://community.learningequality.org/t/kolibri-installation-and-setup-raspberry-pi-4/2715

Thx for highlighting those 🏗️

It's taken a decade (all good things take patience) but Kolibri has made tremendous strides recently.

tim-moody, 2023-03-20:
>Some Kolibri educators spend their time here

seeing some implementers; haven't found any teachers yet

https://community.learningequality.org/t/content-in-khan-academy-vs-ka-lite-vs-kolibri/514/3 is the same experience I had 2 years ago when testing

jvonau, 2023-03-20:
> This is not hard work (lots of people are now installing Kolibri with IIAB) but it will take a few months to cleanly articulate all these common installation tips & tricks.

Knowing Kolibri has their own RasPi [image](https://learningequality.org/download/) I would suspect more Kolibri power users would start with that image rather than IIAB given upstream would respond to issues within their own product first rather than raising an issue here and waiting for a fix.

holta, 2023-03-20:
Around 2019 Kolibri make the decision to distribute across "all" platforms, no matter the OS, no matter the hardware, no matter where in others' software stack Kolibri resides (etc), no matter if used for home schooling or informal libraries or whatever 

Some gaming platforms also pursue this kind of "may a thousand flowers bloom" marketing/outreach strategy.  In any case, very promising.

shanti-bhardwa, 2023-03-20:
yes I agree, kalite should be False as a default. I have always done this
in my local_vars.yml and set kolibri to True.
You can also leave kolibri to False as a default.
Just update our documentation to say this. People can then make the choice
if they still want kalite which become unsupported from our perspective or
choose kolibri.

Regards
Shanti

holta, 2023-03-20:
A video would be better, but I and @shanti-bhardwa have enough experience to put solid Kolibri introductory guidance/documentation together over April/May or so.  Which we will do.

Automation of Kolibri Channel installs (and related implementer workflows) can hopefully increase in due course over coming years.  If not earlier.

> all that is ever seen is negative

You're right.  Negativity predominates across far too many open source communities.

David Farning was far ahead of his time a decade ago, long before trolls became a de facto industry to sew division & r— when he asked people to chin up & contain their toxic negativity — **or "put $1 in the jar."**

_Unfortunately few listened to him in the end..._

(I drove to Wisconsin to meet him after he shut everything down in 2014, where he humbly explained that too many peoples' compulsive negativity... quite simply became intolerable.)

jvonau, 2023-03-21:
So much for having the technical discussion around dropping from the default install, or keeping ka-lite going and how to keep it going in the near future. Nobody is suggesting that ka-lite disappear tomorrow just start herding new users away from putting too much time into using ka-lite, after all with admin-console it very easy to add it when requested. The when requested part can be extended forever if the steps are take create the tar archive for future use, the rest should be just implementation details in the role.

holta, 2023-03-21:
Closed by accident.  I hit the wrong button.

As an aside, I personally appreciate museums & archives.  Even society needs to be more honest, that archival pursuits are fundamentally () a privilege of wealth.

holta, 2023-03-21:
Whereas inspired collections in Kolibri, Kiwix, Calibre-Web (etc!) can be deeply transformational for the world's poor.

(If an engaging local mentor/librarian or parental-like figure is on hand to truly animate + engage minds...)

holta, 2023-03-21:
> just start herding new users away from putting too much time into using ka-lite

Herding is an awful strong word.

But I agree overall.

jvonau, 2023-04-05:
FWIW
```
ubuntu@test-2210-2304-220326:~$ cat /etc/os-release 
PRETTY_NAME="Ubuntu Lunar Lobster (development branch)"
NAME="Ubuntu"
VERSION_ID="23.04"
VERSION="23.04 (Lunar Lobster)"
VERSION_CODENAME=lunar
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=lunar
LOGO=ubuntu-logo

ubuntu@test-2210-2304-220326:~$ apt list python2*
Listing... Done
```

jvonau, 2023-04-05:
https://github.com/iiab/iiab/issues/3526#issuecomment-1497651186

jvonau, 2023-04-05:
> are Tauthon or PyPy viable as alternatives? https://www.infoworld.com/article/3365221/python-2-end-of-life-how-to-survive-the-end-of-python-2.html

Think that might be the only alternative or perhaps using the last deb package(s) released from an older install.
```
/usr/local/kalite/venv/lib/python2.7:
total 20
0 lrwxrwxrwx  1 root root   30 Apr  5 10:17 lib-dynload -> /usr/lib/python2.7/lib-dynload
0 lrwxrwxrwx  1 root root   24 Apr  5 10:17 os.py -> /usr/lib/python2.7/os.py
0 lrwxrwxrwx  1 root root   25 Apr  5 10:17 os.pyc -> /usr/lib/python2.7/os.pyc
4 drwxr-xr-x 11 root root 4096 Apr  5 10:17 site-packages
8 -rw-r--r--  1 root root 6938 Apr  5 10:17 site.py
8 -rw-r--r--  1 root root 6728 Apr  5 10:17 site.pyc
```
The virtual environment is using a symlinks pointing to the system installed python2

jvonau, 2023-04-06:
workable solution found

closed 2023-04-06 by jvonau