iiab/iiab #3298: set-hostname fail abruptly stops install

AgentJeffy, 2022-07-08:
### Expected behavior and actual behavior: 
Initial setup should complete however is failing at TASK [0-init : Set /etc/hostname by running: hostnamectl set-hostname "box"] 

### Steps to reproduce the problem: 
iiab-install or the initial sudo bash install

### IIAB version, OS version, hardware details: 
unknown / DietPi on Raspberry Pi Zero 1 W

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!) 
Can not paste as it gives error saying "pastebinint: command not found"

Have attempted several times but still can not finish the initial setup/configuration as the error below that displays haults the install.

TASK [0-init : Set /etc/hostname by running: hostnamectl set-hostname "box"] ***************************************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ["hostnamectl", "set-hostname", "box"], "delta": "0:00:00.179717", "end": "2022-07-08 18:36:41.864309", "msg": "non-zero return code", "rc": 1, "start": "2022-07-08 18:36:41.684592", "stderr": "\u001b[0;1;31mFailed to connect to bus: No such file or directory\u001b[0m", "stderr_lines": ["\u001b[0;1;31mFailed to connect to bus: No such file or directory\u001b[0m"], "stdout": "", "stdout_lines": []}

AgentJeffy, 2022-07-08:
I have manually set the hostname to "box" in an attempt to get around the error using "dietpi-config". Will update with findings if it succeeds.

jvonau, 2022-07-08:
unsupported OS

holta, 2022-07-08:
Hi @AgentJeffy

[DietPi](https://dietpi.com/)  is not currently supported per:
https://github.com/iiab/iiab/wiki/IIAB-Platforms

fyi IlAB tries to support the most recent version of these 4 OS's especially: 

- Raspberry Pi OS
- Ubuntu
- Linux Mint
- Debian

(If you manually get it to work with DietPi however, can you reply explaining how you did that?)

jvonau, 2022-07-08:
I've been down this road, and I've identified the above issues with corrective PRs. I can guide @AgentJeffy through their journey, would need to see the contents of /etc/os-release should there be interest in supporting DietPi @floydianslips

AgentJeffy, 2022-07-08:
So, no change this time. I could try manually setting all of the hostname settings if I knew what they were. 

But that also raises the question of if the installer will skip the section if it's already correctly configured or not. It doesn't seem like it will.

jvonau, 2022-07-08:
Not without patching, good luck.

AgentJeffy, 2022-07-08:
> Not without patching, good luck.

That's a shame. I think iiab would benefit GREATLY from supporting DietPi. Even on heavy lifting Pi having a lightweight OS would open up more resources additional clients.

Well for the time being I'm going to detach from this venture. Thanks for the input everyone.

closed 2022-07-08 by AgentJeffy

jvonau, 2022-07-08:
Well that is not exactly true setting the hostname to be 'box.lan' might get you past this first issue with hostnamectl or tinkering with local_vars.yml 'iiab_hostname' 'iiab_domain' to match your machine before you start the install. But there will be other issues that bubble up.

jvonau, 2022-07-08:
> > Not without patching, good luck.
> 
> That's a shame. I think iiab would benefit GREATLY from supporting DietPi. Even on heavy lifting Pi having a lightweight OS would open up more resources additional clients.

I've have been trying to open up that path for a long time but I'm blocked by the PTB here.
> Well for the time being I'm going to detach from this venture. Thanks for the input everyone.

I could wipe up a test branch if you so choose, let me know, just make the request at https://github.com/jvonau/iiab/issues